zigpy/zigpy #14: Add support for Xiaomi sensors

ruippeixotog, 2018-02-07:
(previously https://github.com/zigpy/bellows/issues/43)

The protocol used by Xiaomi to connect to their sensors seems to deviate slightly from the ZigBee standard, preventing zigpy from communicating properly with them. It would be great if we could use a generic ZigBee radio to connect Xiaomi sensors to Home Assistant.

jonatanolofsson, 2018-02-08:
Related: https://github.com/zigpy/zigpy/issues/8

rcloran, 2018-02-18:
"slightly"...

iantrich, 2018-04-25:
@ruippeixotog does this include the smart wireless switch? https://xiaomi-mi.com/sockets-and-sensors/xiaomi-mi-wireless-switch/

therve, 2018-06-08:
Can we close this issue? It seems to work well enough now.

AndreasBomholtz, 2018-06-28:
This seams to be fixed in PR #45.

closed 2018-06-28 by AndreasBomholtz

justin8, 2018-07-19:
I'm not sure if this is fixed yet. From the commit history it looks like this fix is in 0.1.0, which is what's running on my hass.io instance, but I still get the following error when adding a v2 Xiaomi motion sensor:

```
2018-07-20 09:17:59 INFO (MainThread) [zigpy.application] Device 0x5dc4 (00:15:8d:00:02:25:3c:19) joined the network
2018-07-20 09:17:59 INFO (MainThread) [zigpy.device] [0x5dc4] Discovering endpoints
Tries remaining: 3
2018-07-20 09:17:59 ERROR (MainThread) [bellows.ezsp] Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/usr/local/lib/python3.6/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
2018-07-20 09:17:59 ERROR (MainThread) [bellows.ezsp] Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/usr/local/lib/python3.6/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
```

therve, 2018-07-23:
I have the same issue, I opened #61 to track it.