zigpy/zigpy #5: Use python3.5+ await syntax

jonatanolofsson, 2018-02-07:
Home assistant is dropping python3.4 support, so it's time to update the bellows syntax to use the await syntax. This also improves the support to e.g. spot unawaited futures.

rcloran, 2018-02-07:
Happy to do this when HA drops 3.4 support.

jonatanolofsson, 2018-02-07:
It was dropped in October last year: https://home-assistant.io/blog/2017/10/06/deprecating-python-3.4-support/

rcloran, 2018-02-07:
Nope. That was an announcement saying it will be dropped. https://github.com/home-assistant/home-assistant/blob/dev/tox.ini#L2

Adminiuga, 2018-03-05:
I think it is reasonable to expect it will be dropped really soon, once [PR #12921 -- upgrade to aiohttp3](https://github.com/home-assistant/home-assistant/pull/12921) goes through.

balloob, 2018-03-05:
The next release, 0.65, will bump the minimum version to Python 3.5.3

AndreasBomholtz, 2018-06-28:
It is fixed in PR #31.

closed 2018-06-28 by AndreasBomholtz