zigpy/zigpy #7: Standalone application

jonatanolofsson, 2018-02-07:
For testing purposes, it is nice to have zigpy running in a standalone mode, which can handle joins, database updates etc without dying (as some devices require some time to complete setup). Always running zigpy through home assistant is no alternative as 1) The home assistant startup time is prohibitive 2) Some people (including me) uses zigpy/bellows for other things than HA, so there should be a strict separation between the two (i.e. HA must not be a requirement for zigpy)

rcloran, 2018-02-07:
This sounds useful. I suggest you develop it in a separate project, and once it’s ready, we can see whether it makes sense to fold in.

AndreasTee, 2019-08-26:
Did anyone start a project to interface zigpy standalone like @jonatanolofsson suggested?

mickeprag, 2019-11-28:
I second this. But I also notice the project is renamed zigby-homeassistant on PyPI. Why is this? I will also use this outside HomeAssistant and would hate to see a requirement for HomeAssistant for this library.

Gamester17, 2020-09-14:
Have you guys follows these very much related discussions with @pipiche38 in regards to getting started with zigpy development?

These discussions all started with https://github.com/zigpy/zigpy/issues/452 and then have snowballed from there:

https://github.com/zigpy/zigpy/pull/470

https://github.com/zigpy/zigpy/pull/477

https://github.com/zigpy/zigpy/issues/471

https://github.com/zigpy/zigpy/issues/469

PS: @pipiche38 is asking questions since he started developing Domoticz-Zigpy https://github.com/pipiche38/Domoticz-Zigpy/

closed 2021-09-07 by github-actions[bot]