zigpy/zigpy #9: Make zigpy forgiving when faced with unknown clusters/endpoints.

jonatanolofsson, 2018-02-07:
Some devices - notably Xiaomi - will not report the available cluster in accordance with the zigbee specifications (in fact, they rarely answer to requests at all). They do however send data on those clusters anyway, which causes bellows/zigpy to fail. Instead of throwing an error, a new cluster could be initialized upon discovery (it is more likely that it's not properly initialized than a properly invalid cluster id makes it through transport and hash checks). The same is true for uninitialized endpoints.

Gamester17, 2020-09-14:
@jonatanolofsson Has this issue been resolved with latest zigpy, its radio libraries, and/or latest zha-device-handlers/zha-quirks?

jonatanolofsson, 2020-09-14:
@Gamester17 Sorry, I abandoned this project long ago due to reasons, I cannot help you with this

closed 2020-09-14 by Adminiuga