zigpy/zigpy #17: Ikea tradfri discovery throws exception

andrasbiro, 2018-02-11:
I'm trying to join an ikea tradfri bulb to home-assistant (v0.63.0), and I traced back the error to this exception:
```
bellows -d/dev/ttyUSB0 permit -D /config/zigbee.db
Joins are permitted for the next 30s...
Device 0x8bb4 (90:fd:9f:ff:fe:03:e4:da) joined the network
[0x8bb4] Discovering endpoints
Device 0x796f (90:fd:9f:ff:fe:03:e4:da) joined the network
[0x796f] Discovering endpoints
[0x796f] Discovered endpoints: [1]
[0x796f:1] Discovering endpoint information
[0x796f:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=544 device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 768, 2821, 4096] output_clusters=[5, 25, 32, 4096]>
Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 35764, []])
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 54, in _initialize
    raise Exception("Endpoint request failed: %s", epr)
Exception: ('Endpoint request failed: %s', [129, 35764, []])
Done
```

The device itself:
```
bellows -d/dev/ttyUSB0 devices -D /config/zigbee.db
Device:
  NWK: 0x796f
  IEEE: (address removed)
  Endpoints:
    1: profile=0xc05e, device_type=DeviceType.COLOR_TEMPERATURE_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Level control (8)
        Color Control (768)
        Diagnostic (2821)
        LightLink (4096)
      Output Clusters:
        Scenes (5)
        Ota (25)
        Poll Control (32)
        LightLink (4096)
```
The coordinator is a Telegesis EZSP dongle, and I don't yet have a tradfri hub, so the bulb might have old firmware.

rcloran, 2018-02-12:
The device joins twice (some devices seem to do that ...). The exception is for the first join, but the initialization succeeds for the device with its second NWK.

  Failed ZDO request during device initialization: ('Endpoint request failed: %s', [129, 35764, []])

35764 = 0x8bb4.

Can you provide hass debug logs? Also, since it's in the database (otherwise it wouldn't show in `bellows devices`, hass should try to set it up on start. What's not working?

andrasbiro, 2018-02-12:
On hass UI, nothing happens
Log during join (I think I've reset the bulb)
[join.log](https://github.com/zigpy/zigpy/files/1717628/join.log)
And from that point, each time hass boots:
[restart.log](https://github.com/zigpy/zigpy/files/1717629/restart.log)

rcloran, 2018-02-18:
OK -- I think this is because Tradfri devices need messages sent to them with profile ID 0x0104 (ZHA), and not the profile ID they report (0xc05e, ZLL).

wixoff, 2018-02-20:
Is this the same issue as my submission [#12213](https://github.com/home-assistant/home-assistant/issues/12213) in the main home-assistant repo?

Sounds similar, but my bulb doesn't report DeviceType.COLOR_TEMPERATURE_LIGHT, only the hex number assigned to that.  And then it acts as a switch, not a dimmable/color temperature bulb.

I'm on 0.63.3 now, so maybe i'll try rerunning bellows to see if I get a different output.

rcloran, 2018-02-20:
It’s related, but different.... and I probably different firmware versions on the bulbs. 

@andrasbiro’s bulb reports the ZLL profile ID, but expects messages to be sent to it with the ZHA profile ID. 

Your bulb reports the ZHA profile ID, but a device type that is valid for ZLL and not ZHA. On your bulb, hass makes a guess about the device based on what clusters are there, since it doesn’t match a known (profile, device type), but it only knows how to look for individual clusters.

rklomp, 2018-03-02:
See also my logs in https://github.com/zigpy/bellows/issues/107

perosb, 2018-03-03:
If I manually update device db:
`update endpoints set profile_id = 0x0104, device_type = 0x0102;`

I can control the bulb from hass.

andrasbiro, 2018-03-04:
@perosb's workaround worked for me as well. Thanks!

chenchen119, 2018-03-09:
@perosb wondering if it is possible to do the device db update with hassio?

ryanwinter, 2018-09-11:
I believe this is resolved, I didn't have an issue during my last pairing.

andrasbiro, 2018-09-11:
I cannot confirm it, I use a tradfri base now instead of ZHA. If you guys think it's fixed, feel free to close it.

jrbenito, 2020-02-07:
As a side note, I do see Ikea bulbs showing twice in the pair screen and, very seldom, an exception. But anyway pairing finishes and I can control the bulb.

closed 2020-05-10 by Adminiuga