zigpy/zigpy #29: Lutron Connected Bulb Remote support

jonathanweinberg, 2018-03-11:
Please advise if this is the incorrect place to place this issue and I'll resubmit elsewhere.

I previously posted this here: [HASS ~ ZHA ZigBee Tested Devices](https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/290?u=jonathanweinberg)

Wanted to share some info I found on the deCONZ REST GitHub regarding the use of Lutron Connected Bulb Remotes: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ca381006cdd85987d4f8cf43b34e0a489429ab17

``` c
+static const Sensor::ButtonMap lutronLZL4BWHLSwitchMap[] = {
+//    mode                          ep    cluster cmd   param button                                       name
+//  vendor specific
+    // top button
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x04, 0xfe,  S_BUTTON_1 + S_BUTTON_ACTION_SHORT_RELEASED, "on" },
+    // second button
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x06, 0x00,  S_BUTTON_2 + S_BUTTON_ACTION_HOLD,           "dimm up" },
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x03, 0x00,  S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED,  "dimm up release" },
+    // third button
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x02, 0x01,  S_BUTTON_3 + S_BUTTON_ACTION_HOLD,           "dimm down" },
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x03, 0x01,  S_BUTTON_3 + S_BUTTON_ACTION_LONG_RELEASED,  "dimm down release" },
+    // bottom button
+    { Sensor::ModeDimmer,           0x01, 0x0008, 0x04, 0x00,  S_BUTTON_4 + S_BUTTON_ACTION_SHORT_RELEASED,  "off" },
+
+    // end
+    { Sensor::ModeNone,             0x00, 0x0000, 0x00, 0,    0,                                           0 }
+};
```

It was referenced here: [deCONZ REST Plugin Issue #232](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/232)

@manup there mentioned ` "For the events we'll create a binding from 0x0008 level control cluster to a newly created device group, so the remote will also work when the gateway is offline." `

Not sure if this would help, as I've seen ya'll are making a ton of progress in [Home Assistant Pull Request #12528](https://github.com/home-assistant/home-assistant/pull/12528/commits/66efb2f74a18f3dcc9e73ac9fbca59fa684a6216)

iantrich, 2018-04-25:
Would love to ditch my Wink hub and this would be the last thing needed to do so. +1

cgarwood, 2018-05-29:
I'm in the same boat as @iantrich - wanting to ditch my Wink but needing support for these remotes to do so. I don't know enough about the code to write support myself, but I'm more than happy to try dev builds, run special commands, and send over log files if it would help get support added.

jonathanweinberg, 2018-05-29:
I moved over to using deCONZ (I have a RaspBee), and the Lutron remotes function there.

Jonathan Weinberg

iantrich, 2018-05-29:
Might have to pick up a RaspBee. Thanks.

tylerleppek, 2018-08-09:
Hey Jonathan, what were the steps that you used to successfully pair the remote? While I can see the remote in the deconz interface It doesn't seem to function and effectively send any commands.

josephyancey, 2018-08-10:
If you make sure that you are on the latest version of deCONZ you should
just be able to pair the remote as a new sensor. In the web interface
nothing seems to happen but if you check using the web api, the device is
there and button presses get pushed out on the websocket connection. HA
sees these button presses as events.

jonathanweinberg, 2018-08-15:
@tylerleppek sorry about that, was locked out of my email for far too long. What @josephyancey mentioned is dead on. If you'd like, I can dig up my HomeAssistant / NodeRED bits and pieces for you.

iantrich, 2018-08-15:
@jonathanweinberg Please and thank you!

tylerleppek, 2018-08-16:
I’d be really interested. I’m using a Conbee with Jowihue on a homeseer hub and for whatever reason the top button on my remote doesn’t register/ send an event.

kevank, 2019-04-06:
Is there any data I ca provide that will help with getting these supported?

bballwiz5, 2019-06-07:
Duplicate of dmulcahey/zha-device-handlers#32

The implementation for HA is being tracked in the quirks repository.

br0nd, 2019-07-12:
I recommend we close this, since this is now supported by quirks

Gamester17, 2020-09-14:
Has this issue been resolved with latest Home Assistant, zigpy, its radio libraries, and/or latest zha-device-handlers/zha-quirks?

br0nd, 2020-10-01:
> Has this issue been resolved with latest Home Assistant, zigpy, its radio libraries, and/or latest zha-device-handlers/zha-quirks?

Yes, this works out of the box in Home-assistant now

closed 2021-09-07 by github-actions[bot]