zigpy/zigpy #53: attribute cache expiration

Adminiuga, 2018-05-23:
We should have some sort of cache expiration for attributes. Currently (if I understand the code correctly) the attribute cache is updated only when:
- an attribute report is received
- an "uncached" Read_Attribute command is issued

so if a zigbee device never sends "attribute reports" and application is only using "cached" attribute_reads, then the attribute cache goes stale.
For example, zha.light component in Home Assistant:
1. I started `hass` (0.70.dev.0)
2. changed the color and brightness of the bulb 
3. stopped `hass`
4. started and stopped hass two times, nothing was changed/controled. Here what async_update() was getting when light was initialized and "updated_before_add":
[image]
[image]
5. after that I stopped has and read the same attributes with `bellows` which makes an "uncached" reads
```
(venv-dev) kex@deb:~/kex/src/ac-hass$ bellows zcl 7c:b0:3e:aa:00:a4:55:95 3 6 read_attribute 0
0=Bool.false
(venv-dev) kex@deb:~/kex/src/ac-hass$ bellows zcl 7c:b0:3e:aa:00:a4:55:95 3 8 read_attribute 0
0=119
(venv-dev) kex@deb:~/kex/src/ac-hass$ bellows zcl 7c:b0:3e:aa:00:a4:55:95 3 768 read_attribute 3
3=9895
(venv-dev) kex@deb:~/kex/src/ac-hass$ bellows zcl 7c:b0:3e:aa:00:a4:55:95 3 768 read_attribute 4
4=22478
```
6. after reading the same attribute with `bellows` I started `hass` again and this time it read "correct" cached attributes:
[image]

for zha.light I can alleviate the issue if I don't load "attribute cache" from the database. But it still not addresses the issue of the cache going stale, since zha.light does not configure attribute reporting. 

So 1st question: do we really need to fill the cache from DB upon cold start?
2nd question: is attribute cache updated on getting a successful attribute write? I don't think so, but would like to confirm

dshokouhi, 2018-11-09:
I dont know if this is on topic or not but would this be the reason why my Iris Contact Sensors always default to `off` when I restart HA even if they are `on`?

Adminiuga, 2018-11-10:
Depends on the contact sensor. If there are no "server" clusters on the device side (like a dimming remote switch), then HA can't read attribute representing the status of the device, as those devices use zcl cluster commands to push changes to ha/zigpy and not attribute updates, which also means there's no cache on zigpy side. 
I don't believe current IAS implementation in HA polls for state upon restart, so those would get out of sync as well. 
There was a PR to HA to restore binary_sensor state from recorder database on restart, but it was rejected. On the phone and can't find it but I'm running it in my private branch

closed 2018-12-13 by Adminiuga