zigpy/zigpy #54: Posible to support cc2531 usb Dongle?

dzungpv, 2018-05-27:
Now there are some solution to use cc2531 like https://github.com/Koenkk/zigbee2mqtt but it is not native with hass. So there is any way to make it work with cc2531, because it is very cheap ($10 from China). I have buy some and the tool to dev it but find no way to start.

ciotlosm, 2018-05-31:
https://github.com/danielwelch/hassio-zigbee2mqtt

Gamester17, 2018-09-25:
Could maybe port cc-znp? cc-znp is an interface for a host to communicate with TI CC253X Zigbee Network Processor(ZNP) over a serial port, this is used by zigbee-shepherd among other projects.

https://github.com/zigbeer/cc-znp
https://github.com/zigbeer/zigbee-shepherd/wiki
https://github.com/zigbeer/documents/tree/master/zigbee-shepherd

Could this be achieved with the a module for Z-Stack firmware for CC253x (like CC2530 and CC2531) used by zigbee-shepherd based projects like zigbee2mqtt?

zigbee-shepherd is an open source ZigBee gateway solution with node.js. It uses TI's CC253X wireless SoC as a zigbee network processor (ZNP), and takes the ZNP approach with cc-znp to run the CC253X as a coordinator and to run zigbee-shepherd as the host.

Source code as well as pre-built firmware for  Z-Stack firmware for CC253x is available for example:

https://github.com/zigbeer/documents/tree/master/zigbee-shepherd
https://github.com/Koenkk/Z-Stack-firmware
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator
https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started

ericmiller777, 2019-01-31:
It would be nice to have it for the home assistant integrated menus.

andriej, 2019-03-20:
+1, would love to see it in. hA

Adminiuga, 2019-05-21:
Same as #85 closing this one to keep everything in one place.

closed 2019-05-21 by Adminiuga

Gamester17, 2020-01-09:
@dzungpv @ciotlosm @ericmiller777 @andriej

FYI, sanyatuning started with CC2531 support for Zigpy here https://github.com/sanyatuning/zigpy-cc

Some more information about that can also be found here https://github.com/zigpy/zigpy/issues/85