zigpy/zigpy #55: Sylvania Lightify Smart+ Dimmer Switch not Passing Button Presses

drjared88, 2018-06-01:
I apologise in advance if this is the wrong place but I was directed to come here to raise an issue with the component. My Lightify Smart+ Dimmer Switch is only showing an on/off state and brightness attribute. The component does not seem to be passing event clicks to Home Assistant.

For example if the switch state is 'on' and I press the 'on' button, nothing gets sent to Home Assistant. This causes problems with my switch getting out of sync with the device that it is controlling. Is there a way to resolve this?

Thanks!

dmulcahey, 2018-06-03:
@drjared88 the linked PR on HA fixes this.

drjared88, 2018-06-06:
@dmulcahey I tried your PR and I don't know if its working correctly. It seems to just send the switch state every 30 seconds which is causing my automatons to run every 30 seconds.

dmulcahey, 2018-06-06:
Did you change polling on your own? That’s the only way that would happen...

drjared88, 2018-06-06:
@dmulcahey No I did not.

I also noticed that with or without your code, my zha switch that if I manually try to change the state, it automatically changed back within a few seconds. It seems like the zha component constantly check the device and is sending an update at that time.

That being said I really don't know what I'm talking about. Just guessing....

ShonkyCH, 2018-06-07:
If you're having trouble with automations you might want them set so they detect a *change* from off to on or on to off rather than just detecting the state. My automations are mostly from "" to "on" or "" to "off" in HA and that works for me - I don't get repeated triggers.

drjared88, 2018-06-07:
@ShonkyCH While this would kind of work, it takes me back to my original problem where when the switch and light are out of sync, it won't work. This happens one of two ways. First is when home assistant restarts the switch defaults to on regardless of previous state. The time this happens is if the light is controlled by automation. What ends up happening is the switch is 'on' and the light is 'off' which prevents we from being able from turning the light on without turning the switch off first.

The fix that @dmulcahey has is almost perfect. It is sending the command 'on' state even if the switch is already 'on'. Unfortunately though with the switch continually sending its state ~30 seconds it renders it useless.

dmulcahey, 2018-06-07:
@drjared88 as I mentioned before that isn’t possible unless polling is on. I have 8 of these connected to 2 different systems and I am not experiencing this. Can you disable all of your automations and just watch state updates in the logs to see if it still happens?

drjared88, 2018-06-07:
@dmulcahey I think I found my problem... For some reason I was missing this
`    @property
    def should_poll(self) -> bool:
        """Let zha handle polling."""
        return False`

I must have copy and pasted over top of that but I am guessing that is my issue. Sorry to bother you. You code looks good! Now hopefully it gets rolled into the master branch.

Adminiuga, 2019-05-21:
Should be supported in HA 0.93.1 via ZHA Events. Please give it a try and reopen if it doesn't work.

closed 2019-05-21 by Adminiuga