zigpy/zigpy #61: Traceback when adding Xiaomi device

therve, 2018-06-08:
I have a few Xiaomi devices, which works great with the new support. But, on join they fail with a traceback:

2018-06-08 15:33:00 ERROR (MainThread) [bellows.ezsp] Exception running handler
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/bellows/ezsp.py", line 194, in handle_callback
    handler(*args)
  File "/usr/lib/python3.5/site-packages/bellows/zigbee/application.py", line 139, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/lib/python3.5/site-packages/bellows/zigbee/application.py", line 160, in _handle_frame
    tsn, command_id, is_reply, args = self.deserialize(device, aps_frame.sourceEndpoint, aps_frame.clusterId, message)
  File "/usr/lib/python3.5/site-packages/zigpy/application.py", line 73, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/usr/lib/python3.5/site-packages/zigpy/device.py", line 97, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1

I feel that's due to the fact that quirks only operate after device_initialized, not add_device (though I'm have only a shallow understanding of the problem).

I'd be happy to help testing.

Thanks!

hoangtuit, 2018-06-19:
Same issues

hoangtuit, 2018-06-23:
It worked for me. Thank you, @Yoda-x 
https://github.com/zigpy/bellows/issues/43

Adminiuga, 2019-05-21:
Xiaomi  are sending attribute updates prior Zigpy finishes device discovery and configure the device to send updates, hence the error. 
If you continue getting this error, then:
1. remove the device
2. repair it
3. during the discovery, keep the device awake by pressing the button ever 1.5s 
4. you may have to repeat the above a few times, as sometimes Xiaomis are PITA to pair

closed 2019-05-21 by Adminiuga

therve, 2019-05-21:
The error only happens when adding the device. But pairing does work. Could we just catch the error?

Adminiuga, 2019-05-21:
Hrm, we could take same approach as `zigpy-deconz` and ignore ZCL traffic until the device is fully initialized or alternatively maybe we should add endpoint/clusters if we receive traffic on those, similar to proposal in #9