zigpy/zigpy #70: King of Fans zigbee receiver dimmable lights

cgarwood, 2018-07-14:
The King of Fans/Hampton  Bay Zigbee fan receiver is able to dim lights. However, the light device shows up in Home Assistant as a `switch` (non-dimmable) instead of a dimmable `light` entity. Not completely sure if this is a Home Assistant issue or Zigpy issue. Figured I'd start here first.

I'm willing to help test or install dev versions of packages.

cgarwood, 2018-08-04:
Additional info, it shows the following Input/Output clusters:
```
Device:
  NWK: 0xbed2
  IEEE: 00:22:a3:00:00:20:ce:42
  Endpoints:
    1: profile=0x104, device_type=14
      Input Clusters:
        KofBasic (0)
        KofIdentify (3)
        KofGroups (4)
        KofScenes (5)
        On/Off (6)
        Level control (8)
        Fan Control (514)
      Output Clusters:
        Identify (3)
        Ota (25)
```

In Home Assistant's `zha` code it's mapping the On/Off cluster to a `switch` entity.
If you add a line to Home Assistant's code to map `LevelControl` to a `light` entity, then it creates a working dimmable light entity. (Switching the HA code to map OnOff to a `light` entity creates a light entity but doesn't allow dimming)

So it seems this is more an issue with HA's code. Though by adding the code to create a `light` entity based on the `LevelControl` cluster, you end up with both a `light` and a `switch` that both control the device. Not sure if zigpy needs some kind of quirk for that, or again, if it should fall on the Home Assistant code.

Adminiuga, 2018-08-12:
~~~Make sure you are running Zigpy version which has #58 merged.~~~
In Home Assistant override device type for KoF by adding the following to `zha:` section:
```
zha:
    device_config:
        00:22:a3:00:00:20:ce:42-1:
            type: 'light'
```
See similar discussion https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/384?u=quatuor

PS: should update HA documentation for zha configuration.

cgarwood, 2018-08-15:
I added the code to my `zha` config in HA but it's still not working correctly. I haven't tried yet with a newer build of Zigpy than what HA ships with. I'll give that a shot and see if it works.

As it stands I can control the lights exactly one time and then I can't do anything with them again until HA restarts. The log shows timeout errors (which I think is what #58 intended to fix)

cgarwood, 2018-08-16:
Working now that I'm running a dev build of `zha`. I'll create a new issue to see if an updated build can be pushed to PyPi so the HomeAssistant requirement can be updated.

closed 2018-08-16 by cgarwood

drjared88, 2018-09-10:
@WhiteNoseThunderPants I am running docker so this might be my issue however I think it should be fine. I have it upgraded and have restarted home assistant. After the restart I checked and I still have 0.7.0 and 0.2.0.  I guess I will just wait for HA 0.78.0 and see if that fixes it.