zigpy/zigpy #76: New PyPi Release

cgarwood, 2018-08-16:
Could we get a new release pushed to PyPi so Home Assistant's requirement can be updated?

AndreasBomholtz, 2018-08-17:
@rcloran Can you make a new release?

cgarwood, 2018-08-30:
Any updates? Seems like @rcloran hasn't been around much lately.

AndreasBomholtz, 2018-08-30:
No, I have not heard anything from him.
So the only thing that I can do is to push it under a new name. Other then that, I do not know what to do.

rcloran, 2018-09-03:
I'll push a new set of releases today.

rcloran, 2018-09-03:
I do want to get #78 fixed first. Seems it shouldn’t be too hard.

Gamester17, 2018-12-05:
Bump this to get a new release in PyPi with support for the newish Ikea Trådfri Plug. Thanks in advance!

Hedda, 2018-12-10:
Please bump PyPi release now that both Aqara Magic Cube and Ikea Trådfri Plug been added

Hedda, 2018-12-21:
Duplicate of #104 however the comments here from @Adminiuga might bare more weight

closed 2019-05-21 by Adminiuga