zigpy/zigpy #78: zha lights stop responding over time

wixoff, 2018-08-30:
I'm going to cross-reference my issue at zigpy/bellows#124 because it doesn't seem to be getting any traction over there, and a similar issue at home-assistant/home-assistant#15259 looks even more dead.

You can read more issue details over in the bellows issue, but the bottom line is that zha lights **work fine** immediately following a Home Assistant restart.  Then, over the course of a few hours, they disappear.  They still show up in the UI, but they are entirely inoperative.  After a **homeassistant.restart** call, the lights will all work again.

For some reason my Visonic MCT-340E door sensors **never** drop off, but all the lights eventually do, one at a time.

And here's the log message when I try to operate one of those missing lights (actually three at once):

```none
2018-07-10 07:11:09 ERROR (MainThread) [homeassistant.core] Error executing service ServiceCall <light.turn_off: entity_id=['light.osram_lightify_a19_tunable_white_0001d4ac_3', 'light.osram_lightify_a19_tunable_white_0001e6c8_3', 'light.osram_lightify_a19_rgbw_000a381d_3']>
Traceback (most recent call last):
File "/usr/src/app/homeassistant/core.py", line 1021, in _event_to_service_call
await service_handler.func(service_call)
File "/usr/src/app/homeassistant/components/light/__init__.py", line 362, in async_handle_light_service
await light.async_turn_off(**params)
File "/usr/src/app/homeassistant/components/light/zha.py", line 127, in async_turn_off
await self._endpoint.on_off.off()
File "/usr/local/lib/python3.6/site-packages/zigpy/device.py", line 89, in request
expect_reply=expect_reply,
File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 213, in request
assert sequence not in self._pending
AssertionError
```

And for the record, this is with the usual HUSBZB-1 stick, and I'm running 0.76.1 now, in Docker.  This issue has persisted for a number of months now, and the wife acceptance factor is flirting with zero at this point.

walthowd, 2018-08-30:
Sorry for the lack of traction and that you haven't gotten much of a response -- the main devs have been a bit quiet lately.

I read back through the code and I'm not sure why you are hitting the AssertionError for self._pending.

Do you have access to modify the source code of zigpy in your Docker setup? Can you try [modifying line 111 of zigpy/application.py](https://github.com/zigpy/zigpy/blob/bae5ef599dc297b9b4bf6e99fb53824cc8f0a149/zigpy/application.py#L111) and changing the 256 to 512 or 1024? That will effectively increase your sequence space  and let you run a bit longer.

You can also add these debugging lines to [line 219 of bellows/zigbee/application.py](https://github.com/zigpy/bellows/blob/c9eb0b85b480a04940176f2008c0ff2e936a4930/bellows/zigbee/application.py#L219):

```
        LOGGER.debug("ASSERTDEBUG - sequence=%s", sequence)
        for pending in self._pending:
            LOGGER.debug("ASSERTDEBUG - pending=%s", pending)
```

This will print a debugging log with your current sequence number and also enumerate the pending sequence list each time you issue a zha command. With those logs you should be able to see if your pending list is building up.

walthowd, 2018-08-30:
After running for a bit I was able to get one sequence (156) stuck in pending:

```
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=250
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=251
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=156
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=251
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=252
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=156
2018-08-30 12:07:36 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=252
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=253
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=156
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=253
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=254
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=156
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=254
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=255
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=156
2018-08-30 12:07:37 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=255
```

walthowd, 2018-08-30:
Here is another example -- An "orphaned" sequence 37 that is stuck in _pending that generated an AssertionError. The operation for that sequence 37 happened just over an hour earlier at 13:13 local time. 

```
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=34
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=120
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=37
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=123
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=34
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=35
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=120
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=37
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=123
2018-08-30 14:27:21 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=35
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=36
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast
2018-08-30 14:27:30 DEBUG (MainThread) [homeassistant.components.api] STREAM 4548111136 WRITING data: {"event_type": 
"call_service", "data": {"domain": "light", "service": "turn_on", "service_data": {"entity_id": ["light.philips_lwb01
4_0344ad6f_11", "light.sengled_e11g13_030b0923_1", "light.sengled_e11g13_030b2f5a_1"]}}, "origin": "LOCAL", "time_fir
ed": "2018-08-30T19:27:30.920485+00:00", "context": {"id": "4ff97fdea29e4e2586e085eac6de6c81", "user_id": null}}
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Sending: b'7075219c542d55b6589c4a24ab1593499c6a03adecea6374f8c6
7d3af97e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Data frame: b'0075a19c54644f027e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Data frame: b'1075b1ed542e14ba59954b65ab5592e263f620eb12316283d
9cd6789b63d7e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Data frame: b'2075b197542d55b6589c4a24ab1593499c0003abed241f7e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received
2018-08-30 14:27:30 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall light.turn_on (c:4ff
97fdea29e4e2586e085eac6de6c81): entity_id=['light.philips_lwb014_0344ad6f_11', 'light.sengled_e11g13_030b0923_1', 'li
ght.sengled_e11g13_030b2f5a_1']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/core.py", line 1117, in _event_to_service_call
    await service_handler.func(service_call)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/light/__init__.py", line 355, in async_handle
_light_on_service
    await light.async_turn_on(**pars)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/light/zha.py", line 108, in async_turn_on
    duration
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/zigpy/device.py", line 89, in request
    expect_reply=expect_reply,
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/zigbee/application.py", line 213, in request
    assert sequence not in self._pending
AssertionError
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=120
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=37
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=123
2018-08-30 14:27:30 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=36
2018-08-30 14:28:14 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - sequence=38
2018-08-30 14:28:14 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=120
2018-08-30 14:28:14 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=37
2018-08-30 14:28:14 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=123
2018-08-30 14:28:14 DEBUG (MainThread) [bellows.zigbee.application] ASSERTDEBUG - pending=38
```

And here is the log from 13:13 when the original sequence 37 got "stuck" in pending:

```
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.uart] Data frame: b'0075b1f100b86727169f4a4aa7556df49c6b2d7e'
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.ezsp] Application frame 89 (incomingRouteRecordHandler) received
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.uart] Data frame: b'1075b1ed542a15b359944a65ab55929b61f373391231625dfdc66389fe23ac7e'
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2018-08-30 13:13:05 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received
[... cut ...]
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/uart.py", line 97, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/ezsp.py", line 169, in frame_received
    assert expected_id == frame_id
AssertionError
```

It seems that over time these orphaned sequence IDs will keep growing until they use all the available 256 IDs and then all commands will fail, just as @wixoff is seeing.

adantoft, 2018-08-31:
I'm in the same boat as @wixoff here. I needed to downgrade to an older version of home assistant in order for my lights to work again. In HA 0.68 bellows was upgraded to version 0.5.2 and this is when the problems started.

Looks like these orphaned sequence IDs are causing the leak - that would make sense since things appear to work well initially after restart then fail after a span of time.

Thanks all for the support on this. My wife acceptance factor is being challenged as well :-0

walthowd, 2018-08-31:
@Yoda-x @AndreasBomholtz Any ideas what is going on here guys?

Yoda-x, 2018-08-31:
The request function in bellows.zigpy.application wait for an answer from a device, if there is not answer it times out after 10 sec and throws an exception:   

`
            v = await asyncio.wait_for(reply_fut, timeout)`
the problem, it does not cleanup the pending table and the sequence number stays for ever.
This would need something like

```
try:    
     v = await asyncio.wait_for(reply_fut, timeout)
except:
     self._pending.pop(sequence)
     raise
```
may fix it.

walthowd, 2018-08-31:
@Yoda-x Thanks, that would all make sense!

It would explain why only lights for @wixoff were being affected and not his sensors. It would also explain why the older versions of bellows that @adantoft is using before the switch to asyncio didn't have the same issue. 

Finally, it would also mean the problem would be exacerbated by a weak Zigbee network -- for everytime out you receive, you'd consume one of the 256 available sequence IDs. I know in my personal setup, I have around 30 bulbs and would usually hit max AssertionErrors after two to three days. After adding five powered Zigbee switches around the house as repeaters I can go for several weeks without any need to restart Home Assistant. 

I'm testing the exception catch code now with a torture test of toggling my most remote bulbs every 5 seconds.

walthowd, 2018-08-31:
@Yoda-x Still getting a few sequence numbers sticking in pending. For each one that is sticking this error is logged:

```
2018-08-31 14:17:38 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback SerialTransport._
read_ready()
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.4_3/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/events.py", 
line 145, in _run
    self._callback(*self._args)
  File "/Users/hass/Library/Python/3.6/lib/python/site-packages/serial_asyncio/__init__.py", line 106, in _read_read
y
    self._protocol.data_received(data)
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/uart.py", line 64, in data_received
    self.frame_received(frame)
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/uart.py", line 76, in frame_received
    self.data_frame_received(data)
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/uart.py", line 97, in data_frame_received
    self._application.frame_received(self._randomize(data[1:-3]))
  File "/Users/hass/.homeassistant/deps/lib/python/site-packages/bellows/ezsp.py", line 169, in frame_received
    assert expected_id == frame_id
AssertionError
```

Is there another leak or failed cleanup in ezsp.py?

xerootg, 2018-08-31:
@walthowd if you share your code I can spend some time trying to find the leak. My home assistant install is getting rebooted nightly over what looks like the same issue.

walthowd, 2018-08-31:
@Yoda-X Here is my full bellows log:

https://www.dropbox.com/s/l8x5vknaz3rwrp5/home-assistant.log?dl=0

The first "stuck" entry is sequence 240 which starts at line number 5713.

There are no other tracebacks from application.py until several minutes later at line number 10655.

walthowd, 2018-08-31:
@xerootg Here is the bellows/zigbee/application.py I'm running. It has the additional debug lines and the exception catch:

https://hastebin.com/opufuyenaw.py

rcloran, 2018-09-03:
Close in favour of https://github.com/zigpy/bellows/issues/124

closed 2018-09-03 by rcloran

walthowd, 2018-09-03:
Tested out some exception catching this afternoon, still seeing a few abandoned/orphaned sequence IDs in pending, not sure why. Here is the modifed request function of zigbee/application.py I'm running:

```
        aps_frame.groupId = t.uint16_t(0)
        aps_frame.sequence = t.uint8_t(sequence)

        try:
            v = await asyncio.wait_for(self._ezsp.sendUnicast(self.direct, nwk, aps_frame, sequence, data), timeout)
        except asyncio.TimeoutError:
            LOGGER.debug("sendunicast timeout %s:%s:%s", nwk, dst_ep, cluster)
            self._pending.pop(sequence)
            send_fut.cancel()
            if expect_reply:
                reply_fut.cancel()
            raise DeliveryError("Message send failure timeout")

        # Wait for messageSentHandler message
        v = await send_fut
        if expect_reply:
            # Wait for reply
            try:
                v = await asyncio.wait_for(reply_fut, timeout)
            except:
                self._pending.pop(sequence)
                raise DeliveryError("Message reply timeout %s" % (sequence, ))
        return v

```

Did another tortue test of toggling remote bulbs and many other bulbs, and had a few sequence IDs get stuck again. The first in the below log is sequence 152 which starts on line 56 of the home assistant log:

https://www.dropbox.com/s/fwjq9rafsu7wl4z/stuck.txt?dl=0

By the end of the log there are two other stuck sequences in self._pending (12 and 228) but those are after the sequence IDs have wrapped around at least once.