zigpy/zigpy #81: Quirks for smartthings sensors to add battery_percentage_remaining attribute

ryanwinter, 2018-09-11:
Working HA, I was looking to add a generic battery layer that uses the battery_percentage_remaining attribute. Unfortunately, many sensors, like the last generation smarthings, don't report this attribute.

I tried to add a quirk to create this attribute based on the existing battery_voltage, but this failed dismally. Was wondering if anyone had the expertise to do this.

This was my attempt based on an example I saw elsewhere, but it is never triggered.
```
class SmartthingsPowerCluster(CustomCluster, PowerConfiguration):
    """Provides battery % remaining based on voltage"""
    minVolts = 15
    maxVolts = 28
    values = {
        28: 100,
        27: 100,
        26: 100,
        25: 90,
        24: 90,
        23: 70,
        22: 70,
        21: 50,
        20: 50,
        19: 30,
        18: 30,
        17: 15,
        16: 1,
        15: 0
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)

        attr_name = self.attributes.get(attrid, [attrid])[0]
        _LOGGER.debug("smartthings update attribute %s", attr_name)
        if attr_name == 'battery_voltage':
            _LOGGER.debug("Found battery voltage %s %s", value)

            value = min(value, self.maxVolts)
            value = max(value, self.minVolts)

            bpr = self.values.get(value, 'unknown')
            bpr_id = self._attridx.get('battery_percentage_remaining')

            super()._update_attribute(bpr_id, bpr)
```

ryanwinter, 2018-10-18:
Is anyone able to provide some guidance here? Happy to do the leg work.

roblandry, 2018-11-28:
It's unfortunate that you couldn't get any assistance and instead closed this. This is one of the annoyances with zha (no battery reporting) and I was hoping you would be able to get it going. If I had any understanding of the inner workings of zha I'd be glad to help. Maybe with any luck, zha will be forked into hass like zwave where more eyes are on it.

ryanwinter, 2018-11-28:
oops, i didnt mean to close this.

Actually the work is already done and is split into two parts. 

1. Addition of the battery reporting based on the "battery_percentage_remaining" attribute, as outlined in [this post](https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/526?u=winter)

2. Adding the quirk for older Smartthings sensors to convert the battery voltage to the battery percentage attribute. @dmulcahey also has some sample code for this for a different sensor type, however I was able to implement a quirk with his help for the smartthings sensor and have it working locally.

Once the HA change is commited, and once the zigpy quirks manufacturing change is in place, I can submit the smartthings quirk.

Adminiuga, 2019-05-21:
Fixed via https://github.com/dmulcahey/zha-device-handlers

closed 2019-05-21 by Adminiuga