zigpy/zigpy #85: Port cc-znp to communicate with TI CC253X Zigbee Network Processor(ZNP) over a serial port

Gamester17, 2018-09-25:
Would it be possible to port cc-znp as a radio module for ZigPy to communicate with TI CC253X based dongles? 

cc-znp is an interface for a host to communicate with TI CC253X Zigbee Network Processor(ZNP) over a serial port, this is used by zigbee-shepherd among other projects with the use of a custom firmware.

https://github.com/zigbeer/cc-znp
https://github.com/zigbeer/cc-znp/wiki

https://github.com/zigbeer/zigbee-shepherd/wiki
https://github.com/zigbeer/documents/tree/master/zigbee-shepherd

Could this be achieved with the a module for Z-Stack firmware for CC253x (like CC2530 and CC2531) used by zigbee-shepherd based projects like zigbee2mqtt?

zigbee-shepherd is an open source ZigBee gateway solution with node.js. It uses TI's CC253X wireless SoC as a zigbee network processor (ZNP), and takes the ZNP approach with cc-znp to run the CC253X as a coordinator and to run zigbee-shepherd as the host.

PS: zigbee2mqtt also have source code as well as pre-built firmware for a custom Z-Stack firmware :

https://github.com/Koenkk/Z-Stack-firmware
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator
https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started

Gamester17, 2019-01-09:
Also see the open-zigbee fork updates https://github.com/open-zigbee/zigbee-bridge-znp

leobel96, 2019-01-24:
+1 for TI CC253X support

highground88, 2019-01-25:
Also keen on testing with the Texas Instruments CC2531 in particular. Cheap device and very flexible.

andriej, 2019-03-20:
+1, would be superb

jedi7, 2019-06-16:
These ticket "closes" are little confusing. Because all are closed with reference to each other.
Is it purpose or bug?
(I hope it is not purpose, because the support for cc2531 will be great)

Hedda, 2019-10-03:
Has anyone made any progress with Zigpy for CC253x USB sticks with custom Z-Stack firmware?

https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

https://github.com/Koenkk/Z-Stack-firmware

Gamester17, 2019-12-12:
FYI, it looks like the Z-Stack firmware from @Koenkk now also has experimental support CC2652R / CC26X2R1 based USB sticks and he has added Zigbee 3.0 support in Z-Stack 3.0 firmware as well

https://github.com/Koenkk/Z-Stack-firmware

CC2652R by Texas Instruments is a new high-performance Zigbee chip which is much faster and contains much more memory than the CC2530 and CC2531 chips

Koenkk posted some more information about CC2652R here https://github.com/Koenkk/zigbee2mqtt/issues/1429

**_Notes

The device can be bought here: http://www.ti.com/tool/launchxl-cc26x2r1
No additional flashing hardware is needed, it has an on-board programmer (no CC debugger needed).
The board has +- the size of a Raspberry Pi
Zigbee 3.0!
Flash using Flash Programmer v2
Firmware can be found here in the bin folder: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.x.0
It's not possible to migrate your Zigbee 1.2 network (requires repairing)
The network restore trick that we used in Zigbee 1.2 doesn't work anymore (due to better security policies). A backup of the CC2652R is taken at every shutdown of zigbee2mqtt, file can be found under data/coordinator_backup.json. This backup is automatically restored when starting with a new flashed CC2652R._**

Gamester17, 2019-12-19:
FYI, Tasmota is another popular project that makes use of the Z-Stack-firmware from Koenkk

https://github.com/arendst/Tasmota/wiki/Zigbee

Z2T (Zigbee to Tasmota) concept allows you to make your own Zigbee to WiFi bridge by combining Z-Stack-firmware on CC2530 and Tasmota firmware on ESP8266 / ESP82xx hardware.

sanyatuning, 2019-12-25:
FYI 
I started working on this :)

Hedda, 2020-01-03:
@sanyatuning Awesome that you started on this! I for one will now be ordering a CC2531 USB sniffer and keep an eye on your repository at https://github.com/sanyatuningI/zigpy-cc

Gamester17, 2020-01-07:
@sanyatuning Very cool! Which version of Z-Stack firmware are you testing? It's from Koenkk or?

https://github.com/Koenkk/Z-Stack-firmware

Z-Stack 3 (3.x) supports the full Zigbee 3.0 specification according to these discussions here:

https://github.com/Koenkk/zigbee2mqtt/issues/1429
https://github.com/Koenkk/zigbee2mqtt/issues/1568

FYI, this Z-Stack 3.0 and later firmware has experimental support in Zigbee2mqtt dev branch.

Gamester17, 2020-01-13:
FYI; ioBroker is another JaveScript based project which now native support for CC2531/CC2530 based Zigbee USB sticks when they are flashed with the same custom Z-Stack firmware from Koenkk

https://github.com/ioBroker/ioBroker.zigbee

https://github.com/ioBroker/ioBroker.zigbee/blob/master/README.md

sanyatuning, 2020-01-16:
@Gamester17 my port is based on zigbee herdsman. I'm testing with z-stack 1, but it can be compatible with v3.
I would like to continue developing it this week.

dzungpv, 2020-01-17:
@sanyatuning Great effort! I use zigbee2mqtt for year, but native component for HA always the best.

Gamester17, 2020-02-02:
@sanyatuning see that @puddly now submitted a pull request for CC26X2R1 support to your zigpy-cc

https://github.com/sanyatuning/zigpy-cc/pull/8

Gamester17, 2020-02-13:
FYI, looks like @sanyatuning has now submitted initial (“Beta”) support for Home Assistant integration:

https://github.com/home-assistant/home-assistant/pull/31621

Hedda, 2020-03-25:
FYI, it looks like Adminiuga (and puddly via a fork) started on zigpy-znp as an alternative to zigpy-cc
- https://github.com/zha-ng/zigpy-znp
It has not been updated by Adminiuga in about 4-months but puddly continues in his fork here:
- https://github.com/puddly/zigpy-znp
Neither of those projects has yet been announced so are still both very early work in progress.

puddly, 2020-03-25:
I'm pretty close to getting zigpy-znp running. I'm just waiting for a second TI radio so that my entire house is not unusable while I fix bugs.

Adminiuga, 2020-05-10:
There's https://github.com/zigpy/zigpy-cc and work being done in https://github.com/zha-ng/zigpy-znp

Can continue discussions in those repos.

closed 2020-05-10 by Adminiuga