zigpy/zigpy #87: Additional inputs for SmartThings multipurpose sensors

ratatine, 2018-10-03:
I've been digging into the code a bit to see if there might be a mechanism to map the inputs on the SmartThings multipurpose sensors. I saw that 0x402 (1026) maps the temperature sensor and that works in HA.

Other inputs include:
- 0xFC02 (64514) tilt and acceleration. attr 0x0010->acceleration, 0x0012->x-axis 0x0013->y-axis 0x0014->z-axis
- 0x500 (1280) zone status, attr 2, 32=closed and 33=open

If anyone has any suggestions I'm willing to help do some leg work. I do see that 0x402 (1026) is mapped in zha.py and that's the temperature sensor. Thanks!

Adminiuga, 2018-10-03:
There's #82 and #83, but then in Home assistant, you would need to update the code and map `zigpy.quirks.smartthings.SmartThingsAccelCluster` to `'sensor'` component and write a class representing the motion sensor.

Adminiuga, 2018-10-03:
You also need to provide correct `manufacturer_id` for `configure_reporting()` cluster method, which is different for different models, check the latest smart things device handler.

ratatine, 2018-10-03:
That's me not having a full grasp on github - I failed to see those PRs. I'm happy to do my part with what I have and I have three of these (all different generations) right now. Thanks!

Adminiuga, 2020-05-10:
closing on no activity

closed 2020-05-10 by Adminiuga