zigpy/zigpy #89: Xiaomi WXKG03LM and quirks

Adminiuga, 2018-10-17:
What Xiaomi device/model the [class TemperatureHumiditySensor(CustomDevice):](https://github.com/zigpy/zigpy/blob/6f1d70256ad1bd0b66fb37d53bdb96d261172d4d/zigpy/quirks/xiaomi/__init__.py#L4) quirks is for? 
I've got [WXKG03LM](https://www.aliexpress.com/item/Xiaomi-Smart-home-Aqara-Smart-Light-Control-Wireless-Switch-Key-and-Wall-Switch-ZigBee-Remote-Control/32813503911.html) the single button, which gets detected with 3 endpoints
```
<SimpleDescriptor endpoint=1 profile=260 device_type=24321 device_version=1 input_clusters=[0, 3, 25, 65535, 18] output_clusters=[0, 4, 3, 5, 25, 65535, 18]>
<SimpleDescriptor endpoint=2 profile=260 device_type=24322 device_version=1 input_clusters=[3, 18] output_clusters=[4, 3, 5, 18]>
<SimpleDescriptor endpoint=3 profile=260 device_type=24323 device_version=1 input_clusters=[3, 12] output_clusters=[4, 3, 5, 12]>
```
and therefore the above quirks takes over it, but it is definitely not a Temperature/humidity device or at least I think it is not. If it has the same clusters as the quirks Temperature/humidity sensor, then maybe we should take model into consideration when matching for a quirk?

unconfigured, 2018-10-19:
I have both, the switches and temperature sensors, they both report the same clusters/profile/devicetype :( So quirks will not work here.
See https://github.com/home-assistant/home-assistant/issues/14609#issuecomment-391815146
Sadly I've not seen any solution to this.

rkoens, 2018-11-27:
I ran into the same issue and coded a way to allow for quirk matching based on manufacturer and model. See pull request #97 for further news.

Adminiuga, 2018-11-27:
Yep, so that one. I'll give it a try.

closed 2019-05-21 by Adminiuga