zigpy/zigpy #96: Orivibo T10D1 Smart Dimmer causing UTF related errors

tyler-vincent, 2018-11-19:
I've picked up one of these dimmer switches for an excellent price and tried it out with ZHA on Hass.io. It pairs correctly but after that Zigpy throws an error like this:

```
Error during setup of component zha
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/setup.py", line 145, in _async_setup_component
    hass, processed_config)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/zha/__init__.py", line 106, in async_setup
    APPLICATION_CONTROLLER = ControllerApplication(radio, database)
  File "/usr/local/lib/python3.6/site-packages/bellows/zigbee/application.py", line 21, in __init__
    super().__init__(database_file=database_file)
  File "/usr/local/lib/python3.6/site-packages/zigpy/application.py", line 25, in __init__
    self._dblistener.load()
  File "/usr/local/lib/python3.6/site-packages/zigpy/appdb.py", line 223, in load
    ep.manufacturer = value.decode('ascii').strip()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe6 in position 0: ordinal not in range(128)
```

I took a look in `zigbee.db` and saw that it the manufacturer name was in Chinese. I corrected this to just `Orvibo` manually and it seems to have solved the problem.

A secondary problem is that the switch only shows up as a switch object with no dimming function. Not sure how to solve that one.

zalke, 2019-02-19:
Just my 2 cents, regarding the secondary issue,
is there a possibility that since the discovery failed because of the first issue, it didn't had a chance to report the dimming function?

dmulcahey, 2019-02-19:
no, switches don’t support dimming in zha, we’ll have to override the profile / component for the IEEE’s in configuration.yaml for you to get dimming. @adminiuga do you know the syntax for that off hand?

Adminiuga, 2019-02-19:
`configuration.yaml` snippet for overriding the component type:
```
zha:
  baudrate: 57600
  usb_path: /dev/ttyUSB1
  database_path: /home/pi/.homeassistant/zigbee.db
  device_config:
    7c:b0:3e:aa:de:ad:be:ef-3:
       type: 'light'
```

zalke, 2019-02-19:
Is there anyway to do it using the "zha integration?"

dmulcahey, 2019-02-19:
The integration only controls a few settings. Things like this go in configuration.yaml per the HA standards

Adminiuga, 2019-05-21:
Fixed in #147

closed 2019-05-21 by Adminiuga