zigpy/zigpy #102: [REQUEST] ZHA Device Handler to handle for unique quirks exception handling implementations in Zigpy

Hedda, 2018-12-11:
@dmulcahey (with the help of a few other developers like @damarco @Adminiuga & @Yoda-x ) have written this ZHA Device Handler as an implementation to handle "quirks" for Zigbee devices when using Zigpy (in Home Assistant) which allow you to write modular device handlers for handling unique exceptions (a.k.a. "quirks") in Zigbee devices. 

https://github.com/dmulcahey/zha-device-handlers

@rcloran @AndreasBomholtz I was wondering it would be a good idea for the Zigpy itself to integrate such a ZHA Device Handler function of its own or similar modular implementation for exception handling of the unique quirks in some Zigbee devices?

This point of this is both to workaround bugs in devices and to deal with exceptions when device manufacturers deviate from the Zigbee standards or even if a device use functions not yet available in the Zigpy library itself, as separate handlers can then be implemented by third.parties without having to change or update the Zigpy library. SmartThings, for example, has custom handlers which perform this function of dealing with exceptions and device functions not yet implemented by Samsung. 

I mean, would it not be better if a modular ZHA Device Handler implementation existing as part of the Zigpy project (so other third-party projects can use it as well) instead of it being only part of Home Assistant?

PS: Something similar was also requested by @jonatanolofsson in #8 ("Low-level hardware specialization")

_"Specialization cannot be exclusively handled above the zigpy abstraction level as some devices require low-level adjustments. One distinguished example is the IKEA Trådfri devices which need to use the ZHA profile ID in the APS frame in ZCL communication."_

dmulcahey, 2018-12-11:
@hedda I think that what quirks offers today is perfect for Zigpy. I personally think that keeping the device-handlers separate is the right move. This offers a few benefits: 

They can evolve on their own and at their own pace

They can take advantage of the platform they were written for and they don’t have to remain generic.

I get wanting them available for other application platforms but I think that would limit what they can do. IMO quirks itself is the generic part that any application can use and the device-handlers are the application specific implementation.

Hedda, 2019-01-14:
@dmulcahey then how about moving your zha-device-handlers repository to the zigpy organization?

That is, move https://github.com/zigpy to https://github.com/dmulcahey/zha-device-handlers

That way developers would have all Zigpy related repositories under one single organization.

Please also see the somewhat related suggestion here => https://github.com/zigpy/zigpy/issues/113

dmulcahey, 2019-01-24:
The lib is geared only towards HA and won’t work for anything else so I don’t think this is a good idea but if others feel strongly I’ll consider it

Adminiuga, 2020-04-02:
Decision was made to keep `zha-device-handlers` as a separate library, moving all quirks into it. This allows `zha-device-handlers` to be developed/extended at a different pace, without waiting on zigpy releases.
Migrating `zha-device-handles` into `zigpy` organization is under progress.

closed 2020-04-02 by Adminiuga