zigpy/zigpy #109: GE Link A19 fails to pair, Failed ZDO Request

prairiesnpr, 2018-12-31:
When attempting to pair, the process times out.  I've tried with allow pair on and off and get the same result.  This is after resetting the bulb, bulb will dim and then blink three times, but isn't added.

Home assistant version:
0.85.0.dev0 

Zigpy version: 0.2.0                                                                                                                                                    

Logs:

```
> 2018-12-29 02:15:32 INFO (MainThread) [zigpy.application] Device 0x88b7 (00:13:a2:00:40:b4:15:17) joined th
> e network                                                                                                  
> 2018-12-29 02:15:32 INFO (MainThread) [zigpy.device] [0x88b7] Discovering endpoints                        
> Tries remaining: 3                                                                                         
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request seq 5                
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] Sequenced command: tx_explicit (00:13:a2:00:40:b4:1
> 5:17, 34999, 0, 0, 5, 0, 0, 32, b'\x05\xb7\x88')                                                           
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (14, 00:13:a2:00:40:b4:15:17, 3
> 4999, 0, 0, 5, 0, 0, 32, b'\x05\xb7\x88')                                                                  
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x0e\x88\xb7\x00\x00\x00'   
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status                          
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] tx_status: [14, 34999, 0, 0, 0]                    
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00@\xb4\x15\x1
> 7\x88\xb7\xe8\xe8\x00\x92\xc1\x05\x01\x01\x10.\x00\x00 '                                                   
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator              
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: opts=1                        
> Tries remaining: 2                                                                                         
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request seq 6                
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.api] Sequenced command: tx_explicit (00:13:a2:00:40:b4:1
> 5:17, 34999, 0, 0, 5, 0, 0, 32, b'\x06\xb7\x88')                                                           
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (15, 00:13:a2:00:40:b4:15:17, 3
> 4999, 0, 0, 5, 0, 0, 32, b'\x06\xb7\x88')                                                                  
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x0f\x88\xb7\x00\x00\x00'   
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status                          
> 2018-12-29 02:15:44 DEBUG (MainThread) [zigpy_xbee.api] tx_status: [15, 34999, 0, 0, 0]                    
> Tries remaining: 1                                                                                         
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request seq 7                
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.api] Sequenced command: tx_explicit (00:13:a2:00:40:b4:1
> 5:17, 34999, 0, 0, 5, 0, 0, 32, b'\x07\xb7\x88')                                                           
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (16, 00:13:a2:00:40:b4:15:17, 3
> 4999, 0, 0, 5, 0, 0, 32, b'\x07\xb7\x88')                                                                  
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x10\x88\xb7\x00\x00\x00'   
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status                          
> 2018-12-29 02:15:56 DEBUG (MainThread) [zigpy_xbee.api] tx_status: [16, 34999, 0, 0, 0]                    
> 2018-12-29 02:16:06 ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization:     
> Traceback (most recent call last):                                                                         
>   File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 51, in _initialize           
>     epr = await self.zdo.request(0x0005, self.nwk, tries=3, delay=2)                                       
>   File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/util.py", line 52, in retry                   
>     r = await func()                                                                                       
>   File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 89, in request               
>     expect_reply=expect_reply,                                                                             
>   File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_xbee/zigbee/application.py", line 81, in reque
> st                                                                                                         
>     v = await asyncio.wait_for(reply_fut, timeout)                                                         
>   File "/usr/local/lib/python3.7/asyncio/tasks.py", line 423, in wait_for                                  
>     raise futures.TimeoutError()                                                                           
> concurrent.futures._base.TimeoutError 
```

Adminiuga, 2019-01-03:
What version of zigpy-xbee are you running? 
Per trace looks like it couldn't get past endpoint discovery. ZDO request was sent out and per "TX Status" successfully.
I see a reply to "active_ep_req" ZDO request:
```
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00@\xb4\x15\x1
> 7\x88\xb7\xe8\xe8\x00\x92\xc1\x05\x01\x01\x10.\x00\x00 '                                                   
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator              
> 2018-12-29 02:15:32 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: opts=1                        
> Tries remaining: 2                                                                                         
```
but for whatever reason looks like it still continues to query for "active endpoints" after 10s (12s) which is the default timeout.

prairiesnpr, 2019-01-03:
I'm running zigpy-xbee version 0.1.1.

Adminiuga, 2019-01-03:
Could you try install zigpy-xbee from github master tree? There were some improvements, although not really applicable to your case.
After you install zigpy-xbee from master, "factory reset" the light and try to join it again, while collecting all the logs.

prairiesnpr, 2019-01-04:
Set the baudrate to 57600.

I reset the bulb and attempted to pair.  A few observations, it doesn't seem to make any difference if I trigger a ZHA.permit prior to resetting the bulb or even if I reset the bulb.  As soon as the bulb is turned on, HA attempts ZDO discovery and the logs are the same.

While I had the xbee out and connected to xctu, I went ahead and grabbed the configuration for it.  I had this running on a custom python script that controlled all of my zha devices and I didn't change the config when I moved to home assistant.  Was wondering if maybe I have an incompatible configuration, (ZDO passthrough?) 

I do have a functioning network, but the only other devices on the system are non standard, 2 Iris V1 smartplugs that follow a slightly different protocol that I'm supporting via quirks and a couple other xbee/arduino devices that are attached, but I haven't found a way to support them yet.  I'm willing to go buy a device if there is a known model that would be helpful.

Xbee Config (XB24-ZB_21A7)
```
<setting command="ID">0</setting>
<setting command="SC">FFFF</setting>
<setting command="SD">3</setting>
<setting command="ZS">2</setting>
<setting command="NJ">FF</setting>
<setting command="DH">0</setting>
<setting command="DL">FFFF</setting>
<setting command="NI">HA Coordinator</setting>
<setting command="NH">30</setting>
<setting command="BH">0</setting>
<setting command="AR">FF</setting>
<setting command="DD">30000</setting>
<setting command="NT">3C</setting>
<setting command="NO">3</setting>
<setting command="CR">3</setting>
<setting command="PL">4</setting>
<setting command="PM">1</setting>
<setting command="EE">1</setting>
<setting command="EO">1</setting>
<setting command="KY"/>
<setting command="NK"/>
<setting command="BD">6</setting>
<setting command="NB">0</setting>
<setting command="SB">0</setting>
<setting command="D7">0</setting>
<setting command="D6">0</setting>
<setting command="AP">2</setting>
<setting command="AO">3</setting>
<setting command="SP">20</setting>
<setting command="SN">1</setting>
<setting command="D0">1</setting>
<setting command="D1">0</setting>
<setting command="D2">0</setting>
<setting command="D3">0</setting>
<setting command="D4">0</setting>
<setting command="D5">1</setting>
<setting command="P0">1</setting>
<setting command="P1">0</setting>
<setting command="P2">0</setting>
<setting command="PR">1FFF</setting>
<setting command="LT">0</setting>
<setting command="RP">28</setting>
<setting command="DO">1</setting>
<setting command="IR">0</setting>
<setting command="IC">0</setting>
<setting command="V+">0</setting>

```
LOGS
```
Read local IEEE address as 00:13:a2:00:40:ac:b4:ab
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] AT command: AI ()
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] Command at (5, b'AI', b'')
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x05AI\x00\x00'
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] AT command: MY ()
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] Command at (6, b'MY', b'')
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x88\x06MY\x00\x00\x00'
2019-01-03 23:51:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2019-01-03 23:51:32 DEBUG (MainThread) [zigpy_xbee.api] AT command: NJ (60,)
[... 36 lines cut ...]
2019-01-03 23:52:02 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x0c\xff\xfd\x00$\x00'
2019-01-03 23:52:02 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2019-01-03 23:52:02 DEBUG (MainThread) [zigpy_xbee.api] tx_status: [12, 65533, 0, 36, 0]
2019-01-03 23:52:04 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request seq 4
2019-01-03 23:52:04 DEBUG (MainThread) [zigpy_xbee.api] Sequenced command: tx_explicit (7c:e5:24:00:00:06:c0:93, 39506, 0, 0, 5, 0, 0, 32, b'\x04R\x9a')
2019-01-03 23:52:04 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (13, 7c:e5:24:00:00:06:c0:93, 39506, 0, 0, 5, 0, 0, 32, b'\x04R\x9a')
2019-01-03 23:52:12 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\r\xff\xfd\x00$\x02'
2019-01-03 23:52:12 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2019-01-03 23:52:12 DEBUG (MainThread) [zigpy_xbee.api] tx_status: [13, 65533, 0, 36, 2]
2019-01-03 23:52:14 ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization: 
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 51, in _initialize
    epr = await self.zdo.request(0x0005, self.nwk, tries=3, delay=2)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/util.py", line 52, in retry
    r = await func()
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 89, in request
    expect_reply=expect_reply,
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_xbee/zigbee/application.py", line 81, in request
    v = await asyncio.wait_for(reply_fut, timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Thanks for helping to troubleshoot this.

prairiesnpr, 2019-01-04:
I'm going to close this, I took another xbee and reset the firmware to default and the swapped it for the current coordinator.  I was able to get both ge lamps to pair.  This is with the new firmware.

It's not perfect, both lamps only respond to commands about 10% of the time, but they are paired, pretty sure this was an xbee configuration issue.

Thanks

closed 2019-01-04 by prairiesnpr