zigpy/zigpy #113: Invite more developers to join "Team Zigpy" and permit repository permissions?

Hedda, 2019-01-14:
@rcloran can I suggest that you extend some invitations to a few more developers who already submitted or are actively working on patches for and forks of zigpy (and radio libraries for it) to join some kind of "Team Zigpy" which could work together to achieve common goals for this upstream zigpy project and its repositories as an organization unit with the same goals? 

https://github.com/orgs/zigpy/people

As mentioned in #104 please then give permission to either all or a trusted few to not only make commits of pull requests to the mainline tree but also tag and make releases on GitHub (and PiPy releases as well).  Long-term, do you not believe that it would be a good idea to have more people with commit and release access to this upstream repo for zigpy?

As a team, you could collaborate using a common chat and/or forum as a communication channel for sharing ideas and discussing short-term and long-term goals for zigpy and other libraries or tools for it?

Having a team collaborating would off-load some of the work from you. Like if you and a few of them are willing to work together as team then you could for example all help review and give feedback on each other’s pull request for patches and once trust is proved you could giving more people write access to the upstream bellows repository to assist in committing push request and fixing issues that other people posts?

To start with some example team-member candidates of the people who look to be actively been working on improvements and enhancements for zigpy and its radio libraries or the ZHA component for Home Assistant are; @AndreasBomholtz @Adminiuga @dmulcahey @Yoda-x @jonatanolofsson @damarco @Seluxit @h3ndrik @SteveEasley @titilambert @puddly @igorbernstein2 @hobbypunk90 @dbajric @kylehendricks ...and perhaps also @balloob from the Home Assistant project too?

Hedda, 2019-01-22:
Maintainers are also needed for releasing and maintaining all PyPI releases as a team when tag versions

- https://pypi.org/project/zigpy
- https://pypi.org/project/bellows
- https://pypi.org/project/zigpy-xbee
- https://pypi.org/project/zigpy-deconz

ryanwinter, 2019-01-24:
It might be easier to just fork the repos and continue work from there.

Hedda, 2019-01-25:
Does anyone here been in contact with @rcloran recently or know of anyone who has talked to him? 

I still think best long-term would be if @rcloran assign additional admins to this existing zigpy organization and all of its repositories, both as everything external already links here and you do not get copies of issues and all pull request discussions when you fork, this as GitHub just forks the code and its revision history, not issues and pull requests.

If @rcloran is truly and completely incommunicado then perhaps someone like @AndreasBomholtz @damarco and/or @Adminiuga who already contributed a lot of code to this existing zigpy organization could contact GitHub support about having GitHub assigning one of them as an additional admin of this existing zigpy organization. They probably have a procedure for this that does not require forking.

I'm sure that GitHub would be willing to one or more additional owners of this existing zigpy organization if other code contributing developers explain that the founding member has gone 'missing in action' from this project and that for it to go on best would be if someone else could also be assigned admin to this existing zigpy organization until if and when @rcloran comes back.

balloob, 2019-01-26:
Stop tagging me. I maintain Home Assistant and have nothing to do with this.

frog32, 2019-01-27:
@Hedda no need to reach out to any support. The easiest way to get a release out is to publish it under a different pypi name and then change homeassistant to use that new release. Once other people have the privilege to push to pypi it is easy to revert back to using the official channel.

Hedda, 2019-01-29:
Zigpy has everything to do with Home Assistant as its only used by it and gives it native Zigbee support.

Sad if the founder of Home Assistant does not care about the library that gives it native Zigbee support.

puddly, 2019-01-29:
@Hedda I use zigpy/bellows independently of Home Assistant (and I assume others do too). Are there any benefits to tightly coupling the two projects? I feel that'd just make using zigpy/bellows more difficult to use.

Adminiuga, 2019-01-29:
ATM I think active devs do have access to Zigpy repository, so as a work around the could be pushed under a slightly different name to PyPi. Homeassistant does this for some other libraries too, so I think that should be ok.

frog32, 2019-01-29:
@Adminiuga that was exactly what I was proposing. The only part here is a PyPi release nothing else is required. This should be doable by tweaking setup.py and then pushing a release. This should not have anything to do with homeassistant other than it being mostly driven by HAs need for a new release to support some features.

Hedda, 2020-03-10:
Submitted https://github.com/zigpy/zigpy/issues/318 as I understand that lack of admin permissions to zigpy org is still causing problems.

Hedda, 2020-03-10:
@Adminiuga should this issue maybe be reopened?

Neustradamus, 2020-04-01:
@Adminiuga: Can you reopen this ticket?

Hedda, 2020-04-01:
@Neustradamus please also see the related and dependent issue discussed in https://github.com/zigpy/zigpy/issues/318

@rcloran was until recently the sole owner and admin of the zigpy organization and repos, meaning that until recently no others have had permissions to add other members or repos.

Hedda, 2020-04-28:
FYI, I just read the news that GitHub basic free account now allows teams private repositories with unlimited collaborators at no cost to make team collaboration easier, see:

- https://github.blog/2020-04-14-github-is-now-free-for-teams/

GitHub is now free for teams
Nat Friedman

We’re happy to announce we’re making private repositories with unlimited collaborators available to all GitHub accounts. All of the core GitHub features are now free for everyone. 🎉

Until now, if your organization wanted to use GitHub for private development, you had to subscribe to one of our paid plans. But every developer on earth should have access to GitHub. Price shouldn’t be a barrier.

This means teams can now manage their work together in one place: CI/CD, project management, code review, packages, and more. We want everyone to be able to ship great software on the platform developers love.

Teams who need advanced features (like code owners), enterprise features (like SAML), or personalized support can upgrade to one of our paid plans.

We’re also reducing the price of our paid Team plan from $9 per user/month to $4 per user/month, effective immediately. Existing customers will have their bills automatically reduced going forward.

Learn more in the FAQ, or compare plans on our pricing page.

- https://help.github.com/github/getting-started-with-github/faq-about-changes-to-githubs-plans

- https://github.com/pricing

Neustradamus, 2020-04-28:
It was previously FREE !!

Only more free options included.

pipiche38, 2020-08-10:
I'm watching the project since a while , and this looks pretty great. However I have the feeling that the development is very HA focussed and not open to be used as a pure Python Library as per its origin. For instance I didn't find any documentation or HOW-TO to start using the library.

Adminiuga, 2020-08-10:
It is a python library, not a "standalone application" and that was the intent originally and it is still the intent today.
It is a separate library as per HA requirements, but I don't think are any HA specific features in the library itself. Probably because it is a library and not a standalone application.

Documentation could be better, just like with any open source project, but it is open source so PRs a welcome, otherwise the code is the documentation 🤷

Hedda, 2020-08-11:
>I didn't find any documentation or HOW-TO to start using the library.

@pipiche38 While you have a valid point please understand that your question is off-topic for this specific issue topic discussion so please instead create new separate issues in regards to different questions about library usage and/or missing documentation. See https://github.com/zigpy/zigpy-cc/issues/61 as an example on how-to formulate such a question in a new issue.

Again discussion about this is off-topic here, however, a tip is to as per reply in linked issue is to look at the code for zha in Home Assistant). Please do post a pull request with documentation improvements when you learn more as that is the open source way. And other tip is to look at some non-HA stand-alone application projects that used the bellows library without zigpy, like example https://github.com/daniel17903/ZigCoHTTP ZigCoHTTP and qzig https://github.com/Seluxit/qzig and hue-thief https://github.com/vanviegen/hue-thief

pipiche38, 2020-08-11:
@Hedda  if I'm succesful in using it, I'll make such HOW-To but right now struggling a lot :-(
Will have a look to the pointers

Hedda, 2020-08-11:
> @Hedda if I'm succesful in using it, I'll make such HOW-To but right now struggling a lot :-(
> Will have a look to the pointers

That is cool! Suggest that you then create a new separate issue with specific questions -> https://github.com/zigpy/zigpy/issues

pipiche38, 2020-08-11:
> > @Hedda if I'm succesful in using it, I'll make such HOW-To but right now struggling a lot :-(
> > Will have a look to the pointers
> 
> That is cool! Suggest that you then create a new separate issue with specific questions -> https://github.com/zigpy/zigpy/issues

#452

closed 2021-09-07 by github-actions[bot]