zigpy/zigpy #115: [REQUEST] Support for the ZiGate ( http://zigate.fr ) open source ZigBee USB adapter hardware

Hedda, 2019-01-21:
Please consider supporting the ZiGate open source ZigBee USB adapter hardware and its firmware in zigpy.

http://zigate.fr

ZiGate is a open source ZigBee USB adapter that was initially launched on Kickstarter by @fairecasoimeme

https://www.kickstarter.com/projects/1361563794/zigate-universal-zigbee-gateway-for-smarthome

The Zigate USB adapter communicates via a PL2303HX Serial interface module.

Here is the documents that layout the communication protocol used by the Zigate:

https://github.com/fairecasoimeme/ZiGate/tree/master/Protocol

Developer @elric91 has a "pyzigate" Python library for ZiGate that is being developed here:

https://github.com/elric91/ZiGate

He @elric91 has also made a ZiGate component for Home Assistant which might be ported:

https://github.com/elric91/homeassistant_zigate

Hedda, 2019-01-21:
Apparently @doudz have already started development here https://github.com/doudz/zigpy-zigate

doudz, 2019-03-18:
I'm working on support of zigate and I'm getting an error but I don't understand what's going wrong, any help will be appreciate 

```
2019-03-16 17:02:01 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler RESPONSE 0x8002 - Data indication : status:0, profile_id:0, cluster_id:32773, source_endpoint:0, destination_endpoint:0, lqi:201, source_address_mode:2, source_address:b8c6, dst_address_mode:2, dst_address:0000, payload_size:1, payload:b'\x00\xc6\xb8\x01\x01'
2019-03-16 17:02:01 ERROR (MainThread) [zigpy_zigate.api] Exception running handler
Traceback (most recent call last):
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy_zigate/api.py", line 59, in handle_callback
    handler(*args)
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy_zigate/zigbee/application.py", line 60, in zigate_callback_handler
    response['cluster_id'], response['payload'])
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy/application.py", line 74, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy/device.py", line 98, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy/zdo/__init__.py", line 36, in deserialize
    args, data = t.deserialize(data, cluster_details[2])
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/home/doudz/.homeassistant/deps/lib/python3.6/site-packages/zigpy/types/basic.py", line 12, in deserialize
    r = cls.from_bytes(data[:cls._size], 'little', signed=cls._signed)
  File "/usr/lib/python3.6/enum.py", line 293, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.6/enum.py", line 535, in __new__
    return cls._missing_(value)
  File "/usr/lib/python3.6/enum.py", line 548, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 198 is not a valid Status
```

doudz, 2019-03-18:
Another information, I made a HA component too https://github.com/doudz/homeassistant-zigate

Adminiuga, 2019-03-18:
It tries serialize ZDO 0x8005 `Active_EP_rsp` which has ZDO Status field and it gets 198/0xC6 for Status, which is not defined in https://github.com/zigpy/zigpy/blob/450b27d441d51bd4dbea755321368a48c83c8cff/zigpy/zdo/types.py#L128

```
source_address:b8c6, dst_address_mode:2, dst_address:0000, payload_size:1, payload:b'\x00\xc6\xb8\x01\x01'
```
IMO you're stripping somewhere too much data. Per Payload it interprets `\xc6` as Status, but looks like it is really a part of `0xb8c6` NWK address of the sending device. 
If I read the payload correctly, then
\x00 -- Status.SUCCESS
\xc6\xb8 -- NWK 0xb8c6
\x01 -- number of endpoints/list length - 1
\x01 -- Active endpoint #1

doudz, 2019-03-18:
But I pass the full payload to deserialize ...

Looking the code I see that the first bit should be `frame_control` but it's not in the payload provided by the zigate, I have to check the firmware
What frame_control is supposed to be ?

    https://github.com/zigpy/zigpy/blob/450b27d441d51bd4dbea755321368a48c83c8cff/zigpy/endpoint.py#L149-L152

Adminiuga, 2019-03-18:
Could it be Zigate interprets and strips that field for you? 
Check Page 2-6 of [ZCL Specs](http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf) section 2.4.1 for `frame control` field

Adminiuga, 2019-03-18:
On a second look I think Zigate already did "endpoint" deserialization, so the actual "payload" of  `b'\x00\xc6\xb8\x01\x01'` should be fed to `zcl.Cluster.deserialize()`, but to do so you also need to provide it with TSN, frame_type, is_reply and command_id. I see only `command_id` logged. I don't know internals of Zigate to comment further.

Adminiuga, 2019-03-18:
In other words, I think the function of the following code
https://github.com/zigpy/zigpy/blob/450b27d441d51bd4dbea755321368a48c83c8cff/zigpy/endpoint.py#L149-L156
was already done somewhere in Zigate, and your 'b'\x00\xc6\xb8\x01\x01'` pyaload is what should have been assigned to `data` in line 156

doudz, 2019-03-18:
That make sense, thanks for your help I'll make a test

doudz, 2019-03-18:
it works ! :smile: 
But I'm surprised, XIAOMI devices are not well supported/recognized on HA

[image]

Adminiuga, 2019-03-18:
With https://github.com/zigpy/zigpy/pull/131 it should get better. Currently zigpy attribute cache is lost on quirks application.

doudz, 2019-03-20:
I'm not sure we're are talking about the same problem.
lumi.sensor_switch.aq2 is discovered as a switch on HA but it couldn't be a switch since you can't control it. It should be a binary sensor
There's also a missing sensor for the "multiclick" attribute of that device

Another problem is that both devices have a temperature sensor in HA but real devices have no temperature sensor ! The temperature cluster doesn't exist on such devices so I don't understand why it shows up in HA

dmulcahey, 2019-03-20:
The temp is device temp and it’s experimental. It’s from the quirk. The device shouldn’t show up as a switch though... not sure what’s up there.

Hedda, 2019-04-02:
@doudz FYI, unique Zigbee device quirks are handled with ZHA Device Handlers for Home Assistant:

https://github.com/dmulcahey/zha-device-handlers

_"ZHA Device Handlers are custom quirks implementations for [Zigpy](https://github.com/zigpy/zigpy), the library that provides the [Zigbee](http://www.zigbee.org) support for the [ZHA](https://www.home-assistant.io/components/zha/) component in [Home Assistant](https://www.home-assistant.io). These ZHA Device Handlers can be used to parse custom messages to and from Zigbee devices. Custom quirks implementations for zigpy implemented as ZHA Device Handlers are a similar concept to that of [Hub-connected Device Handlers for the SmartThings Classics platform](https://docs.smartthings.com/en/latest/device-type-developers-guide/) as well that of [Zigbee-Shepherd Converters as used by Zigbee2mqtt](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html), meaning they are virtual representation of a physical device that expose additional functionality that is not provided out-of-the-box by the existing integration between these platforms. See [Device Specifics](#Device-Specifics) for details."_

doudz, 2019-07-28:
I just publish zigpy-zigate v0.1.0, it's ready to test

Adminiuga, 2019-08-13:
Fixed via https://github.com/doudz/zigpy-zigate

closed 2019-08-13 by Adminiuga