zigpy/zigpy #122: Python 3.4 support

doudz, 2019-02-01:
I know support of python 3.4 has been dropped https://github.com/zigpy/zigpy/issues/5 when HA did it too but since zigpy is supposed to be "universal" I think it could be useful to still support python 3.4 for other system.
For example Jeedom still use python 3.4 on some hardware

Additionnally Python 3.4 is not dead yet, the next release 3.4.10 is planned for March 2019
https://www.python.org/dev/peps/pep-0429/

doudz, 2019-02-01:
and the same for zigpy-xbee, zigbee-deconz, bellows...

dmulcahey, 2019-02-01:
I don’t speak for everyone but I definitely prefer async await syntax so IMHO I don’t think going backwards is a good idea. I’d say fork the project if that’s what you need but that’s just my opinion.

closed 2019-02-02 by doudz