zigpy/zigpy #123: Attribute Report prior to Init causes exception

prairiesnpr, 2019-02-03:
If a device reports an attribute value prior to completing initialization. The attribute will be written to the database, but the corresponding device entry isn't created.  This will cause an exception during the Home Assistant startup and prevent any ZHA devices from working until the database is manually edited to remove the offending entry.

homeassistant==0.88.0.dev0                                                     
zigpy==0.2.0                                                                                                                               
zigpy-xbee==0.1.1

```
2019-02-03 11:20:15 ERROR (MainThread) [homeassistant.config_entries] Error setti
ng up entry /dev/ttyUSB0 for zha                                                 
Traceback (most recent call last):                                               
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/config_entri
es.py", line 258, in async_setup                                                 
    result = await component.async_setup_entry(hass, self)                       
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/z
ha/__init__.py", line 140, in async_setup_entry                                  
    application_controller = ControllerApplication(radio, database)              
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_xbee/zigbee/applicat
ion.py", line 15, in __init__                                                    
    super().__init__(database_file=database_file)                                
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/application.py", lin
e 25, in __init__                                                                
    self._dblistener.load()                                                      
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/appdb.py", line 214,
 in load                                                                         
    dev = self._application.get_device(ieee)                                     
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/application.py", lin
e 116, in get_device                                                             
    return self.devices[ieee]                                                    
KeyError: 00:13:a2:00:40:c0:4e:54 
```

dmulcahey, 2019-02-03:
We’re going to need the full logs from both Zigpy libraries at debug level to confirm your assertion. 

I don’t use xbee personally but the only time I have seen these errors is when a device times out during pairing and it isn’t fully inserted into the DB as a result (that also needs to be corrected) 

The reason I am skeptical of the cause is because Xiaomi devices need to be kept awake during pairing which means you have to press the button constantly. This causes them to send reports on each press. I would expect every Xiaomi device to fail with this error if the reporting itself was in fact causing an issue and that isn’t the case.

Attach logs that also show the full set of network traffic and we’ll take a look.

prairiesnpr, 2019-02-03:
In an attempt to prove I'm not crazy.

Logs are here: https://hastebin.com/rajikidafi.http

DB Tables; https://hastebin.com/aroyukepin.rb

Note: This is caused by my debugging of a custom device I'm writing, so really not an issue, just wanted to bring it up, since the changes should really be rolled back when the join fails.

dmulcahey, 2019-02-03:
Please don’t truncate the logs. Can you post the non edited version please.

prairiesnpr, 2019-02-03:
Logs from join attempt, nothing happens after last line: https://hastebin.com/upozotewiw.coffeescript
Then restart home assistant.
Logs from restart: https://hastebin.com/kimohamogo.coffeescript

dmulcahey, 2019-02-03:
You can see from the logs that the device in question has 4 endpoints. You’re only getting discovery info for endpoint 1. This device isnt Joined correctly. If it had completed there would be simple descriptor logs for all 4 endpoints. I don’t think attribute reporting has anything to do with the issue at hand.

Adminiuga, 2019-02-08:
@dmulcahey I can confirm the symptoms as I had it happened a few times with Xiaomi devices. Could be related to https://github.com/zigpy/zigpy/pull/108 

What I think is happening:
1. Device is being discovered
2. one endpoint was successfully discovered, but other are still being processed
3. Zigpy receives attribute update on already discovered endpoint, which means deserarialization completes successfully so attribute update lands into the cache db
4. the discovery of other endpoints fails, so the devices is not in the persistent storage
5. upon next restart, zigpy fails, untill you clean zigpy db manually

beatsWorking, 2019-02-11:
Is there any way to resolve this while the fix is being developed? I tried clearing my zigbee.db database but the issue persists.

Adminiuga, 2019-02-11:
@beatsWorking 
1. Call `zha.remove` service with `{"ieee_address": "00:22:a3:00:00:25:83:74"}` service data a few times (give it 10s between the calls)
2. Factory Reset GE switch
3. check/clear database if needed

Re-join GE dimmer and you **have to move coordinator closer** to GE dimmer during the process. Make sure it successfully finishes the discovery, you'll know that if you are going to see a bunch of messages similar to:
```
2019-02-11 11:54:58 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.xiaomi.TemperatureHumiditySensor'>
2019-02-11 11:54:58 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1, 2, 3]) {1, 196}
2019-02-11 11:54:58 DEBUG (MainThread) [zigpy.quirks] Considering <class 'zigpy.quirks.xiaomi.AqaraTemperatureHumidityPressureSensor'>
2019-02-11 11:54:58 DEBUG (MainThread) [zigpy.quirks] Fail because endpoint list mismatch: dict_keys([1]) {1, 196}
```
Assuming you have debug logging on.  Also, if the discovery is complete, you should see your device in the output of `bellows devices -D zigbee.db` command with two endpoints.
You need to keep coordinator close to GE dimmer only during initial joning/re-joining. After that you can put coordinator back were it was assuming you have other Zigbee devices acting as Zigbee routers (usually mains powered devices with the exception of Senled bulbs and Wink fan controller)

beatsWorking, 2019-02-11:
I was not able to call the zha.remove service without clearing the zigbee.db file first. When I clear the zigbee file, restart HA, call zha.remove a few times, then try to reconnect (with the controller right next to the dimmer), I don't see the device but I don't get an error. Then when I restart HA, I get the same error again.

Any other ideas?

Adminiuga, 2019-02-11:
Enable debug logging. 
Did you factory reset the dimmer prior rejoining it? The blue led should blink slowly while it is searching for the network.

beatsWorking, 2019-02-11:
Debug logging is on. Yes, I've reset the dimmer prior to rejoining. the blue led flashes for about 10 seconds and then stops (I'm assuming it joined). I grabbed the relevant sections out of the debug logs:

2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e5d000700aa00
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x175d000800010001
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x175d002a002300220200000102cc8701040119000c0001200100241100000600000000afffa3000104c9
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=2 address=0>, 1, <DeconzAddress address_mode=2 address=34764>, 1, 260, 25, b'\x01 \x01\x00$\x11\x00\x00\x06\x00\x00\x00
', 0, 175, 255, 163, 0, 1, 4, -55]
2019-02-11 11:50:30 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame on uninitialized device 00:22:a3:00:00:25:83:74 endpoint 1 for cluster: 25
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e5e000700aa00
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x175e000800010001
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x175e002a002300220200000102cc8701040119000c0001210100241100000600000000afffa3000104c9
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=2 address=0>, 1, <DeconzAddress address_mode=2 address=34764>, 1, 260, 25, b'\x01!\x01\x00$\x11\x00\x00\x06\x00\x00\x00
', 0, 175, 255, 163, 0, 1, 4, -55]
2019-02-11 11:50:33 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame on uninitialized device 00:22:a3:00:00:25:83:74 endpoint 1 for cluster: 25
2019-02-11 11:50:35 INFO (Thread-14) [homeassistant.components.device_tracker.nmap_tracker] Scanning...
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e5f000700aa00
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x175f000800010001
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x175f002a002300220200000102cc8701040119000c0001220100241100000600000000afffa3000104ca
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=2 address=0>, 1, <DeconzAddress address_mode=2 address=34764>, 1, 260, 25, b'\x01"\x01\x00$\x11\x00\x00\x06\x00\x00\x00
', 0, 175, 255, 163, 0, 1, 4, -54]
2019-02-11 11:50:36 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame on uninitialized device 00:22:a3:00:00:25:83:74 endpoint 1 for cluster: 25
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e60000700aa00
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1760000800010001
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1760002a002300220200000102cc8701040119000c0001230100241100000600000000afffa3000102ca
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=2 address=0>, 1, <DeconzAddress address_mode=2 address=34764>, 1, 260, 25, b'\x01#\x01\x00$\x11\x00\x00\x06\x00\x00\x00
', 0, 175, 255, 163, 0, 1, 2, -54]
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame on uninitialized device 00:22:a3:00:00:25:83:74 endpoint 1 for cluster: 25

I can send the entire log if helpful.

Adminiuga, 2019-02-11:
Did you call `zha.permit` after resetting GE Dimmer? Log Line
```
2019-02-11 11:50:39 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Received frame on uninitialized device 00:22:a3:00:00:25:83:74 endpoint 1 for cluster: 25
```
implies `zigpy-deconz` started but did not finish new device discovery. Post the logs after 
`Device 0x87cc (00:22:a3:00:00:25:83:74) joined the network`
Also, I'm perplexed you are getting `KeyError: 00:22:a3:00:00:25:83:74` error with Deconz as there are safeguards to ignore ZCL messaging until device is fully discovered, so your issue could be something else.

beatsWorking, 2019-02-11:
yes, I ran zha.permit right before I brought the GE dimmer back on line.
Here are the logs you requested:

```
2019-02-11 10:47:53 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0x87cc, 00:22:a3:00:00:25:83:74
2019-02-11 10:47:53 INFO (MainThread) [zigpy.application] Device 0x87cc (00:22:a3:00:00:25:83:74) joined the network
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy.zdo] [0x87cc:zdo] ZDO request 0x0013: [9603, 00:00:8e:00:22:a3:00:00, 0]
2019-02-11 10:47:53 INFO (MainThread) [zigpy.device] [0x87cc] Discovering endpoints
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 1, data: b'01cc87'
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_request (18, 1, 0, <DeconzAddress address_mode=2 address=34764>, 0, 0, 5, 0, b'\x01\xcc\x87', 0, 0)
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12110019001200010002cc87000000050000030001cc870000
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121100090002002201
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 1]
2019-02-11 10:47:55 INFO (Thread-9) [homeassistant.components.device_tracker.nmap_tracker] Scanning...
2019-02-11 10:48:00 INFO (Thread-9) [homeassistant.components.device_tracker.nmap_tracker] No MAC address found for 192.168.0.113
2019-02-11 10:48:00 INFO (Thread-9) [homeassistant.components.device_tracker.nmap_tracker] No MAC address found for 192.168.0.105
2019-02-11 10:48:00 INFO (Thread-9) [homeassistant.components.device_tracker.nmap_tracker] nmap scan successful
2019-02-11 10:48:00 DEBUG (Thread-9) [homeassistant.components.device_tracker.nmap_tracker] Nmap last results [Device(mac='48:3B:38:1A:0D:58', name='', ip='192.168.0.107', last_update=datetime.datetime(2019, 2, 11, 10, 41, 39, 182218, tzinfo=<D
stTzInfo 'America/Los_Angeles' PST-1 day, 16:00:00 STD>)), Device(mac='B4:3A:28:F5:F5:A6', name='', ip='192.168.0.103', last_update=datetime.datetime(2019, 2, 11, 10, 41, 39, 182218, tzinfo=<DstTzInfo 'America/Los_Angeles' PST-1 day, 16:00:00 S
TD>)), Device(mac='18:D6:C7:24:7C:CD', name='', ip='192.168.0.1', last_update=datetime.datetime(2019, 2, 11, 10, 41, 39, 182218, tzinfo=<DstTzInfo 'America/Los_Angeles' PST-1 day, 16:00:00 STD>)), Device(mac='50:DC:E7:4F:9E:23', name='', ip='19
2.168.0.109', last_update=datetime.datetime(2019, 2, 11, 10, 41, 39, 182218, tzinfo=<DstTzInfo 'America/Los_Angeles' PST-1 day, 16:00:00 STD>)), Device(mac='CC:F7:35:37:BC:99', name='', ip='192.168.0.106', last_update=datetime.datetime(2019, 2,
 11, 10, 41, 39, 182218, tzinfo=<DstTzInfo 'America/Los_Angeles' PST-1 day, 16:00:00 STD>)), Device(mac='BC:3B:AF:A5:21:5E', name='', ip='192.168.0.104', last_update=datetime.datetime(2019, 2, 11, 10, 42, 9, 486405, tzinfo=<DstTzInfo 'America/L
os_Angeles' PST-1 day, 16:00:00 STD>))]
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e12000700a600
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 0]
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_confirm (0,)
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04120007000000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04120013000c00220102cc8700000000000000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 1: 00
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e13000700aa00
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1713000800010001
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17130025001e00220200000002cc87000000058007000100cc8702010200afd7a3000102c0
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, 34, <DeconzAddress address_mode=2 address=0>, 0, <DeconzAddress address_mode=2 address=34764>, 0, 0, 32773, b'\x01\x00\xcc\x87\x02\x01\x02', 0, 175, 21
5, 163, 0, 1, 2, -64]
2019-02-11 10:48:03 INFO (MainThread) [zigpy.device] [0x87cc] Discovered endpoints: [1, 2]
2019-02-11 10:48:03 INFO (MainThread) [zigpy.endpoint] [0x87cc:1] Discovering endpoint information
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 2, data: b'02cc8701'
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_request (19, 2, 0, <DeconzAddress address_mode=2 address=34764>, 0, 0, 4, 0, b'\x02\xcc\x87\x01', 0, 0)
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1214001a001300020002cc87000000040000040002cc87010000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121400090002002202
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 2]
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e15000700a600
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 0]
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_confirm (0,)
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04150007000000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04150013000c00220202cc8700000000000000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 2: 00
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e16000700aa00
```

beatsWorking, 2019-02-11:
Adding a link to additional log items after the cutoff: https://justpaste.it/6emrq

Adminiuga, 2019-02-11:
Well, maybe this is a candidate for `zigpy-deconz`. 
@beatsWorking as of it is right after the above trace was taken, does ZHA still fails to load if you restart HA?

I see that it finishes discovery of the two endpoints
```
2019-02-11 10:47:53 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0x87cc, 00:22:a3:00:00:25:83:74
2019-02-11 10:47:53 INFO (MainThread) [zigpy.application] Device 0x87cc (00:22:a3:00:00:25:83:74) joined the network
2019-02-11 10:47:53 DEBUG (MainThread) [zigpy.zdo] [0x87cc:zdo] ZDO request 0x0013: [9603, 00:00:8e:00:22:a3:00:00, 0]
2019-02-11 10:47:53 INFO (MainThread) [zigpy.device] [0x87cc] Discovering endpoints
...
2019-02-11 10:48:03 INFO (MainThread) [zigpy.device] [0x87cc] Discovered endpoints: [1, 2]
2019-02-11 10:48:03 INFO (MainThread) [zigpy.endpoint] [0x87cc:1] Discovering endpoint information
...
2019-02-11 10:48:03 INFO (MainThread) [zigpy.endpoint] [0x87cc:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=257 device_version=0 input_clusters=[0, 3, 4, 5, 6, 8, 2821, 1794] output_clusters=[10, 25]>
...
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Manufacturer: Jasco Products
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Model: 45857
2019-02-11 10:48:03 INFO (MainThread) [zigpy.endpoint] [0x87cc:2] Discovering endpoint information
...
2019-02-11 10:48:03 INFO (MainThread) [zigpy.endpoint] [0x87cc:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=260 device_version=0 input_clusters=[0, 3, 2821] output_clusters=[3, 6, 8]>
...
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x041e0013000c00220502cc870202a600000000
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 5: a6
2019-02-11 10:48:03 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xa6
2019-02-11 10:48:30 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1f000700aa00
```

So it discovers two endpoints, but I never see it getting model/manufacturer for the 2nd endpoint -> device discovery incomplete and there's error sending frame a6 which my guess is the frame getting model/manufacturer 

@damarco any ideas?

beatsWorking, 2019-02-12:
yep - ZHA fails to start and I get the same error message "KeyError: 00:22:a3:00:00:25:83:74".

About 30-40 lines up from tha warning you referenced, I'm seeing: 
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Manufacturer: Jasco Products
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Model: 45857

So it must be getting model/mfg somehow. Could it be a different frame that's causing it to fail?

Adminiuga, 2019-02-12:
>About 30-40 lines up from tha warning you referenced, I'm seeing:
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Manufacturer: Jasco Products
2019-02-11 10:48:03 DEBUG (MainThread) [zigpy.endpoint] [0x87cc:1] Model: 45857

There're two endpoints. These model & manufacturer for the Endpoint `#1` you should see similar but with `[0x87cc:2]` denoting model & manufacturer for endpoint `#2`

can you run `sqlite3 zigbee.db .dump | grep '00:22:a3:00:00:25:83:74'` and post it here? Did you clear the `attributes` table, because if you did I don't get how attribute update got through? Initially I assumed you were running `bellows` not `zigpy-deconz`

beatsWorking, 2019-02-12:
I see. I didn't realize that it was going to get the data twice. I don't see anything for [0x87cc:2].

Regarding the zigbee.db table, I cleared the entire DB (deleted the DB and re-created it). Here's the requested info:

```
sqlite3 zigbee.db .dump | grep '00:22:a3:00:00:25:83:74'
INSERT INTO "attributes" VALUES('00:22:a3:00:00:25:83:74',1,0,4,X'4A6173636F2050726F6475637473');
INSERT INTO "attributes" VALUES('00:22:a3:00:00:25:83:74',1,0,5,X'3435383537');
```

Adminiuga, 2019-02-12:
Argh, that is attribute update, but not sent by the device, but requested
by zigpy endpoint discovery. So yes, part of the issue is related to this
one, but does not explain why it doesn't finish the discovery.

beatsWorking, 2019-02-12:
Update -

I added `baudrate: 38400` to the configuration file (deconz recommends the slower speed and I thought that may have to do with why I'm only getting one endpoint) and I'm getting a new error.

When I start Home Assistant I get this:
`Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/zigbee/application.py", line 34, in _reset_watchdog
    await self._api.write_parameter(NETWORK_PARAMETER['watchdog_ttl'][0], 3600)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/api.py", line 207, in write_parameter
    timeout=COMMAND_TIMEOUT
  File "/usr/lib/python3.5/asyncio/tasks.py", line 400, in wait_for
    return fut.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
Exception: write_parameter, status: 4`

It seems like pairing goes ok, but I get a new error when I try to turn the light on and off. I get this error ~3 times everytime I turn the switch on and off:

`Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/uart.py", line 63, in data_received
    self._api.data_received(frame)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/api.py", line 159, in data_received
    getattr(self, '_handle_%s' % (command, ))(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/api.py", line 235, in _handle_aps_data_indication
    data[16])   # rssi
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/zigbee/application.py", line 130, in handle_rx
    raise Exception("Unsupported address mode in handle_rx: %s" % (src_addr.address_mode))
Exception: Unsupported address mode in handle_rx: 3`

damarco, 2019-02-12:
This looks similar to [#11](https://github.com/zigpy/zigpy-deconz/issues/11) and [#9](https://github.com/zigpy/zigpy-deconz/issues/9).

Which firmware version are you running? The baudrate is hardcoded to 38400 in [zigpy-deconz](https://github.com/zigpy/zigpy-deconz/blob/6c25a95fda9039bfb539696197d59441427209a4/zigpy_deconz/api.py#L116), so changing it in the configuration file should not make any difference.

beatsWorking, 2019-02-12:
Version 0x260b0500

beatsWorking, 2019-02-12:
Just upgraded to the newest firmware - 0x26300500 - and the write parameter error goes away. Not sure how that happened.

So I'm back to the original error on this thread: 
"KeyError: 00:22:a3:00:00:25:83:74"

beatsWorking, 2019-02-12:
I am getting a new warning though (it may not be new, but I'm seeing it now.

2019-02-12 15:10:26 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xa6

Here's the log leading up to the warning:

`2019-02-12 15:10:25 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-12 15:10:25 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1719000800010001
2019-02-12 15:10:25 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1719003e0037002202000001021b1601040100002000080301040000420e4a6173636f2050726f64756374730500004205343538353700afffa3000102c6
2019-02-12 15:10:25 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [55, 34, <DeconzAddress address_mode=2 address=0>, 1, <DeconzAddress address_mode=2 address=5659>, 1, 260, 0, b'\x08\x03\x01\x04\x00\x00B\x0eJasco Products\x05\x00\x00B\x0545857', 0, 175, 255, 163, 0, 1, 2, -58]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy.endpoint] [0x161b:1] Manufacturer: Jasco Products
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy.endpoint] [0x161b:1] Model: 45857
2019-02-12 15:10:26 INFO (MainThread) [zigpy.endpoint] [0x161b:2] Discovering endpoint information
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 4, data: b'041b1602'
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_request (19, 4, 0, <DeconzAddress address_mode=2 address=5659>, 0, 0, 4, 0, b'\x04\x1b\x16\x02', 0, 0)
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121a001a0013000400021b160000000400000400041b16020000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121a00090002002204
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 4]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1b000700a600
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 0]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_confirm (0,)
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x041b0007000000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x041b0013000c002204021b1600000000000000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 4: 00
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1c000700aa00
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_indication (1, 1)
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x171c000800010001
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x171c00370030002202000000021b160000000480190004001b16140204010401000300000300050b0303000600080000afffa3000102c6
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [48, 34, <DeconzAddress address_mode=2 address=0>, 0, <DeconzAddress address_mode=2 address=5659>, 0, 0, 32772, b'\x04\x00\x1b\x16\x14\x02\x04\x01\x04\x01\x00\x03\x00\x00\x03\x00\x05\x0b\x03\x03\x00\x06\x00\x08\x00', 0, 175, 255, 163, 0, 1, 2, -58]
2019-02-12 15:10:26 INFO (MainThread) [zigpy.endpoint] [0x161b:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=260 device_version=0 input_clusters=[0, 3, 2821] output_clusters=[3, 6, 8]>
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 5, data: b'00050004000500'
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_request (22, 5, 0, <DeconzAddress address_mode=2 address=5659>, 2, 260, 0, 2, b'\x00\x05\x00\x04\x00\x05\x00', 0, 0)
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121d001d0016000500021b160204010000020700000500040005000000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121d00090002002205
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 5]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1e000700a600
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 0]
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] Command aps_data_confirm (0,)
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x041e0007000000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x041e0013000c002205021b160202a600000000
2019-02-12 15:10:26 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 5: a6
2019-02-12 15:10:26 WARNING (MainThread) [zigpy_deconz.zigbee.application] Error while sending frame: 0xa6`

damarco, 2019-02-13:
Can you maybe try the following fix: [link](https://github.com/damarco/zigpy-deconz/commit/80554dcbdded734d0f0763b05c2e413a1a20b0fa)

You only need to change a single line in `zigpy_deconz/zigbee/application.py`.

beatsWorking, 2019-02-13:
That did it!

Adminiuga, 2019-03-12:
@damarco @dmulcahey just bumping this up. 
Question: during the discovery of the new device, should we update the db with each endpoint as we receive update for each endpoint? eg
1. upon receiving `Active_EP_rsp` we dump all endpoints with status `NEW` into db
2. as we receive `Simple_Desc_rsp` for each endpoint we write it into DB
This is more complex but would allow as to resume device discovery if the discovery fails. 
Thoughts?

closed 2019-04-20 by damarco