zigpy/zigpy #124: Error when adding GE Smart Dimmer

beatsWorking, 2019-02-11:
i'm getting an error when adding a GE smart dimmer switch on zigbee-deconz:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/config_entries.py", line 258, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/zha/__init__.py", line 140, in async_setup_entry
    application_controller = ControllerApplication(radio, database)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy_deconz/zigbee/application.py", line 21, in __init__
    super().__init__(database_file=database_file)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/application.py", line 25, in __init__
    self._dblistener.load()
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/appdb.py", line 214, in load
    dev = self._application.get_device(ieee)
  File "/srv/homeassistant/lib/python3.5/site-packages/zigpy/application.py", line 116, in get_device
    return self.devices[ieee]
KeyError: 00:22:a3:00:00:25:83:74

Any help would be appreciated! Happy to provide additional information

Adminiuga, 2019-02-11:
This is duplicate of #123 Please move it there.

beatsWorking, 2019-02-11:
Duplicate of #123

closed 2019-02-11 by beatsWorking