zigpy/zigpy #125: Switch turns on but doesn't turn off

beatsWorking, 2019-02-15:
I've been able to add the Jasco switch to my home assistant using zigpy-deconz package, but now I'm able to turn off the light but I'm not able to turn it back on.

I've added and remove the light several times and here are the scenarios and how HA responds.
1. I can always turn the light off in the HA portal. 
2. I can see the current status of the switch in the HA portal whenever I turn the switch on or off at the switch. 
3. If the first command I give to the switch through the HA portal is to turn it on, then it works. Every subsequent attempt to turn the switch on remotely doesn't work.

Here's a copy of my logs (I removed anything that doesn't include Zigbee in the log entry to reduce the size of the file).
https://pastebin.com/wep19P0S

Adminiuga, 2019-02-15:
can you please include `[homeassistant.components.zha]` log lines?

Adminiuga, 2019-02-15:
> 3\. If the first command I give to the switch through the HA portal is to turn it on, then it works. Every subsequent attempt to turn the switch on remotely doesn't work.

Can you do the following test from HA:
1. turn switch on
2. descres the brightness from UI
3. increase the brightness from UI

beatsWorking, 2019-02-15:
New paste with the requested lines. The new log file includes the tests above.

https://pastebin.com/Mxvqiur6

Can you do the following test from HA:

1. turn the light on - 
   response - as long as it's the first thing done after device setup, IO can turn the light on from HA
2. descrese the brightness from UI
    response - as long as the device starts in the "on" state, I can decrease brightness. If the device is in the "off" state, I cannot decrease brightness
3. increase the brightness from UI
    response - as long as the device starts in the "on" state, I can increase brightness. If the device is in the "off" state, I cannot increase brightness. Even if the device is at the minimum brightness (almost off) I can still increase brightness.

Adminiuga, 2019-02-15:
I think you misinterpreted me. I'd like you to do the following, while the light is kept on in UI, don't switch it off:
1. decrease the brightness (increase it prior to that if there's no room for a decrease
2. increase the brightness -- the idea for HA to send `move_to_level_with_on_off` command with higher than the current level

Adminiuga, 2019-02-15:
while the light is on in the UI (but still hasn't turned on physical light) I'm expecting the light to actually turn on if you increase the brightness. In other words I'm trying to understand if it is the same problem as in https://github.com/home-assistant/home-assistant/issues/12813

beatsWorking, 2019-02-15:
> problem as in home-assistant/home-assistant#12813

I see. So if the light is off but it's on the the UI, the light comes on
and adjusts to the desired brightness. Starting with either increase and
decrease commands produce the same effect (light comes on to the brightness
level set in the UI).

beatsWorking, 2019-02-16:
The issue I'm seeing looks like https://github.com/home-assistant/home-assistant/issues/12813, but it looks like that issue has been resolved and committed. Is this a new bug or are my settings incorrect somehow?

Adminiuga, 2019-02-16:
it wasn't resolved and was closed because of lack of traction. The issue is really with HA and not zigpy. Do you want to open a new issue with HA? May get this through the more feedback we get.

beatsWorking, 2019-02-16:
Got it. Reported on https://github.com/home-assistant/home-assistant/issues/21107

beatsWorking, 2019-02-16:
Feel free to pile on

Adminiuga, 2019-05-04:
should be fixed via https://github.com/home-assistant/home-assistant/pull/21156
reopen if the problem still persists.

closed 2019-05-04 by Adminiuga