zigpy/zigpy #127: Suggest to flesh out README.md with inspiration from zsmartsystems/com.zsmartsystems.zigbee

Gamester17, 2019-02-18:
I like to suggest fleshing out README.md with inspiration from zsmartsystems/com.zsmartsystems.zigbee

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/README.md

This to make README.md contain more informative information about its functionality and capabilities.

My thinking is that a more explanatory README.md might attract more developers?

meremortals70, 2020-08-03:
Another vote for more documentation please?

Gamester17, 2020-09-14:
> Another vote for more documentation please?

@meremortals70  are you following these related discussions with @pipiche38 in regards to getting started with zigpy development?

These discussions all started with https://github.com/zigpy/zigpy/issues/452 and then have snowballed from there:

https://github.com/zigpy/zigpy/pull/470

https://github.com/zigpy/zigpy/pull/477

https://github.com/zigpy/zigpy/issues/471

https://github.com/zigpy/zigpy/issues/469

PS: @pipiche38 is asking questions since he started developing Domoticz-Zigpy https://github.com/pipiche38/Domoticz-Zigpy/

closed 2021-09-07 by github-actions[bot]