zigpy/zigpy #132: Support AlertMe/Iris proprietary clusters

maladaptly, 2019-03-18:
With the release of some (most?) of the source code of the Iris system there is now public documentation of the behavior of these devices and how to interface them.

The meat and potatoes appears to be here: https://github.com/arcus-smart-home/arcusplatform/tree/d5691b01f40cac5e5d4de9e1c8f8b6410a16daf4/common/arcus-protocol/src/main/irp (the ame-* files)

They may no longer be sold by Lowe's but there's still a lot of them out there second-hand and people selling off old stock. It'd be nice to have more options.

prairiesnpr, 2019-03-19:
I have two of the iris smart plugs and wasted a bunch of time writing quirks so they would work with home assistant.  Unfortunately, as I added devices I discovered that some newer devices would only connect if the network was encrypted and when I encrypted the network, the iris smart plugs would no longer connect to the network.  Anyways, I really think its a dead end at this point.  I've tried several times to get them to connect to the network, xctu will report that it has seen them, but they are not available and I never see traffic from them on the coordinator.

Adminiuga, 2020-05-10:
If there's still interest, feel free to submit PR for it.

closed 2020-05-10 by Adminiuga