zigpy/zigpy #159: UnicodeDecodeError: 'ascii' codec can't decode byte 0x91 in position 3: ordinal not in range(128)

anderwm, 2019-05-17:
Hello, I am using zigpy_xbee inside homeassistant. It looks like sometimes parts of the API frame header are trying to be interpreted as ASCII. This happens randomly but 10-15 times a day. Is this a problem in how I have setup my radio or a real bug?

> 2019-05-17 05:42:25 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback <bound method SerialTransport._read_ready of SerialTransport(<uvloop.Loop running=True closed=False debug=False>, <zigpy_xbee.uart.Gateway object at 0x722ef550>, Serial<id=0x722ef570, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))>
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 66, in uvloop.loop.Handle._run
  File "/usr/local/lib/python3.7/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.7/site-packages/zigpy_xbee/uart.py", line 78, in data_received
    self.command_mode_rsp(rsp)
  File "/usr/local/lib/python3.7/site-packages/zigpy_xbee/uart.py", line 55, in command_mode_rsp
    data = data.decode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0x91 in position 3: ordinal not in range(128)

Adminiuga, 2019-05-17:
can you please move this to https://github.com/zigpy/zigpy-xbee (as it is the issue with xbee) and post full log along with what Xbee are you using and XBee firmware version?
During normal operations, there shouldn't be any use of `command_mode_rsp()` as those are used only once when XBee is not configured for API mode.

Adminiuga, 2019-05-22:
this belongs to `zigpy-xbee`. Please move it.

closed 2019-05-22 by Adminiuga