zigpy/zigpy #174: convert install code gets byte order wrong

rayzorben, 2019-06-03:
```python
def convert_install_code(code):
    if len(code) < 8:
        return None

    real_crc = bytes([code[-1], code[-2]])
    crc = CrcX25()
    crc.process(code[:len(code) - 2])
    if real_crc != crc.finalbytes():
        return None

    return aes_mmo_hash(code)
```

This code converts incorrectly. With a code of `313233343536E672` the value of real_crc is b'r\xe6' whereas the CrcX25 is b'xe6r'. To fix, the line needs to be changed to `real_crc = bytes([code[-2], code[-1]])`

Adminiuga, 2019-10-18:
is `313233343536E672` the real install code you've got? Just reading base device behavior specification and it says the install code is 128bit key + 16 bit crc. Just thinking if this should be enforced or not.

The 3320L centralite door sensor i've got scans with a 128 + 16bit code, although the install code from a ConBeeII stick is just 64bit

Adminiuga, 2019-10-18:
Nevermind, install code is
>While Zigbee smart energy networks allow the installation code to be comprised of either 6-, 8-, 12-, or 16-byte random, hexadecimal
number with a 2-byte CRC appended to the end, Zigbee 3.0 (Z3) networks specifically require 16-byte hexadecimal installation codes,

So Zigbee Energe devices could use 48 bits + 16bit crc.

closed 2019-10-18 by Adminiuga