zigpy/zigpy #193: Philips Hue Dimmer (RWL021) causing repeated exceptions

andrewpc, 2019-07-03:
Ive started to add a number of Philips Hue Dimmers to my zigbee network. They work fine however exceptions continue to be thrown in the Home Assistant log which look related to Zigpy/bellows. Here is a log excerpt. This seems to happen about every 30 minutes per dimmer.

```
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0400000600000000000000a9e7d08c2affff09b0fdff040101190000'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, <EmberApsFrame profileId=0 clusterId=6 sourceEndpoint=0 destinationEndpoint=0 options=0 groupId=0 sequence=169>, 231, -48, 0x2a8c, 255, 255, b'\xb0\xfd\xff\x04\x01\x01\x19\x00\x00']
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [zigpy.zdo] [0x2a8c:zdo] ZDO request 0x0006: [0xfffd, 260, [25], []]
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:17:88:01:06:72:a9:a4/0x2a8c
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (00:17:88:01:06:72:a9:a4, True)
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Sending: b'06da21d6f08367b4581c5d25ab2de97e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Data frame: b'61daa1d68c077e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 10892, <EmberApsFrame profileId=0 clusterId=32774 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=241>, 241, b'\xf1\x00\x00\x00\x01\x01')
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Sending: b'17db219c54a63fb25992ca25aa1593499cbfd6ad1cce678bfcc79cc37e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Data frame: b'72dba19c54fb16447e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
Jul 03 22:59:04 ha hass[29281]: 2019-07-03 22:59:04 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00d1'
Jul 03 22:59:08 ha hass[29281]: 2019-07-03 22:59:08 DEBUG (MainThread) [bellows.uart] Data frame: b'02dbb19754a63fb25992ca25aa1592499c9fd6cded227b7e'
Jul 03 22:59:08 ha hass[29281]: 2019-07-03 22:59:08 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
Jul 03 22:59:08 ha hass[29281]: 2019-07-03 22:59:08 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'008c2a00000680000040000000d1f16600'
Jul 03 22:59:08 ha hass[29281]: 2019-07-03 22:59:08 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 10892, <EmberApsFrame profileId=0 clusterId=32774 sourceEndpoint=0 destinationEndpoint=0 options=64 groupId=0 sequence=209>, 241, <EmberStatus.DELIVERY_FAILED: 102>, b'']
Jul 03 22:59:09 ha hass[29281]: 2019-07-03 22:59:09 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Jul 03 22:59:09 ha hass[29281]: Traceback (most recent call last):
Jul 03 22:59:09 ha hass[29281]:   File "/home/andrew/ha-config/deps/lib/python3.7/site-packages/bellows/zigbee/application.py", line 365, in request
Jul 03 22:59:09 ha hass[29281]:     res = await asyncio.wait_for(req.send, timeout=APS_ACK_TIMEOUT)
Jul 03 22:59:09 ha hass[29281]:   File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
Jul 03 22:59:09 ha hass[29281]:     return fut.result()
Jul 03 22:59:09 ha hass[29281]: zigpy.exceptions.DeliveryError: [0x2a8c:0:0x8006]: message send failure: EmberStatus.DELIVERY_FAILED
```

Adminiuga, 2019-07-03:
Ignore those for now. Those are replies to `Match_Desc_req` of HUE lights looking for an OTA upgrade server.

andrewpc, 2019-07-04:
Thank you. I will close for now

closed 2019-07-04 by andrewpc