zigpy/zigpy #206: How to get values from a device?

JimmyTheNerd, 2019-08-31:
Hello!

I used bellows CLI commands to join a device by doing the following:
```
export EZSP_DEVICE=/dev/ttyUSB0
bellows form
bellows permit
```

It worked. The following command outputs the device:
```
bellows devices
```

I'm wondering how I should use _zigpy/zigpy_ or _zigpy/bellows_ to get values, for example a temperature change, using Python. I couldn't find anything so far. 

Code snippets or a link to the docs are highly appreciated.

Thank you very much!

martin3000, 2019-08-31:
`bellows -d /dev/ttyUSB0 zcl '00:14:1b:00:07:19:64:ed' 1 1794 read-attribute 0`

but this works only for devices which are connected to mains. battery devices normally sleep and send their values as events, they cannot be polled.

dmulcahey, 2019-08-31:
You can poll them if you keep them awake... it’s just not ideal.

JimmyTheNerd, 2019-08-31:
Thanks @martin3000 for your reply.

> `bellows -d /dev/ttyUSB0 zcl '00:14:1b:00:07:19:64:ed' 1 1794 read-attribute 0`
I tried your command (adjusted to my device), but it fails. I'm using `Python 3.5.2`. Error log:
```
(zigbee_venv) jim@tower:~/zigbee_playground$ bellows zcl 00:0d:6f:00:0d:69:52:54 1 1026 read-attribute 0
Traceback (most recent call last):
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/bellows/cli/application.py", line 259, in read_attribute
    v = await cluster.read_attributes([attribute], allow_cache=False, manufacturer=manufacturer)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/zigpy/zcl/__init__.py", line 211, in read_attributes
    result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/zigpy/zcl/__init__.py", line 179, in read_attributes_raw
    v = await self.request(True, 0x00, schema, attributes, manufacturer=manufacturer)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/zigpy/device.py", line 89, in request
    expect_reply=expect_reply,
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/bellows/zigbee/application.py", line 241, in request
    v = await send_fut
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
zigpy.exceptions.DeliveryError: Message send failure: EmberStatus.DELIVERY_FAILED
  File "/home/jim/zigbee_playground/zigbee_venv/bin/bellows", line 10, in <module>
    sys.exit(main())
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/bellows/cli/util.py", line 71, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/bellows/cli/util.py", line 58, in async_inner
    await f(ctx, *args, **kwargs)
  File "/home/jim/zigbee_playground/zigbee_venv/lib/python3.5/site-packages/bellows/cli/application.py", line 268, in read_attribute
    except bellows.zigbee.exceptions.DeliveryError as e:
AttributeError: module 'bellows.zigbee' has no attribute 'exceptions'
```

> but this works only for devices which are connected to mains. battery devices normally sleep and send their values as events, they cannot be polled.

Do you know how I could listen to that event in a Python script?

JimmyTheNerd, 2019-08-31:
> You can poll them if you keep them awake... it’s just not ideal.

Thank you, but I'd like not to drain the battery. :) Do you know how to catch the event of a value change with Python?

martin3000, 2019-08-31:
you used cluster 1794 (0x702) which is the metering cluster for plugs. You have to adapt the cluster number. See http://www.zigbee.org/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf

JimmyTheNerd, 2019-08-31:
> you used cluster 1794 (0x702) which is the metering cluster for plugs. You have to adapt the cluster number. See http://www.zigbee.org/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf

Thanks and sorry for my lack of knowledge, but I think I adjusted the cluster number correctly.

```
belows devices
``` 
Gave me the following information:

```
Device:
  NWK: 0xf6c7
  IEEE: 00:0d:6f:00:0d:69:52:54
  Endpoints:
    1: profile=0x104, device_type=DeviceType.IAS_ZONE
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        Poll Control (32)
        Temperature Measurement (1026)
        Relative Humidity Measurement (1029)
        Occupancy Sensing (1030)
        IAS Zone (1280)
```

I tried to read the temperature (1026) of endpoint 1 via the following command:
```
bellows zcl 00:0d:6f:00:0d:69:52:54 1 1026 read-attribute 0
```

Afterwards I got the Python error pasted earlier.

Adminiuga, 2019-08-31:
1. The error tells you that read attribute request did not reach the device. Don't test your code with xiaomi devices as there are many unpleasant surprises with xiaomi
2. Apparently there's a bug in bellows cli, but your issue is still delivery failure
3. To listen for attribute value change in python:
  - bind the device
  - configure attribute reporting for the required cluster and attribute (not all attributes support reporting)
  - attach a listener to cluster with `cluster_istance.add_listener(your_listener_instance)` it would call `attribute_updated()` method on your listener.
Check zha component source from around Feb 2018 as it was much simpler to understand.

JimmyTheNerd, 2019-09-01:
Thanks for your advice, @Adminiuga . I had a look at the older ZHA components (Feb/Mar 2018) and tried to follow your instructions.

The following script starts the controller, lists all it's devices and their clusters, attaches the listener, but `attribute_updated()` is never called.

I added some comment within the code to explain a bit more.

**What am I missing?** 
Thanks again!

```
import asyncio
import bellows.ezsp
from bellows.zigbee.application import ControllerApplication

DATABASE = '/home/jim/.config/bellows/zigbee.db'
DEVICE_PORT = '/dev/ttyUSB1'
BAUDRATE = 57600

class ClusterListener:
    """
    Listens to events of each cluster
    """
    def __init__(self):
        # I'm called.
        print('ClusterListener init')

    def attribute_updated(self, attribute, value):
        # I should be called when a cluster's attribute value changes,
        # but I'm not called.
        print('ClusterListener attribute_updated:')
        print(str(attribute))
        print(str(value))

async def handle_zwave():
    radio = bellows.ezsp.EZSP()
    await radio.connect(DEVICE_PORT, BAUDRATE)

    controller = ControllerApplication(radio, DATABASE)
    await controller.startup(auto_form=True)

    for device in controller.devices.values():
        print('Device '+str(device.ieee)) # works
        for endpoint_id, endpoint in device.endpoints.items():
            if endpoint_id > 0:
                print('Endpoint {0} ({1}, {2})'.format(endpoint_id, endpoint.manufacturer, endpoint.model)) # works
                for cluster_id, cluster in endpoint.in_clusters.items():
                    print('Cluster {0} ({1})'.format(cluster_id, cluster.name)) # works
                    # Adding the ClusterListener to listen to future value changes:
                    cluster_listener = ClusterListener()
                    cluster.add_listener(cluster_listener)

def main():
    loop = asyncio.get_event_loop()

    # Start the controller, add the listeners one time.
    loop.run_until_complete(handle_zwave())

    # Listen to the clusters changes forever
    loop.run_forever()

if __name__ == '__main__':
    main()
```

Adminiuga, 2019-09-01:
You need to call `cluster.bind()` and `cluster.configure_reporting()` (check the args for reporting, should be attr_id, min_report_interval_seconds, max_report_interval_seconds, reportable_chang) but only once, as devices store that configuration in NVRAM
You only want to do this for clusters and attributes you want to receive updates for, as binding tables size is limited on devices

JimmyTheNerd, 2019-09-01:
Thanks @Adminiuga. I had a look at the implementation of HA and the zigpy tests, but `cluster.bind()` unfortunately fails with `Message send failure: EmberStatus.DELIVERY_FAILED`.

I interpret the error as: the device is not available, that's why the delivery of the bind command failed. It's the same behavior after waking the device up.

Here's the added code (I updated the post above as well to have it in context):
```
if cluster_id == 1026:  # bind only Temperature Measurement
    try:
        await cluster.bind()  # <- the script breaks here
        await cluster.configure_reporting(0, 0, 600, 1)
        cluster_listener = ClusterListener()
        cluster.add_listener(cluster_listener)
    except Exception as e:
        print(str(e)) # -> Message send failure: EmberStatus.DELIVERY_FAILED
```

1. Is the code correct?
2. What could cause the failed delivery?

I appreciate your help.

martin3000, 2019-09-01:
While binding, the device has also to be awake. You can achieve this by pressing the button every second or two.

closed 2020-04-29 by Adminiuga