zigpy/zigpy #207: zigpy_zigate : devices are lost after HA restart

doudz, 2019-08-31:
I'm currently fixing various problem on zigpy_zigate but there's one I can't figure out.
Devices are lost after HA restart, what is missing ?

Devices are still paired to zigate but not recognize anymore by zha, why ?

Adminiuga, 2019-08-31:
Zigpy keeps application state (devices) in zigbee.db sqlite database.  Make sure it is not deleted or lost between the restarts.

doudz, 2019-09-02:
zigbee.db is present
Missing entities are visible in entities registry
[image]
And visible in Integrations
[image]

But not in devices management
[image]

doudz, 2019-09-02:
Looking into zigbee.db I see that all tables are empty...
is there something missing in my zigpy_zigate to be able to save devices ?

Adminiuga, 2019-09-03:
yep, you need to call `ControllerApplication.handle_join(nwk_of_new_device, ieee_of_new_device, parent_is_ignored)` For really new devices it is going to schedule device initialization and upon successful initialization it is going to store the device state into `zigbee.db`

doudz, 2019-09-03:
handle_join is called, so maybe there's something wrong during initialization

Adminiuga, 2019-09-03:
Most likely. Hard to say without zigpy debug logs. We did make a change in
Zigpy requiring all endpoints to be initialized and successfuly reading
`manufacturer` and `model` attributes from at least one of the endpoints,
otherwise the device is dropped from the database.
When a brand new device joins, to be saved into `zigbee.db` the following
requests must succeed
- ZDO Node Descriptor Request
- Active Endpoint Req
- for each endpointd from `Active_Ep_Request`
  - simple descriptor request
- at least for one of the endpoints:
`BasicCluster.read_attributes(['manufacturer', 'model'])`

pipiche38, 2019-09-03:
This is an unfair assumption to get Model from devices. 
There are some devices which do not provide any Model name ( Cluster 0x0000 and Attribute 0x0005 )

Adminiuga, 2019-09-03:
HRM, I personally haven't seen one yet. Do you know which devices do not provide model?
However I've just checked the code and while we do try to read model&manufacturer, we skip the errors for those, so successfully reading model/manufacturer is not a requirement for now.

pipiche38, 2019-09-03:
Profalux a brand which manufacture Blinds and Venetians do not provide Model and if you query attribute 0x0005 if I’m not mistaken you don’t get any answer 

I’m seen others but do not remember.

For me what is mandatory in the zigbee world is ProfileId and DeviceId

Adminiuga, 2019-09-03:
Basic is a mandatory cluster in HA 1.2 profile. Not replying to attribute read request is not acceptable. Should at least reply with `unsupported attribute` ZCL status  (

doudz, 2019-09-08:
This is the log when pairing a Xiaomi square button
[image]

```
2019-09-08 21:21:37 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Received {'type': 'zha/devices/permit', 'id': 30}
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x49 b'fffc3c00'
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x00I\x00\x04r\xff\xfc<\x00'
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x10I\x02\x10\x02\x14r\xff\xfc<\x02\x10'
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800000056d00a1004900
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00a10049' LQI:0
2019-09-08 21:21:37 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 161, 73, b'']
2019-09-08 21:21:37 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'result', 'success': True, 'result': None}
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000cc4adef00158d00016c47f58000
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'adef00158d00016c47f580' LQI:0
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0xadef, 00:15:8d:00:01:6c:47:f5, 128]
2019-09-08 21:21:45 INFO (MainThread) [zigpy.application] Device 0xadef (00:15:8d:00:01:6c:47:f5) joined the network
2019-09-08 21:21:45 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'first_occured': 1567970505.9174163, 'timestamp': 1567970505.9174163, 'level': 'INFO', 'message': 'Device 0xadef (00:15:8d:00:01:6c:47:f5) joined the network', 'exception': '', 'root_cause': None, 'source': '/srv/homeassistant/lib/python3.7/site-packages/zigpy/application.py', 'count': 1}}}
2019-09-08 21:21:45 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'event', 'event': {'type': 'device_joined', 'nwk': 0xadef, 'ieee': '00:15:8d:00:01:6c:47:f5'}}
2019-09-08 21:21:45 INFO (MainThread) [zigpy.device] [0xadef] Requesting 'Node Descriptor'
Tries remaining: 2
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xadef, 0, 0x0002, 0, 0, 7, b'\x07\xef\xad', True, 10)
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02adef00000002000000000307efad'
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f>\x02\xad\xef\x00\x00\x00\x02\x00\x00\x00\x00\x03\x07\xef\xad'
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f>\x02\x12\xad\xef\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x02\x17\xef\xad'
2019-09-08 21:21:45 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'first_occured': 1567970505.9219959, 'timestamp': 1567970505.9219959, 'level': 'INFO', 'message': "[0xadef] Requesting 'Node Descriptor'", 'exception': '', 'root_cause': None, 'source': '/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py', 'count': 1}}}
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000005a50015053000
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'00150530' LQI:0
2019-09-08 21:21:45 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 21, 1328, b'']
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001af90000000013000002adef02fffd65efadf5476c01008d150080d2
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000000013000002adef02fffd65efadf5476c01008d150080' LQI:210
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0xadef>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'e\xef\xad\xf5Gl\x01\x00\x8d\x15\x00\x80']
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy.zdo] [0xadef:zdo] ZDO request 0x0013: [0xadef, 00:15:8d:00:01:6c:47:f5, 128]
2019-09-08 21:21:46 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'first_occured': 1567970506.173294, 'timestamp': 1567970506.173294, 'level': 'DEBUG', 'message': '[0xadef:zdo] ZDO request 0x0013: [0xadef, 00:15:8d:00:01:6c:47:f5, 128]', 'exception': '', 'root_cause': None, 'source': '/srv/homeassistant/lib/python3.7/site-packages/zigpy/zdo/__init__.py', 'count': 1}}}
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002002f3a0001040000010102adef02000018000a050042166c756d692e73656e736f725f7377697463682e61713201002003d5
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0001040000010102adef02000018000a050042166c756d692e73656e736f725f7377697463682e61713201002003' LQI:213
2019-09-08 21:21:46 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 0, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0xadef>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x00\n\x05\x00B\x16lumi.sensor_switch.aq2\x01\x00 \x03']
[... cut ...]
2019-09-08 21:22:04 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2754869328] Sending {'id': 30, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'first_occured': 1567970524.2976182, 'timestamp': 1567970524.2976182, 'level': 'DEBUG', 'message': '[0xadef](lumi.sensor_switch.aq2): completed initialization', 'exception': '', 'root_cause': None, 'source': 'components/zha/core/device.py', 'count': 1}}}
2019-09-08 21:22:08 DEBUG (SyncWorker_0) [homeassistant.helpers.storage] Writing data for core.device_registry
2019-09-08 21:22:12 DEBUG (SyncWorker_12) [homeassistant.helpers.storage] Writing data for zha.storage

```

martin3000, 2019-09-08:
I had a similar problem and I fixed it in homeassistant/helpers/storage.py by checking if data is "None":

```
    async def _async_load(self):
        """Load the data."""
        # Check if we have a pending write
        if self._data is not None:
            data = self._data

            # If we didn't generate data yet, do it now.
            if 'data_func' in data:
                data['data'] = data.pop('data_func')()
        else:
            data = await self.hass.async_add_executor_job(
                json_util.load_json, self.path)

```
            if data == {} or **data==None**:
`
                return None

`

Adminiuga, 2019-09-09:
HRM, looks like the dvice discovery goes through, but I'm concerned about `error calling listeneres`
```
2019-09-08 21:21:57 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200355b0001040000010102adef020000180a0104000042044c554d4905000042166c756d692e73656e736f725f7377697463682e617132d2
2019-09-08 21:21:57 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0001040000010102adef020000180a0104000042044c554d4905000042166c756d692e73656e736f725f7377697463682e617132' LQI:210
2019-09-08 21:21:57 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 0, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0xadef>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\n\x01\x04\x00\x00B\x04LUMI\x05\x00\x00B\x16lumi.sensor_switch.aq2']
2019-09-08 21:21:57 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
2019-09-08 21:21:57 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.
2019-09-08 21:21:57 DEBUG (MainThread) [zigpy.endpoint] [0xadef:1] Manufacturer: LUMI
2019-09-08 21:21:57 DEBUG (MainThread) [zigpy.endpoint] [0xadef:1] Model: lumi.sensor_switch.aq2
2019-09-08 21:21:57 WARNING (MainThread) [zigpy.util] Error calling listener.raw_device_initialized: Error binding parameter 0 - probably unsupported type.
```

That would explain why the database stays empty. Have no ideas why this error happens. Can you put some breakpoints and check what listeneres throws the error and what it is send to `listener_event()` method

doudz, 2019-09-09:
And what about :+1: 

```
2019-09-08 21:21:46 ERROR (MainThread) [zigpy_zigate.api] Exception running handler
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_zigate/api.py", line 151, in handle_callback
    handler(*args)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_zigate/zigbee/application.py", line 112, in zigate_callback_handler
    response[2], response[-1])
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/application.py", line 82, in deserialize
    return sender.deserialize(endpoint_id, cluster_id, data)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/device.py", line 155, in deserialize
    return self.endpoints[endpoint_id].deserialize(cluster_id, data)
KeyError: 1
```

Adminiuga, 2019-09-09:
That happens with Xiaomi devices because they send attribute updates prior any binding/attribe_reporting configured and in that point in time zigpy hasn't completed device discovery yet, so  it doesn't know the device has `endpoint_id==1`  and it fails. We probably should catch that exception though.

doudz, 2019-09-09:
> That happens with Xiaomi devices because they send attribute updates prior any binding/attribe_reporting configured and in that point in time zigpy hasn't completed device discovery yet, so it doesn't know the device has `endpoint_id==1` and it fails. We probably should catch that exception though.

yes, by ignoring the update if endpoint_id is not in self.endpoints
maybe log a warning

doudz, 2019-09-09:
`2019-09-08 21:22:01 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: Error binding parameter 0 - probably unsupported type.`
The error message comes from sqlite, I have to found why

Adminiuga, 2019-09-09:
Check what do you send for `ieee` address.

doudz, 2019-09-10:
> Check what do you send for `ieee` address.

that's it !
currently ieee is custom EUI64 instance and this is the problem because it is not registered by sqlite3.register_adapter

The problem is zigpy.types.EUI64 is buggy (for me) since it doesn't serialize/deserialiaze
example :
```
>>> ieee = zigpy.types.EUI64(b'\x01\x23\x45\x67\x89\xab\xcd\xef')
>>> ieee
01:23:45:67:89:ab:cd:ef
>>> ieee.serialize()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/sramage/.local/lib/python3.7/site-packages/zigpy/types/named.py", line 26, in serialize
    return b''.join([i.serialize() for i in self[::-1]])
  File "/home/sramage/.local/lib/python3.7/site-packages/zigpy/types/named.py", line 26, in <listcomp>
    return b''.join([i.serialize() for i in self[::-1]])
AttributeError: 'int' object has no attribute 'serialize'
```

But the following works

```
>>> types.zigpy.types.EUI64.deserialize(b'\x01\x23\x45\x67\x89\xab\xcd\xef')
(ef:cd:ab:89:67:45:23:01, b'')
>>> types.zigpy.types.EUI64.deserialize(b'\x01\x23\x45\x67\x89\xab\xcd\xef')[0]
ef:cd:ab:89:67:45:23:01
>>> types.zigpy.types.EUI64.deserialize(b'\x01\x23\x45\x67\x89\xab\xcd\xef')[0].serialize()
b'\x01#Eg\x89\xab\xcd\xef'
```

So the problem is on EUI64 `__init__`

Adminiuga, 2019-09-10:
Most zigpy types do not like to be initialized from the class
instantiation. So to intialize ieee, the typical flow is to `ieee, data =
zigpy.types.EUI64.deserialize(your_bytes_data)` or `ieee =
zigpy.types.EUI64.deserialize(your_bytes_data)[0]`

doudz, 2019-09-10:
ok thank you.
Just an other question about EUI64 
deserialize b'\x01\x23\x45\x67\x89\xab\xcd\xef' gives ef:cd:ab:89:67:45:23:01 but shouldn't it be 01:23:45:67:89:ab:cd:ef ?

Adminiuga, 2019-09-10:
> deserialize b'\x01\x23\x45\x67\x89\xab\xcd\xef' gives ef💿ab:89:67:45:23:01
> but shouldn't it be 01:23:45:67:89🆎cd:ef ?

Depends on the radio endianess. Zigbee I think is little endian that's why
EUI64 deserializes like you see. If zigate is different, then you would
need to implement a custom class. check zigpy-deconz it does exactly that
in its own eui64

doudz, 2019-09-10:
But if I create a custom EUI64, then sqlite3 won't be able to store the value... (like above)

doudz, 2019-09-10:
zigpy-deconz doesn't use it's own EUI64, it reverse byte order before deserializing

Adminiuga, 2019-09-10:
hrm, maybe we removed it at some point. 
in what form do you get ieee in https://github.com/doudz/zigpy-zigate/blob/76380ebb491be193e1b2a47b6692892255b668a0/zigpy_zigate/zigbee/application.py#L92 ?
I think all you need to is to pass `zigpy.types.EUI64` instance to `handle_join()` to get this sorted out. Depending how you get your IEEE,  a simple `ieee = zigpy.types.EUI64([zigpy.types.uint8_t(a) for a in response[1]])` could work?

Adminiuga, 2019-09-10:
in other words, just pass a list of `uint8_t` to EUI64 init, that would take care of serialization error.

doudz, 2019-09-10:
I confirm that using zigpy.types.EUI64 instead of my custom EUI64 solve the problem
Devices are now saved and restored accross HA restart :+1:

closed 2019-09-10 by doudz