zigpy/zigpy #237: Osram Smart+ Plug becoming unavailable intermittently

ErnstEeldert, 2019-09-29:
I'm running a fairly large (~30 lights, ~15 sensors, ~10 switches) ha+zigpy-deconz network, and overall things seem to be pretty stable (better than deconz itself).

I have two Osram Smart+ plugs. One is working fine, the other one (which is a lot closer to the controller antenna) is becoming unavailable intermittently. The only way to get it back online is by un/re-plugging it.

The last time it became unavailable was 2:08AM this morning, where I suppose there was not a lot going on on the network.

Here are the logs, hopefully it provides some insight. I'm running HA inside docker, and can swap to other branches if that aids troubleshooting this.

```
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e20000700aa20
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 32]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1720000800010001
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1720002a0023002202fdff0002c3e900000013000c0000c3e9e3640e00002618848e00afffa3000102ce
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xe9c3>, 0, 0, 19, b'\x00\xc3\xe9\xe3d\x0e\x00\x00&\x18\x84\x8e', 0, 175, 255, 163, 0, 1, 2, -50]
home-assistant   | 2019-09-29 11:36:41 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0xe9c3, 84:18:26:00:00:0e:64:e3
home-assistant   | 2019-09-29 11:36:41 INFO (MainThread) [zigpy.application] Device 0xe9c3 (84:18:26:00:00:0e:64:e3) joined the network
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 84:18:26:00:00:0e:64:e3
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy.zdo] [0xe9c3:zdo] ZDO request 0x0013: [0xe9c3, 84:18:26:00:00:0e:64:e3, 142]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xe9c3>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'00c3e9e3640e00002618848e'
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 44, data: b'002c000000'
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 44, 0, <DeconzAddressEndpoint address_mode=2 address=59843 endpoint=3>, 260, 6, 1, b'\x00,\x00\x00\x00', 2, 0)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1221001b0014002c0002c3e90304010600010500002c0000000200
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 45, data: b'002d000700'
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 45, 0, <DeconzAddressEndpoint address_mode=2 address=59843 endpoint=3>, 260, 0, 1, b'\x00-\x00\x07\x00', 2, 0)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1222001b0014002d0002c3e90304010000010500002d0007000200
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Zigbee request with id 46, data: b'002e000b05'
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 46, 0, <DeconzAddressEndpoint address_mode=2 address=59843 endpoint=3>, 260, 2820, 1, b'\x00.\x00\x0b\x05', 2, 0)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1223001b0014002e0002c3e9030401040b010500002e000b050200
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12210009000200222c
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 44]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e22000700a620
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 32]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04240007000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12220009000200262d
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 38, 45]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e23000700a620
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 32]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12230009000200262e
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 38, 46]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e24000700a620
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 32]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04240013000c00262c02c3e903010000000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 44: 00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04250007000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2c 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04250013000c00262d02c3e903010000000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 45: 00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04260007000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04260013000c00222e02c3e903010000000000
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 46: 00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
^CERROR: Aborting.
pi@charlie:~/homeassistant$
pi@charlie:~/homeassistant$ docker-compose logs -f --tail=100000 | grep 0xe9c3
home-assistant   | 2019-09-28 20:37:09 DEBUG (MainThread) [zigpy_deconz.api] Received simplified beacon frame: source=0xe9c3, pan_id=0x274f, channel=11, flags=0x00, update_id=0x01
home-assistant   | 2019-09-28 20:37:11 DEBUG (MainThread) [zigpy_deconz.api] Received simplified beacon frame: source=0xe9c3, pan_id=0x274f, channel=11, flags=0x00, update_id=0x01
home-assistant   | 2019-09-28 21:28:45 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xe7 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
home-assistant   | 2019-09-28 21:28:45 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xe8 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
[... cut ...]
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
home-assistant   | 2019-09-29 11:36:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
home-assistant   | 2019-09-29 11:36:47 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x30 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe9c3 endpoint=3>, status: 0x00
```

Adminiuga, 2019-09-29:
Please move this to zigpy-deconz.

Aside from that, next time when it becomes unavailable, go to zha config panel, pick the device and try reading model attribute of the basic cluster and provide full debug log. Grepping by nwk could omit some details. You could DM me on discord.

Is this conbee 1 or 2. From logs it looks like transmission is acked by device but I'm not seeing the response back. I also had Osram Smart plug becoming unresponsive on EZSP based radios but haven't happened for a while now.

Are you using Osram bulbs? On discord users were complaining about those, but I could not reproduce.

ErnstEeldert, 2019-09-29:
Thanks. Continuing [here](https://github.com/zigpy/zigpy-deconz/issues/72).

closed 2019-09-29 by ErnstEeldert