zigpy/zigpy #244: Smartthings motionv5 not reporting battery, and continuously reporting Temperature(draining Energy)

meauxt, 2019-10-15:
Smartthing not reporting battery  
Smartthing reporting temperature very often and during the battery much faster 
am using zigpy as part of hassio integration so am not even seeing an option to enable debugging to grab more information

meauxt, 2019-10-18:
Every few minutes (5 minutes at most) 
And for battery i have 7 sensor (smart thing motion) haven’t report any battery for last 45 days..

Adminiuga, 2019-10-18:
does it report the same temperature or does temperature change from the previous reading and how much?

meauxt, 2019-10-18:
The temperature changes different by 0.x Fahrenheit

Adminiuga, 2019-10-18:
Zigbee temp sensor reports temperature in hundredth of C. You still haven't provided any logs to support your claim nor you provided the details about your setup. What kind of radio? what hardware. 
Zigpy does not run on hassio. Home assistant runs on hassio and ZHA component of home assistant is using zigpy. ZHA configures the temperature sensors to send updates on 0.5C temperature change. If the device is sending updates more often, then it is not the problem with zigpy.
Remove the device from home assistant and re-pair it again. If the device ignores the requested report change of 0.5C then there's nothing could be done on ZHA  side.
I have a samjin water sensor it is seems to be ignoring the 0.5C temperature delta for reporting and sends updates for every 0.1C change. It was running over 1.5 year and is still on the original battery, so if you battery is drained faster than that, then it is not because of the temperature updates.

meauxt, 2019-10-20:
I did repair all on my devices and now am seeing more normal update... Thank you Adminiuga! appreciate your help!

meauxt, 2019-10-20:
i will watch if i will get any battery update also now... hopefully this will be solved as well :+1:

closed 2019-10-20 by meauxt