zigpy/zigpy #250: KeyError is raised if device is not found in database

martin3000, 2019-10-26:
After removin a zigbee device in home assistant, it seems that the device has not been completely removed. After a restart of home assistant I get this error:
```
2019-10-26 00:08:24 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/ttyUSB0 for zha
Traceback (most recent call last):
  File "/home/nuc/.local/lib/python3.7/site-packages/homeassistant/config_entries.py", line 273, in async_setup
    hass, self)
  File "/home/nuc/.local/lib/python3.7/site-packages/homeassistant/components/zha/__init__.py", line 94, in async_setup_entry
    await zha_gateway.async_initialize(config_entry)
  File "/home/nuc/.local/lib/python3.7/site-packages/homeassistant/components/zha/core/gateway.py", line 87, in async_initialize
    radio, database)
  File "/home/nuc/.homeassistant/deps/lib/python3.7/site-packages/bellows/zigbee/application.py", line 34, in __init__
    super().__init__(database_file=database_file)
  File "/home/nuc/.homeassistant/deps/lib/python3.7/site-packages/zigpy/application.py", line 26, in __init__
    self._dblistener.load()
  File "/home/nuc/.homeassistant/deps/lib/python3.7/site-packages/zigpy/appdb.py", line 249, in load
    _load_attributes()
  File "/home/nuc/.homeassistant/deps/lib/python3.7/site-packages/zigpy/appdb.py", line 218, in _load_attributes
    dev = self._application.get_device(ieee)
  File "/home/nuc/.homeassistant/deps/lib/python3.7/site-packages/zigpy/application.py", line 154, in get_device
    return self.devices[ieee]
KeyError: 00:15:8d:00:03:f4:12:b3

```

In application.py getDevice there is a "raise KeyError" if a device is not found. I think this error is not a good solution because _load_attributes in appdb.py does not catch the exception.

Adminiuga, 2019-10-26:
Lemme guess, that was a xiaomi device, wasn't it?
What version of zigpy?
What radio?
Can you do `sqlite zigbee.db .dump | grep mac_addr_in_question` to see what is left in db.

We also need to see the logs of what happened when the device was removed, so can you reliably reproduce it?

Adminiuga, 2019-10-26:
Unless you killed zigpy after it removed the device but prior it deleted from db I don't see how that could happen

martin3000, 2019-10-26:
Sorry that I did not say it clear. The issue is about this code:
```
def get_device(self, ieee=None, nwk=None):
        if ieee is not None:
            return self.devices[ieee]

        for dev in self.devices.values():
            # TODO: Make this not terrible
            if dev.nwk == nwk:
                return dev

        raise KeyError
```
It is not good to raise a KeyError here because this error is not catched by some callers.
For example in this code in appdb.py there is no "try":
```
        def _load_attributes():
            for (ieee, endpoint_id, cluster, attrid, value) in self._scan("attributes"):
                dev = self._application.get_device(ieee)

```

Adminiuga, 2019-10-26:
Nah, I'm gonna disagree to that. It is right to raise key error there. The
problem is that db is in inconsistent state and I'd rather find the root
cause which leads to inconsistent db state.

And please provide all other details. I don't want to waste your time if
this was already fixed in newer version.

Asnof now I'm not aware of the events chain which leads to inconsistent db
state, unless you kill zigpy during db cleanup.

Adminiuga, 2020-04-02:
Please try with the latest  version. There shouldn't be any device remnants in the database on unsuccessful join.

closed 2020-04-02 by Adminiuga

tomtoump, 2021-01-11:
I am running the latest version of Home Assistant. I got the same error for 00:12:4b:.... 
Using the command above, I found the following entries in `zigbee.db`

```sql
INSERT INTO attributes VALUES('00:12:4b:...',1,1029,0,7361);
INSERT INTO neighbors VALUES('7c:b0:3e:...','00:21:2e:...','00:12:4b:...',812,66,1,1,16);
INSERT INTO neighbors VALUES('00:0d:6f:...','00:21:2e:...,'00:12:4b:...',812,18,0,2,145);
```

I manually deleted them and the error is gone.