zigpy/zigpy #257: Hue dimmer RWL021 short dim_down does not work

rchovan, 2019-11-11:
Hi, 
in HA I have automation for short press dim_up and dim_down, 
```
- id: '1573500068408'
  alias: Hostovska_Stmievac_Svetlo_Dimm
  description: ''
  trigger:
  - device_id: 367e9b2956754bd8ac16822aaf398c09
    domain: zha
    platform: device
    subtype: dim_down
    type: remote_button_short_press
  condition: []
  action:
  - service: script.hostovska_dimm
- id: '1573500922021'
  alias: Hostovska_Stmievac_Svetlo_bright
  description: ''
  trigger:
  - device_id: 367e9b2956754bd8ac16822aaf398c09
    domain: zha
    platform: device
    subtype: dim_up
    type: remote_button_short_press
  condition: []
  action:
  - service: script.hostovska_bright
```
Dim_up is working correctly, but Dim_down does not work. 

I'm logging brightness level to logbook, but automation for dim_down was not executed:
```
9:38 PM Time changed to 21:38
9:38 PM Time (UTC) changed to 20:38
9:38 PM Time & Date changed to 21:38, 2019-11-11
9:38 PM Date & Time ISO changed to 2019-11-11T21:38:00
9:38 PM Date & Time changed to 2019-11-11, 21:38
9:36 PM Internet Time changed to @900
9:36 PM hostovska_dimm Brightning 119
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 109
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 99
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 89
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 79
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 69
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
9:36 PM hostovska_dimm Brightning 59
9:36 PM hostovska_bright started
9:36 PM Hostovska_Stmievac_Svetlo_bright has been triggered
```

This is debug log from zigpy. 
```
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ec1000700aa00
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17c1000800010001
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17c10025001e00220200000102262701040108000700014a02001e090000af67aecc0000af
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2726>, 1, 260, 8, b'\x01J\x02\x00\x1e\t\x00', 0, 175, 103, 174, 204, 0, 0, -81]
2019-11-11 21:36:05 DEBUG (MainThread) [zigpy.zcl] [0x2726:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=74 command_id=2>
[... 568 lines cut ...]
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x170a002a002300220200000102262702040119000c00015f01000b100901b645004200af5ff4cc0000ae
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2726>, 2, 260, 25, b'\x01_\x01\x00\x0b\x10\t\x01\xb6E\x00B', 0, 175, 95, 244, 204, 0, 0, -82]
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy.zcl] [0x2726:2:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=95 command_id=1>
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy.zcl] [0x2726:2:0x0019] ZCL request 0x0001: [0, 4107, 265, 1107314102, None]
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy.zcl] [0x2726:2:0x0019] OTA query_next_image handler for 'Philips RWL021': field_control=0, manufacture_id=4107, image_type=265, current_file_version=1107314102, hardware_version=None
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x2726>, ep: 2, profile: 0x0104, cluster_id: 0x0019, data: b'015f01000b100901b6450042'
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy.zcl] [0x2726:2:0x0019] No OTA image is available
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 95 under 25 request id, data: b'195f0298'
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (19, 25, 0, <DeconzAddressEndpoint address_mode=2 address=10022 endpoint=2>, 260, 25, 1, b'\x19_\x02\x98', 2, 0)
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x120b001a00130019000226270204011900010400195f02980200
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x120b00090002002219
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, 34, 25]
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e0c000700a600
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [166, 0]
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x040c0007000000
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x040c0013000c00221902262702010000000000
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 25: 00
2019-11-11 21:48:17 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x19 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x2726 endpoint=2>, status: 0x00
2019-11-11 21:48:43 DEBUG (MainThread) [zigpy_deconz.api] Command Command.write_parameter (5, <NetworkParameter.watchdog_ttl: 38>, b'\x10\x0e\x00\x00')
2019-11-11 21:48:43 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x0b0d000c00050026100e0000
2019-11-11 21:48:43 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0b0d000800010026
2019-11-11 21:48:43 DEBUG (MainThread) [zigpy_deconz.api] Write parameter watchdog_ttl: SUCCESS

```

dmulcahey, 2019-11-11:
This is fixed in the next version of HA

Adminiuga, 2019-11-11:
@dmulcahey Thank you.

Gonna close this one, since it should belong to HA issue.
@rchovan try beta when it is released in a few days.

closed 2019-11-11 by Adminiuga