zigpy/zigpy #258: Various profile support?

smdjeff, 2019-11-17:
Could anyone comment on the viability of supporting various profiles? EZSP isn't profile soecific, but the network join and form implemention here seems HA specific (especially security modes). I'd be curious to hear how ZLL or SEP could be supported.

Adminiuga, 2019-11-17:
Not at the moment. But PRs are welcome any time.
PS: what do you mean by zll support? ZLL bulbs can be joined and controlled, no?

smdjeff, 2019-11-17:
Between home assistant, bellows and zigpy... it's not clear that anything is actually calling the EZSP ZLL APIs. All the security modes seem to be HA specific. Where are the ZLL differences actually implemented?

smdjeff, 2019-11-17:
Wait a moment a clarification. Although bellows supports the join operations. Is it true that the higher layers: zigpy and home assistant only support forming HA networks as the ZC? And if that's true I suppose ZLL is only supported in that the ZLL bulbs have a fallback to HA mode when they're disconnected from their original network.

Adminiuga, 2019-11-18:
That's correct.

Gamester17, 2020-01-07:
Does anyone know if anyone has plans to extend zigpy to also fully Zigbee Light Link (ZigBee LL) application layer protocol or maybe even extend zigpy to support the full Zigbee 3.0 stack?

By the way, there is a quite good explanation meant for end-users to this dilemma of interoperability between different application layer protocols which causes headaches for end-users who don't know that there is a difference https://www.unifore.net/home-alarm-system/introduction-to-zigbee-3-0-zigbee-ha-protocol.html

Gamester17, 2020-09-14:
Ziggee Smart Energy Profile (SEP) and Zigbee Green Power (ZGP) is not supported yet as as as I know but the very latest bellows do support other Zigbee 3.0 profiles if coondinator firmware does

Hedda, 2021-02-15:
FYI, zoic21 have made progress with ZGP support for zigpy and is requesting regression testing of his new pull request in https://github.com/zigpy/zigpy/pull/656

> Can someone can test with this code if all it's ok ? No side effect. If it I think we can merge this first step (after I fix coveralls).

That is, he is asking people do functional and non-functional tests to ensure that his patch does not break any existing functions. If it does not break anything obvious in zigpy then it could possible be merged 'as-is' even if ZGP might be be fully production ready.

closed 2021-09-07 by github-actions[bot]