zigpy/zigpy #263: Ability to add specific radio commands

doudz, 2019-11-26:
It's not really the right place to put this, but it would be cool to able to add specific radio commands as  Home Assistant service

For example, ZiGate as a command to erase the internal eeprom, so I would like to have something like `zha.zigate_erase`
Another example, ZiGate firmware could be upgraded, so I would like to have `zha.zigate_upgrade_firmware`

Additionnally some radios may need special configuration options
For example, ZiGate has an onboard led and sometimes it's annoying, luckily there's a command to disable the led, but right now I can't call it using zigpy/zha
I would like to have a configuration option handled by zigpy-zigate like led: off/on 
and/or a service `zha.enable_led`

dmulcahey, 2019-11-26:
These would be web socket api’s services are only for things you would call from automations. That’s the rule of thumb. This should probably be moved to the home assistant GH. Thoughts @Adminiuga ?

Adminiuga, 2019-11-27:
My opinion: Unless the service can be handled/offered by all 4 radio types, it shouldn't be part of ZHA. Alternatively, make Zigate being able to run both as a module and as a script? same way `bellows` does CLI? of cause that would work only if LED settings are persistent across Zigate restarts.

In a short term, I'd implement it as a custom component which hooks into ZHA, eg https://github.com/zha-ng/zha-map thus only those who really need/want it can have it. 
In a long term: I really like being able to run `bellows` from command line and want to be able to run those commands using other radios as well. in other words something like `zigpy-cli` Wanted to be able to update Network key, radio channel, etc from command line, without introducing additional complexity into ZHA. ATM this is more of a wishlist, with time being a limiting factor.

dmulcahey, 2019-11-27:
I’d disagree on the ZHA part... I bet we can conditionally register the services based on radio. Is it that big of a deal to do it like that? I’d like to talk to the core devs to see if we can add functions like these to the coords device page. Does that make sense???

Adminiuga, 2019-11-27:
Actually, registering those conditionally make sense. Just be prepared to handle issues like: why there's no zigate_firmware_update service for EZSP :D

dmulcahey, 2019-11-27:
😂

Adminiuga, 2020-04-29:
So is it OK to close this?
**Resolution**: This is for ZHA component of HA Core. We can register zha services conditionally based on radio type. 
Perhaps should come up with a clean way to keep APIs in different files for different radios.

closed 2020-04-29 by Adminiuga