zigpy/zigpy #276: Donate money to existing Zigpy developers for them to buy CC2531 USB sticks?

Gamester17, 2020-01-13:
I was wondering if any existing Zigpy contributing developers are interested in receiving donations earmarked for you to buy CC2531 USB sticks in order to help sanyatuning with the development of the "zigpy-cc" module for Zigpy?

https://github.com/sanyatuning/zigpy-cc

Do not have much money to spare myself but I would personally be willing to make a small token donation of $10 US-dollars each to a maximum of three developers who already contributed to the Zigpy project (or its existing radio modules) in order for them to each buy one CC2531 USB stick with an external antenna to help with zigpy-cc module development.

That is, I am willing to personally donate $30 US-dollars via PayPal towards a radio module which supports CC2531 with Z-Stack firmware for Zigpy. Perhaps others would be willing to donate money to in order som even more developers con buy CC2531 USB sticks as well for the same reason?

@Adminiuga @damarco @puddly @AndreasBomholtz @dmulcahey @Yoda-x @rcloran Would anyone of you be willing to take a small token donation in order to buy CC2531 USB sticks to be able to help sanyatuning with the development of the zigpy-cc module?

Gamester17, 2020-01-13:
For more information please also see https://github.com/zigpy/zigpy/issues/85

puddly, 2020-01-14:
Thanks for the offer but I'm going to try out the CC26X2R1 dev kit instead. It should have the same interface as the CC253x so I'll try to upstream any fixes if I have time.

Gamester17, 2020-01-14:
Cool! Yes that SimpleLink CC26X2R1 LaunchPad Development Kit from Texas Instruments is probably an even better choice for the more advanced developers, though that dev kit is more expensive.

http://www.ti.com/tool/LAUNCHXL-CC26X2R1

For developers interested, it is also possible to still buy the little less expensive China copies of TI CC2530 development kit. You find those by searching for "CC2530 Eval Kit" on Aliexpress. Ex:

https://www.aliexpress.com/item/1885598072.html
https://www.aliexpress.com/item/1885625230.html

FYI, the Zigbee2mqtt community is also working on developing inexpensive USB sticks based on TI CC2538 (with integrated CC2592 amplifier), which has faster MCU and more RAM plus longer range than CC2530/CC2531. When such CC2538 USB sticks are more readily available they should both performance-wise and price-wise fall somewhere between CC2531 and CC2652R based products.

https://github.com/Koenkk/zigbee2mqtt/issues/1568

Tip to developers who want also to experiment with CC2538 too is to buy this dev board from China:

https://www.aliexpress.com/item/32836343416.html
https://www.aliexpress.com/item/32816483947.html

verjus, 2020-01-14:
I've just ordered this board (I can't confirm it works since I've not received it yet) but it looks like a good option too:
[CC2538 Dev Board](https://www.aliexpress.com/item/4000522313451.html?spm=a2g0s.9042311.0.0.721e4c4d4AzjR5)

Jtag programmer (based on experience from other members of the board): [Jlink](https://www.aliexpress.com/item/32833502027.html?spm=a2g0s.9042311.0.0.721e4c4d4AzjR5)

Gamester17, 2020-02-02:
FYI, "zig-a-zig-ah" ("zzh" for short) USB-stick form-factor dev board based on CC2652R / CC2652P has now been announced as an open-source hardware design by Electrolama (  @OmerK and friends ) under open source Solderpad Hardware Licence Version 2.0

https://github.com/electrolama/zig-a-zig-ah/

https://github.com/electrolama/docs/blob/master/docs/projects/zig-a-zig-ah.md

"Assembled versions of zig-a-zig-ah! will be made available on Tindie when it is ready"

https://electrolama.com/projects/zig-a-zig-ah

https://electrolama.com/_assets/zzh-RevA-Draft1-010220.pdf

https://twitter.com/OmerK

Gamester17, 2020-02-13:
FYI, looks like @sanyatuning has now submitted initial (“Beta”) support for Home Assistant integration:

https://github.com/home-assistant/home-assistant/pull/31621

Hedda, 2020-03-11:
@Adminiuga @damarco @AndreasBomholtz @dmulcahey @sanyatuning Checkout related https://github.com/electrolama/zig-a-zig-ah/issues/5 where @omerk of electrolama.com express a willingness to collaborate and could provide some zig-a-zig-ah adapter prototypes to a few zigpy and ZHA developers to help the development of TI CC2652 support. You can send him an e-mail if you are interested.

Hedda, 2020-03-25:
FYI, ITEAD is now for just $4 US-dollars selling a CC2531 USB-dongle preflashed with a coordinator firmware which ZNP is said to be plug-and-play compatible with Zigbee2mqtt out-of-the-box: 

- https://www.itead.cc/cc2531-usb-dongle.html

I'm sure that ITEAD if contact would be willing to donate a bunch of them to developers of well-known open-source projects like Home Assistant and its ZHA integration with Zigpy by affiliation. 

Regardless, might be good enough for most devs even if it is a model with only integrated antenna.

closed 2020-09-14 by Gamester17