zigpy/zigpy #288: Exception happens often

jrbenito, 2020-02-07:
Although system continues to work, I see a exceptions like:

```
2020-02-07 02:41:31 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 161, in request
    with self._pending.new(sequence) as req:
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 236, in new
    return Request(self, sequence)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 205, in __init__
    assert sequence not in pending
AssertionError
2020-02-07 02:41:33 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 161, in request
    with self._pending.new(sequence) as req:
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 236, in new
    return Request(self, sequence)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 205, in __init__
    assert sequence not in pending
AssertionError
2020-02-07 02:41:34 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 186, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0xae77:1:0x0019]: Message send failure
2020-02-07 02:41:34 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 186, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0xae77:1:0x0019]: Message send failure

```

This happens from time to time but I could no figure out why yet.

Hedda, 2020-04-02:
A lot has changed in zigpy, ZHA, and in Home Assistant Core during the last year so is this still an issue?

Regardless you should probably post this in Home Assistant's ZHA forum: https://community.home-assistant.io/c/configuration/zha/48

Adminiuga, 2020-04-02:
fixed by https://github.com/zigpy/zigpy/pull/312

But those errors are typical tale tell signs of a battery operated device which can communicate with coordinator, but coordinator can't communicate with it. Usually happens when it changes its parent device, but the old parent does not remove it from the neighbors table (was offline)

closed 2020-04-02 by Adminiuga