zigpy/zigpy #293: Forming a new ZigBee Network

junsaw, 2020-02-09:
bellows --device /dev/ttyUSB1 form
Usage: bellows form [OPTIONS]
Try "bellows form --help" for help.

Error: Invalid value for "-D" / "--database": File "/root/.config/bellows/app.db" does not exist.

Adminiuga, 2020-02-09:
This belongs to `bellows`. Create `/root/.config/bellows/` folder and try again.

closed 2020-02-09 by Adminiuga

junsaw, 2020-02-13:
I tried that and its still giving the same error. 

ls -al /root/.config/
bellows/ procps/ 

bellows ^C
root@junior-raspblnx01-edge:~ # bellows --device /dev/ttyUSB1 devices
Usage: bellows devices [OPTIONS]
Try "bellows devices --help" for help.

Error: Invalid value for "-D" / "--database": File "/root/.config/bellows/app.db" does not exist.

Adminiuga, 2020-02-13:
form the network first.