zigpy/zigpy #297: [REQUEST] Add network support for Zigbee adapters

Hedda, 2020-02-17:
This is a feature request asking for zigpy or a dependencycy library to support making a serial connection to remote connected Zigbee adapters over your local-network using TCP/IP.

Adding this feature to zigpy should for allow Home Assistant Core with ZHA running in HASSIO to remotely utilize a example HUSBZB-1 adapter connected to a Raspberry Pi Zero W located someone else on your local-network, including Wi-Fi. 

Having this feature integrated would be very usefull to HASSIO users who want to have their Zigbee coordinator adapter located somewhere else in a other area of their house which is relativly far away from their HASSIO installation.

If I understand zigpy dependecies correctly guess for the problem I want to solve here is technically really asking for zigpy, each of its radio libraries, or a helper library to have built-in support to connect to a remote TTY serial device using for example ser2net?

As a proof-of-concept pyRFXtrx, which has similar concept as zigpy but for RFXCom RF devices, recently added integrated network support based on ser2net for used with Home Assistant.

https://github.com/Danielhiversen/pyRFXtrx/pull/85

Something similar was before also submitted as a feature requested by @rbflurry for bellows radio library in https://github.com/zigpy/bellows/issues/183

That idea suggestion is based on the fact that Zigbee2mqtt can remotley connect to a Zigbee serial device on a remote location over the network using ser2net which allows to connect to a serial port over TCP/IP.

- https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html
 - https://www.jpaul.me/2019/01/how-to-build-a-raspberry-pi-serial-console-server-with-ser2net/

Zigbee2mqtt remote adapter concept involves using ser2net to make the serial device available over the network

 ##Zigbee
 3334:raw:0:/dev/ttyUSB1:57600 8DATABITS NONE 1STOPBIT

Typically using socat you can make a "virtual" serial device to utilize this connection

 `pty,link=/dev/ttyUSB1,raw,user=0,group=0,mode=777 tcp:IP_address:3334`

However currently there is no way to get socat running in HASSIO.

I saw that network support was added to RFXtrx ([Danielhiversen/pyRFXtrx#85](https://github.com/Danielhiversen/pyRFXtrx/pull/85))

Could the same thing be added to this library?

For reference I am using the HUSBZB-1 with https://www.gl-inet.com/products/gl-ar150/

Hedda, 2020-02-19:
@foxy82 suggested in https://github.com/zigpy/bellows/issues/183 that one other solution could be to get socat added to HASSIO as he requested to them in https://github.com/home-assistant/docker/issues/92

As I understand it, integrated socat is what Zigbee2mqtt provides in its add-on for HASSIO, it is described from an end-users point of view here https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Gamester17, 2020-02-21:
Also see related discussion in Home Assistant Community Forum about others wanting this for ZHA  https://community.home-assistant.io/t/add-network-protocol-support-to-zigbee-and-zwave-integrations/136056

Hedda, 2020-03-06:
Might be better if ZHA integration for Home Assistant Core interfaces with zigpy over MQTT or similar?

zigpy with the Zigbee database could then run on a separate Raspberry Pi, similar to Zigbee2MQTT?

If that was possible when your Zigbee network would not go down if reboot Home Assistant Core?

Hedda, 2020-04-17:
It would be cool to see Tasmota extended to do the same for TI CC2530 as the ZiGate WiFi module
 
- https://www.cnx-software.com/2019/12/23/tasmota-open-source-firmware-now-supports-zigbee-connectivity/
  - https://tasmota.github.io/docs/Zigbee/

Itead just announced a similar Sonoff Zigbee Bridge / Gateway but based on an EZSP module:

- https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

Hedda, 2020-04-20:
FYI, Itead has just launched Sonoff ZBBridge as an inexpensive Sonoff Zigbee Bridge (which is somewhat based on a similar concept/idea as the product that they have in their Sonoff RF Bridge)

- https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

This new Sonoff ZBBridge (and their existing Sonoff RF Bridge) uses ESP8266 ("ESP8266EX") for WiFi and bridge/gateway/controller software, so the main difference is that they use different types of MCU for the RF radio module part.

- https://www.itead.cc/sonoff-zbbridge.html

According to the teardown on notenoughtech.com it sounds as if it the new Sonoff ZBBridge is based on Silicon Labs EFR32 (Mighty Gecko) for Zigbee 3.0 radio module support so uses EZSP ?

- https://notenoughtech.com/home-automation/sonoff/sonoff-zigbee-bridge-preview/

As you probably already know, Tasmota does something similar with Zigbee2Tasmota (Z2T) but by connecting an ESP8266 to a Texas Instrument CC2530 module instead however it too is using a serial communication protocol, see:

- https://tasmota.github.io/docs/Zigbee/

Hedda, 2020-04-20:
Again, a workaround for Home Assistant users could be to simply present a Zigbee radio module as a remote adapter to Home Assistant's ZHA (their Zigbee Home Automation integration component)?

- https://www.home-assistant.io/integrations/zha/

You could use something like ser2net to allow ZHA to remote connect to it via serial over network, like:

- https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Then you need a radio library for zigpy that works with the serial protocol used. For example with that Sonoff ZBBridge you would need a zigpy radio library for EZSP protocol used by Silicon Labs EFR32 

- https://github.com/zigpy/zigpy

Hopefully, their bellows radio library should already be compatible or could be made to be if updated

- https://github.com/zigpy/bellows

zigpy devs already do something similar to connect remotely with ESP8266 based "ZiGate Pack WiFi"

- https://zigate.fr/produit/zigate-pack-wifi-v1-3/
- https://github.com/fairecasoimeme/ZiGate/
- https://github.com/zigpy/zigpy-zigate

As proof-of-concept you could do this with a CC2530 remote to zigpy-cc from a ESP8266 using ser2net 

https://github.com/zigpy/zigpy-cc

Hedda, 2020-04-21:
@Equidamoid regarding this request and with newly announced Sonoff ZBBridge discussed in https://github.com/zigpy/zigpy/issues/385

Did you have some ideas for an "rpc bridge" that could add TPC/IP support to zigpy radio libraries?

Also mentioned was not yet realized "bridge" project by @vitiral only posted as an design idea:
- https://github.com/vitiral/bridge-rpc/blob/master/README.md
  - https://vitiral.github.io/bridge-rpc/#artifacts/REQ-0

Equidamoid, 2020-06-14:
@Hedda,

Sorry for disappearing.

Let me clarify my position here a little bit.

What I want to achieve is to be able to call zigbee methods (like "on with timed off" that I need the most) via some kind of RPC. Without them being hidden behind various abstraction layers.

What does exist now:
The closest thing is zigbee2mqtt (currently struggling to get it running on my old raspbian). Not sure how stable the network will be yet, but already see 2 downsides of it:
 - nodejs
 - only one zigbee dongle supported 

zigpy: it's a library, so the only "client" of it has to care about everything: automations, device management, gui, etc. And right now only one such client is homeassistant, so
 - writing complex automations is a bit of a pain, 
 - configuring the gui requires restart of everything and AFAIR some end devices really don't appreciate when the coordinator node stops responding
 - debugging is a bit harder due to overall comlplexity

Next, deCONZ. Only provides a "high level" api. Kind of works, but sometimes loses devices in my setup. Also, bound to only one kind of hardware.

The protocol.
I can definitely say that using this `bridge-rpc` in this context is not the best idea IMO. It's better to use one of the widely used protocols that already exist.

Even json over mqtt will be okay-ish.

Hedda, 2020-06-14:
@Equidamoid then I think you would be interested in https://github.com/arendst/Tasmota/issues/8583 as the Tasmota developer working on a compatible TCP/IP serial bridge for Tasmota so that third-party developerrs can access the Zigbee chip inside the Sonoff Zigbee Bridge over the network via serial

Hedda, 2020-06-22:
FYI; @s-hadinger has now also added a TCP/IP serial bridge to Tasmota firmware for remote access:

- https://github.com/arendst/Tasmota/pull/8702

> ## Description:
> Add a new feature: Serial to TCP bridge (similar to esp-link). This allows to remotely communicate to a MCU through ESP8266. Needs `#define USE_TCP_BRIDGE`
> 
> It adds 2 GPIO types: `TCP Tx (208)` and `TCP Rx (209)` and can work with hardware or software serial.
> 
> Commands:
> 
> * `TCPBaudRate <x>`: sets the baudrate for serial (only 8N1 mode), min `1200`, max `115200` by 1200 increments.
> * `TCPStart <port>`: listens to port `<port>`. This features supports 2 parallel TCP connexions, which can be useful if you need a terminal + a specific protocol (like XMODEM). The 3rd connection will disconnect an previous connection. The number of parallel connections is a compile-time option.
> * `TCPStart 0` or `TCPStart`: shuts down the TCP server and disconnects any existing connection.
> 
> For security reasons, the TCP bridge is not started at boot, and requires an explicit `TCPStart` command (can be automated with Rules).

Adminiuga, 2020-09-04:
- Not a zigpy issue
- Works with latest pyserial in select radio lib implementaitons.

closed 2020-09-04 by Adminiuga

Hedda, 2020-12-15:
@Equidamoid FYI, started a new separate discussion about possible Zigbee bridge solutions for zigpy here -> https://github.com/zigpy/zigpy/discussions/580

robajr, 2021-02-23:
It feels like it's 98% there already. socat already exists on HassOS via apk, but is gated by read only permissions on /dev/ (IOW - you can reach the port, but it can't mount). You can validate this by mounting to /tmp/XXX

If there were a plugin that could mount socat hardware with elevated rights to mount to /dev, we would have a "local" solution that would solve this already (or another OS adjustment to /dev/ RW on root or other).

tube0013, 2021-02-23:
Works out of the box. Just specify the port as socket://host:port

MattWestb, 2021-02-23:
Its also working out of the box with tuya TBGW that is having Ethernet only installing the Socat in the device (getting the root password is needed) and connecting to its port. Shall being possible doing the same with all coordinators that using serial port for communicating with one ESP and tasmota or ESPHome with WiFi or Ethernet if the ESP board supporting it.

MattWestb, 2021-02-23:
The good thing its working great also is running HA in docker and dont need struggling with mapping comports from the host system to the container.

Hedda, 2021-02-23:
As this is closed for zigpy please instead discuss the ZHA (Home Assistant) feature request part in -> https://github.com/zigpy/zigpy/discussions/667