zigpy/zigpy #303: home-assistant light status not updated after power on using a light switch

henne49, 2020-02-29:
Hi,

when my Ikea Tradfri light devices using zha are powered on, their status change is not correctly reflected to home-assistant. If I control zha devices from ha, all is good. Devices are not bind to any remote control or dimmer. 

I also understand if there would only a delay of a a couple of seconds to a minute, but it does not change until I try to power on or off from ha, wait a long time > 15 min? then the status is right again.

I can see below messages in ha logs, so the devices is recognized, when powered on, but, why is the status not automatically polled and ha gui updated? When I issue on_off cluster command, then I get the right status. If this cannot be fixed using zigpy, what would be an service call to poll every minute for now?

2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e03000700aa00
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [170, 0]
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1703000800010001
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1703002a0023002202fdff00024cfa00000013000c00814cfa3a9625feff570b008e00afdf130c0700be
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfa4c>, 0, 0, 19, b'\x81L\xfa:\x96%\xfe\xffW\x0b\x00\x8e', 0, 175, 223, 19, 12, 7, 0, -66]
2020-02-29 14:49:57 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0xfa4c, 00:0b:57:ff:fe:25:96:3a
2020-02-29 14:49:57 INFO (MainThread) [zigpy.application] Device 0xfa4c (00:0b:57:ff:fe:25:96:3a) joined the network
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:0b:57:ff:fe:25:96:3a
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy.zdo] [0xfa4c:zdo] ZDO request 0x0013: [0xfa4c, 00:0b:57:ff:fe:25:96:3a, 142]
2020-02-29 14:49:57 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfa4c>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'814cfa3a9625feff570b008e'

thank you

Adminiuga, 2020-02-29:
This is by design. Physically powered off devices cannot be queried. HomeAssistant-core is going to mark them unavailable in two hours when there's no traffic from the lights.

henne49, 2020-02-29:
but when I physically power on a device and see it as above on the logs, why is then HomeAssitant-core not showing the light as on, which is the default what happens with Ikea devices the are turned on, when power comes back?

for powered off devices, I get it 2h might be too long, but that is not my main problem.

Adminiuga, 2020-02-29:
Yes, when you power on the bulb, it joins the network again and can be queried again. 
```
2020-02-29 14:49:57 INFO (MainThread) [zigpy.application] Device 0xfa4c (00:0b:57:ff:fe:25:96:3a) joined the network
```
ZHA integration expects the light to report its status.. It does not do polling. When light is physically off it's not sending any updates.

henne49, 2020-02-29:
I understand the physical off and that is not what I'm after, only the case after power on. 

It seems that the bulbs do not send an update after Power on, but the LED Driver of IKEA does sometimes. Anyway, is there a way to get an event or any kind of trigger from join network, that I can use to query the status? That would solve my problem.

I remember deconz and original IKEA hub did show correct status, after power on, but may be wrong here.

puddly, 2020-03-01:
I use this event to automatically reset my lights to their previous state after they physically power off and on:

https://github.com/zigpy/zigpy/blob/8236bc1a037b4a296cf565de2341471cd5fcecd5/zigpy/zdo/__init__.py#L69-L70

henne49, 2020-03-01:
cheers, would you mind giving me some more guiding information how that works, I listened to * events,  all events on ha and was not able to see any message relating to a zigbee device, when I powered the bulb on. 

Related to that question, where are these details like OTA updated documented or how to listen to events? I was quering ha documentation, but that seems to not have this kind of information.

Adminiuga, 2020-03-01:
Documentation needs some love. Unfortunately current structuring is not quite suitable so efforts were stalled until HA integration docs are restructured. Until then, zigbee channel on discord server is your best bet.

henne49, 2020-03-01:
well, then I need to have a look on discord!

Adminiuga, 2020-03-05:
This is strange. Although we do not handle ZDO device_annce message, when you power on the light it should rejoin the network and upon a rejoin, we do call `async_initialize()` on "on_off" channel which should refresh light state.

henne49, 2020-03-06:
I can see the bulb joining by going from uninitialized to off, but it does not have the right status. If I run the cluster attribute command it is right. Do you need any logs for this? Here are 2 join requests:
https://pastebin.com/Ec6MUxEf
and 
https://hastebin.com/ebehasojir.rb

Adminiuga, 2020-03-06:
Is that all? Was expecting to see homeassistant.components.zha logs as well querying the light and channel initialization

henne49, 2020-03-06:
yes using this log setup:

logger:
  default: info
  logs:
    zigpy: debug
    homeassistant.components.zha: debug

Adminiuga, 2020-03-06:
add:
zigpy_deconz.zigbee.application: debug
zigpy_deconz.api: debug

henne49, 2020-03-06:
done, but still not more than this https://hastebin.com/xajufikico.rb

Adminiuga, 2020-05-10:
Please try with latest HA version and reopen if the issue still persists.

closed 2020-05-10 by Adminiuga