zigpy/zigpy #307: Unhandled timeout error

balloob, 2020-03-01:
I see this unhandled error in Home Assistant: (running 107dev)

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 172, in request
    use_ieee=use_ieee,
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 503, in request
    self.direct, device.nwk, aps_frame, message_tag, data
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

I then also notice this assertion error later:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 858, in _handle_cluster_request
    await self._handle_image_block(*args, tsn=tsn)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/general.py", line 977, in _handle_image_block
    tsn=tsn,
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 162, in request
    with self._pending.new(sequence) as req:
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 239, in new
    return Request(self, sequence)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 208, in __init__
    assert sequence not in pending
AssertionError
```

I don't know if they are related, but they originate in same place so I thought I would mention it.

puddly, 2020-03-01:
It looks like this is happening during the middle of an OTA upgrade and you have more than 256 pending image block responses, which is a little strange. What device(s) are you upgrading? Also, could you post the full zigpy/bellows logs?

Adminiuga, 2020-03-01:
What device and is it battery operated? I've seen something similar happening, when you could see traffic from the far end device, but device was not getting our updates. 

Can you enable zigpy, bellows.zigbee.application and bellows.ezsp debug logging, try to catch this situation again and DM me the logs?

balloob, 2020-03-02:
Aaah, logs. Makes sense. I'll put it on my to do. It's happening in prod so logs is a bit more cumbersome (somebody should fix that :P)

FYI, I have some test Zigbee equipment that is disconnected from power: an IKEA light, an IKEA switch, and a Sylvania Smart+.

PR #312 (2020-03-04): Catch exceptions when scheduling tasks in handlers.
Fixes #307
closed 2020-03-04 by Adminiuga