zigpy/zigpy #308: [REQUEST] Support more network meshes with seperate coordinators - Multiple adapter support?

Hedda, 2020-03-02:
Not sure if this request is really for zigpy or more for the zha implementation in Home Assistant. 

Would it be possible for zigpy to support more than coordinator at the same time as long as each coordinator have their own mesh network?

Personally I would like to have unstable devices on their own mesh network and would like to achieve this by using multiple adapters in zha.

This could also be good for testing as one Home Assistant instance could possible have one hardware adapter of each type?

PS: I know that the Zigbee standard is limited to only one coordinator per Zigbee network.

Adminiuga, 2020-03-02:
This is more for ZHA integration for Homeassistant-core. There could be only one coordinator on Zigbee Network. And ZHA integrations allows configuration only of a single coordinator. 

ATM supporting more than one instance of ZHA integration is not on the priority list. And IMO unless it is some experiment, I personally see little benefit of running two Zigbee networks in parallel, as you potentially get two weakier mesh networks, instead of a stronger single mesh networks.

Adminiuga, 2020-03-02:
You could try copying and renaming ZHA integration folder into `custom_component` and run second network as a custom component. Needs update to `manifest.json` as well. Just a guess, haven't tried personally.

Gamester17, 2020-09-14:
I think this idea of multiple adapters might be more interesting now that bellows for example support new and old EZSP protocols.

Might be a good idea to put unstable devices on a separate adapter if you are buying a new one anyway to upgrade to Zigbee 3.0

Another reason would be the WOW Factor to have a separate adapter and Zigbee network for testing new devices at home.

Wife does not like it when mess around with the devices that are already working as they are practically "in production" at home.

Gamester17, 2020-09-14:
Another readon is lmitation of amount of "Direct children" devices supported by Z-Stack firmware on CC2530 and CC2531

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

Z-Stack | Device | Zigbee | Direct children | Routes | Notes
-- | -- | -- | -- | -- | --
Z-Stack_Home_1.2 (default) | CC2531 | 1.2 HA | 20 | 30/0 |  
CC2530, CC2530 + CC2591, CC2530 + CC2592 | 1.2 HA | 16 | 30/0 |  
Z-Stack_Home_1.2 (source_routing) | CC2531, CC2530, CC2530 + CC2591, CC2530 + CC2592 | 1.2 HA | 5 | 40/40 |  
Z-Stack_3.0.x | CC2531 | 3.0 | 15 | 40/0 | - Discussion #1445- Max 40 Zigbee 3.0 devices
CC2530, CC2530 + CC2591, CC2530 + CC2592 | 3.0 | 10 | 40/0 | - Discussion #1445- Max 40 Zigbee 3.0 devices
CC2538 + CC2592 | 3.0 | 100 | 200/400 | - Discussion #1568- Max 200 Zigbee 3.0 devices
Z-Stack_3.x.0 | CC26X2R1, CC1352P_2 | 3.0 | 50 | 100/200 | - Discussion #1429- Max 200 Zigbee 3.0 devices

Adminiuga, 2020-09-14:
running multiple meshes to solve the limitation of the older CC2530 and CC2531 is a wrong approach here. 
ZHA would need some modifications to support multiple instances, but with current resources that is at the bottom of priority list.

Hedda, 2020-09-14:
Also checkout feature request https://github.com/zigpy/bellows/issues/342 which would allow multiple PANs on a single adapter.

Adminiuga, 2020-09-14:
if you want to support multiple child devices, you add more routers.
Having 100 child devices all joined to a single parent only works if all those children are in the same room.

closed 2021-09-07 by github-actions[bot]