zigpy/zigpy #318: [SUGGESTION] Make more developers admin permission for the zigpy organization?

Hedda, 2020-03-10:
@rcloran can you give more developers administrator permission to the upstream zigpy organization?

The problem as I understand it is that without you giving them full admin permissions your fellow zigpy developers can at this time not create new repositories under the under zigpy organization or migrate/move existing related repositories under zigpy organization a repository was initially created elsewhere.

- https://github.com/zigpy/zigpy

I like to suggest that you consider giving full admin permissions for the zigpy organization to the main code contributing developers who are still active and are willing to directly or indirectly collaborate respectfully together on zigpy and ZHA related projects in different repositories that all fall under the same umbrella organization on GitHub, like; @Adminiuga & @damarco & @AndreasBomholtz & @dmulcahey & @puddly & @doudz & @presslab-us & @Yoda-x

- https://github.com/zigpy/zigpy/graphs/contributors

This is a follow-up to issue https://github.com/zigpy/zigpy/issues/113 which was about giving more developers repo permissions.

New related repositories who creators might want to migrate/move them to zigpy organization are:

 - https://github.com/zha-ng/zigpy-znp = zigpy-znp by Adminiuga
 - https://github.com/zha-ng/zigpy-deconz-parser = zigpy-deconz-parser by Adminiuga
 - https://github.com/zha-ng/zha-map = zha-map by Adminiuga
 - https://github.com/doudz/zigpy-zigate = zigpy-zigate by doudz
 - https://github.com/sanyatuning/zigpy-cc = zigpy-cc by sanyatuning
 - https://github.com/Equidamoid/pyconz = pyconz by Equidamoid
 - https://github.com/dmulcahey/zha-device-handlers = zha-device-handlers by dmulcahey

Perhaps only some of those developers want to move some repositories under the zigpy organization.

Hedda, 2020-03-10:
Have anyone been in contact with @rcloran recently or know anyone else who has talked to him? 

@Adminiuga & @dmulcahey who are working actively on ZHA for HA might know how to reach him?

I believe the best long-term would be if @rcloran assign additional admins upstream to the existing zigpy organization, both most external references and other projects already links there and you do not have to get copies of issues and separate pull request discussions all over the place when developers are forced to create related projects elsewhere.

If @rcloran is truly and completely incommunicado then perhaps someone who helped him with zigpy or bellows early on like @AndreasBomholtz or @damarco who has previously contributed a lot of code to this existing zigpy organization could contact GitHub support about having GitHub assigning one of you guys as an additional admin of this existing zigpy organization with full administrator permissions. Hopefully, they have a procedure for this scenario that does not require forking.

Again, I'm sure that GitHub support would be willing to one or more additional owners of this existing zigpy organization if other code contributing developers explain that the founding member of this zigpy organization has gone 'missing in action' and that for it to go on best would be if someone else could also be assigned admin to this zigpy organization until if and when @rcloran comes back.

rcloran, 2020-03-11:
@Adminiuga and @damarco are both owners of the zigpy organization on Github.

dmulcahey, 2020-03-11:
@rcloran thanks a bunch! Are you back or is this just a quick pop in?

Adminiuga, 2020-03-11:
@rcloran Thank you!!!

Hedda, 2020-03-12:
@rcloran Thank you! Please consider to maybe do the same for these related PyPI projects as well? 

- https://pypi.org/project/zigpy
- https://pypi.org/project/bellows
- https://pypi.org/project/zigpy-xbee

Not sure if there are other zigpy related projects on PyPi which rcloran is the only owner/admin of?

Anyway, I believe rcloran is the only owner/admin of those and that is why these others now exist?

- https://pypi.org/project/zigpy-homeassistant/
- https://pypi.org/project/bellows-homeassistant/
- https://pypi.org/project/zigpy-xbee-homeassistant/

As I understand some developers use these in other projects than Home Assistant so it would, therefore, be most correct if PyPI projects names without a "homeassistant" suffix could be used. 

Thank you again @rcloran for all your work on ZHA, zigpy and all related projects! 

PS: I apologize to all if my OCD is very annoying to handle sometimes :/

Hedda, 2020-03-12:
@Adminiuga and @damarco Will you now please consider making more devs owners or admins too?

I don't exactly know which developers are still active today, perhaps @dmulcahey and @puddly or?

Hedda, 2020-03-30:
@rcloran Could you maybe also OK the renaming of bellows library & repo at https://github.com/zigpy/bellows/issues/215 ?

Hedda, 2020-04-01:
@rcloran I see that you have now added @damarco to zigpy project on PyPi, so wondering if you can also add @Adminiuga @dmulcahey and @puddly to both zigpy & bellows projects on PyPi?

- https://pypi.org/project/zigpy
- https://pypi.org/project/bellows

That way there at least a few more trusted people who are owners/admins for those projects too, and there might not even be a need to keep these duplicated PyPI projects with "homeassistant" suffix:

- https://pypi.org/project/zigpy-homeassistant
- https://pypi.org/project/bellows-homeassistant

PS: Not seen that @damarco been active with Zigpy for quite some time, however, @Adminiuga @dmulcahey and @puddly all seem to be very active (considering these are all just hobby projects).

Hedda, 2020-04-28:
FYI, I just read the news that GitHub basic free account now allows teams private repositories with unlimited collaborators at no cost to make team collaboration easier, see:

- https://github.blog/2020-04-14-github-is-now-free-for-teams/

GitHub is now free for teams
Nat Friedman

We’re happy to announce we’re making private repositories with unlimited collaborators available to all GitHub accounts. All of the core GitHub features are now free for everyone. 🎉

Until now, if your organization wanted to use GitHub for private development, you had to subscribe to one of our paid plans. But every developer on earth should have access to GitHub. Price shouldn’t be a barrier.

This means teams can now manage their work together in one place: CI/CD, project management, code review, packages, and more. We want everyone to be able to ship great software on the platform developers love.

Teams who need advanced features (like code owners), enterprise features (like SAML), or personalized support can upgrade to one of our paid plans.

We’re also reducing the price of our paid Team plan from $9 per user/month to $4 per user/month, effective immediately. Existing customers will have their bills automatically reduced going forward.

Learn more in the FAQ, or compare plans on our pricing page.

- https://help.github.com/github/getting-started-with-github/faq-about-changes-to-githubs-plans

- https://github.com/pricing

closed 2021-09-07 by github-actions[bot]