zigpy/zigpy #319: [SUGGESTION] Migrate related zigpy project repositories to upstream zigpy organization?

Hedda, 2020-03-10:
@rcloran I like to suggest that zigpy related project repos are moved to upstream zigpy organization.

This directly relates to suggestions/discussion in https://github.com/zigpy/zigpy/issues/318 and and https://github.com/zigpy/zigpy/issues/113

Existing and future developers of zigpy, the ZHA integration for Home Assistant, and related zigpy or ZHA projects would probably benefit if all or at least most zigpy and ZHA related projects which fall under the same umbrella was migrated/moved to the upstream zigpy organization on GitHub.

- https://github.com/zigpy/zigpy

New related repos who have creators that might want to migrate/move them to zigpy organization:

https://github.com/zha-ng/zigpy-znp = zigpy-znp by @Adminiuga
https://github.com/zha-ng/zigpy-deconz-parser = zigpy-deconz-parser by @Adminiuga
https://github.com/zha-ng/zha-map = zha-map by @Adminiuga
https://github.com/doudz/zigpy-zigate = zigpy-zigate by @doudz
https://github.com/sanyatuning/zigpy-cc = zigpy-cc by @sanyatuning
https://github.com/Equidamoid/pyconz = pyconz by @Equidamoid
https://github.com/dmulcahey/zha-device-handlers = zha-device-handlers by @dmulcahey

Perhaps only some of those developers want to move some repositories under the zigpy organization.

Equidamoid, 2020-03-11:
Just wanted to say that pyconz in kind of dead right now. 
I still want to make a lowlevel-enough zigbee to some rpc bridge, but right now my zigbee network is pretty unstable by some reason and it makes development practically impossible.

Hedda, 2020-03-12:
@dmulcahey would you now consider migrating/moving your zha-device-handlers project to the zigpy organization now that @Adminiuga is also co-owner of the zigpy and (I assume) would very likely also make you a fellow co-owner of the zigpy organization as well?

- https://github.com/dmulcahey/zha-device-handlers

@dmulcahey related follow-up question is if your zha-device-handlers project is really hardcoded to ZHA or generic for all zigpy projects? If not hardcoded to ZHA then would it maybe be a good idea to rename your project from "zha-device-handlers" to "zigpy-device-handlers" and then migrate/move that repository to the zigpy organization?

If your zha-device-handlers project is hardcoded to ZHA then might it be a good idea to try to remove all such hardcodings in order to try to make it a more generic device-handlers repo for zigpy so that other projects than ZHA for Home Assistant can also more easily take advantage ot it.

- https://github.com/zigpy

@Adminiuga may I ask you same questions about your zha-map and zigpy-znp projects?

- https://github.com/zha-ng/zha-map
- https://github.com/zha-ng/zigpy-znp

zigpy-znp you might just want to move/migrate to the zigpy origination right away, but would it not maybe also a good idea to try to make it so that the zha-map also belonged under the same zigpy origination too? Either 'as is' or make it more generic to can more easily be used by any project that want to utilize zigpy and not just for ZHA in Home Assistant?

 @Adminiuga & @dmulcahey My main point here is still only that is would be great is all zigpy related projects was moved to under the zigpy organization at https://github.com/zigpy/ as a shared umbrella so that will be easy for every new developer to find all related projects in one location.

dmulcahey, 2020-03-12:
We are already discussing this and planning to do it. Hold off on suggestions for a few weeks and give us some time to let the dust settle 😀

Hedda, 2020-03-12:
Nice! Sorry, I will cool down, blaming my OCD :P

Hedda, 2020-03-12:
Nice! Sorry, I will cool down from now on, blaming my OCD :P

dmulcahey, 2020-03-12:
> Nice! Sorry, just love ZHA usability experience concept, will cool down from now on, I blame my OCD :P

I just meant for a week or 2 so we can figure some stuff out. Not forever 😀

sanyatuning, 2020-03-18:
I like the idea moving zigpy-cc to zigpy org.

Adminiuga, 2020-03-19:
I've extended the invitation. Please make sure 2FA is enabled on your account.

sanyatuning, 2020-03-19:
Zigpy cc transfered! Thanks for the membership

Hedda, 2020-03-20:
@Equidamoid even if off-topic here I was very curious what you meant by "make a lowlevel-enough zigbee to some rpc bridge"? Do you mean something like the design described in this not yet realized project by @vitiral ? => https://github.com/vitiral/bridge-rpc/blob/master/README.md

Adminiuga mentioned in https://github.com/zigpy/bellows/issues/183 that he thought that it would be better to implement something like https://github.com/cgarwood/homeassistant-zwave_mqtt for https://github.com/OpenZWave/qt-openzwave which is a wrapper for https://github.com/OpenZWave/Zwave2Mqtt (Zwave2Mqtt "Zwave to MQTT gateway" project by the OpenZWave organization). 

I understand that the Home Assistant Core developers are already working on new native integration for a Z-Wave controller/hub implementation based on that zwave2mqtt project, see: https://developers.home-assistant.io/blog/2020/02/04/new-zwave/

I guess for similar reasons as for https://github.com/Koenkk/zigbee2mqtt they use MQTT because it has already been adopted by most open-source DIY home automation applications.

I posted some suggestions in https://github.com/zigpy/zigpy/issues/297 which might be related as it was a request for "Add network support for Zigbee adapters" but my suggestions there was a more basic "serial over TCP/IP" idea which has huge downsides when it comes to usability.

vitiral, 2020-03-20:
Notice: bridge-rpc was a thought experiment and is in no way real software :smile:

Equidamoid, 2020-03-20:
@Hedda, 
Continuing offtopic =)
No not that protocol. I wasn't sure about protocol at the moment. 
For the experiment it was dbus. It actually worked(*) and had all the perks of "proper" rpc, like "method call" concept or built-in introspection, so you can take a thirdparty tool (such as d-feet or qdbusviewer) and manipulate devices as "objects". 

MQTT is not really made for such "call-response" way of working and the common way of abusing it for such purpose feels a but... workaround'ey to me.

For network support there is already a "zigbee encapsulation protocol". No need to do even that via mqtt.

(*) the dbus-zigbee part worked fine, I was able to call things like "OnWithTimedOff", get device properties, etc. The network itself misbehaved (seems to be fixed recently for me).

Hedda, 2020-04-21:
@Equidamoid & @vitiral Continuing offtopic ;) Follow-up regarding your "rpc bridge" ideas =)

Have you guys heard of the new Sonoff ZBBridge (Sonoff Zigbee Bridge) that was just announced?

See the related requests for zigpy-> https://github.com/zigpy/zigpy/issues/385 and here https://github.com/zigpy/bellows/issues/183

Sonoff ZBBridge is basically a complete Wi-Fi (ESP8266) to Zigbee (EmberZNet/EZSP) bridge/gateway

Hedda, 2020-08-10:
@puddly Suggest that you consider forking or moving zha-custom-radios repo to zigpy organization on GitHub for reference:

https://github.com/zha-ng/zha-custom-radios

@Adminiuga While is no longer used by @puddly for his [zha-ng/zigpy-znp](https://github.com/zha-ng/zigpy-znp) or others projects currently it might be good to keep a copy of this repo under the zigpy organization so that other new developers can easily find it if they would wish to develop new radio libraries for zigpy and ZHA. 

https://github.com/zigpy

puddly, 2020-08-10:
It's unused because it doesn't work. I wanted something installable through HACS but it isn't possible to make the patching code run early enough without making it a legacy custom component, which is what is inside of zigpy-znp.

It might be better to write developer documentation for Zigpy itself, including how to write a radio library, and link to the working custom component there.

Adminiuga, 2020-08-10:
same with `zha_custom` it is very raw and it is used just for "hot" testing of a few things.

Adminiuga, 2020-09-04:
Any projects suitable for General Release will be considered to be included into zigpy organization.

closed 2020-09-04 by Adminiuga

Hedda, 2020-09-07:
@Adminiuga & @puddly I just noticed that zigpy-znp (v0.1.1) will be supported soon by the upcoming Home Assistant so will that its repo be considered to be included into zigpy organization and as such have its repository moved to https://github.com/zigpy or?

https://github.com/home-assistant/core/pull/39700

https://github.com/zha-ng/zigpy-znp

PS: @puddly & @Adminiuga Congratulations on the milestone of adding zigpy-znp support to Home Assistant ZHA integration!