zigpy/zigpy #325: Change attribute Bool

albalaing, 2020-03-20:
When I try to change an attribute of type Bol from the UI I get the following error:

`2020-03-20 13:13:07 ERROR (MainThread) [zigpy.zcl] [0x33ea:1:0x0010] 'Bool.false' is not a valid Bool
2020-03-20 13:13:08 ERROR (MainThread) [zigpy.device] Failed to parse message (b'1839049a') on cluster 16, because Data is too short to contain 2 bytes'`

I have tried false 0 but it always gives the same error

puddly, 2020-04-02:
The error seems to be caused by `Cluster.write_attributes` being passed an invalid bool:

```Python
In [1]: import zigpy.types as t
In [2]: try:
  ...     t.Bool('Bool.false')
  ... except Exception as e:
  ...     print(str(e))
  ...
'Bool.false' is not a valid Bool
```

`Bool.false`, `0`, and `false` are all invalid values if you pass them as strings. Is there no way to specify the data type?

Adminiuga, 2020-04-02:
> When I try to change an attribute of type Bol from the UI I get the following error:

UI is currently limited and does not support specifying the type of data, everything is passed as a string and for now it only works for `int`s and `CharacterStrings`. 
For now the only option is to use `zha.set_zigbee_cluster_attribute`

eg from dev panel services call `zha.set_zigbee_cluster_attribute` with following data:
```
ieee: 00:0d:6f:00:05:7d:2d:34
endpoint_id: 1
cluster_id:  16
cluster_type: "in"
attribute: 85
value: 0
```

Adminiuga, 2020-04-02:
Not zigpy UI but ZHA integration UI in HomeAssistant. Per se this is a HA ZHA issue.

albalaing, 2020-04-12:
I have tried calling the service zha.set_zigbee_cluster_attribute and it gives the same error.
Is there a way to write a Bool attribute?
Can it be done with a ZHA custom quirks? 
This should be able to be done as there are devices that need a particular configuration.

Adminiuga, 2020-04-12:
Home assistant core version?
Debug logs?

albalaing, 2020-04-12:
zha.set_zigbee_cluster_attribute with following data:
`ieee: '00:15:8d:00:02:2a:b5:52'
endpoint_id: 1
cluster_id: 6
cluster_type: "in"
attribute: 0 
value: 0`

Log:
` ERROR (MainThread) [zigpy.zcl] [0xad8f:1:0x0006] '0' is not a valid Bool  `
Now the message from [ziggy.device] is not coming out

Version:
`arch: x86_64
dev: false
docker:  false
hassio: false
name: Linux
os version: 4.15.0-91-generic
python_version: 3.7.7
timezone: Europe/Madrid
version: 0.108.3
virtualenv: true
`

Adminiuga, 2020-04-12:
So the problem is really with ZHA, not with zigpy. ZHA services passes it as a `str` and it causes problem. 
As a workaround, change in your setup https://github.com/home-assistant/core/blob/200e140e8094fb645923c6430c057b3052076100/homeassistant/components/zha/api.py#L102 to be `vol.Required(ATTR_VALUE): vol.Any(int, cv.bool, cv.string),`

and raise this issue in home assistant. 
On the side note, you really can't set attribute `0x0000` of the `on_off` cluster, in most instances it is a read only attribute. And where it's not, it is a case of improper implementation.

albalaing, 2020-04-13:
I have made the patch you have told me and it has worked ok
I wanted to activate interlock mode in Aqara wireless relay controller
`ieee: '00:15:8d:00:03:cb:e6:0d' 
endpoint_id: 1
cluster_id: 16
cluster_type: in
attribute: 65286 
value:  1
manufacturer: 4447`

Another question has arisen
I want to change the reverse direction mode on the Aqara curtain motor ZNCLDJ11LM 
To do this, send Data Type: Character String (0x42) with these data 070002000000ff01
How can it be done?

Adminiuga, 2020-04-13:
I hate when xiaomi uses CharacterString for binary data.
I haven't tried sending `bytes` as a CharacterString in zigpy. Not positive this is going to work. And at very least you would need to create a zigpy quirk for this device, as 0x0401 is not a ZCL attribute and zigpy needs that information in order to serialize data correctly. See `zha-device-handlers` package on how to create a quirk.

Adminiuga, 2020-04-13:
I've checked and unfortunately you won't be able to serialize data correctly for CharacterString data type for raw bytes. You would need to do zigpy calls at lower levels, essentially sending raw data. 
you would need to call something like:
```
import zigpy.types as t
import zigpy.zcl.foundation as foundation

cluster.request(True, foundation.Command.Write_Attributes, (t.uint8_t, t.LVBytes), (0x042, b"\x00\x02\x00\x00\x00\xff\x01"), manufacturer=4447)
```

albalaing, 2020-04-14:
I have created a quirk with:
````Python 
class BasicClusterAdd (Basic):
	def __init__(self, *args, **kwargs):
		self.attributes.update({ 0x0401: ("Xiaomo reverse", t.CharacterString)})
		super().__init__(*args, **kwargs)
````
I have done tests of mailing stings and I have put a snifer to see what it commands and to see what has really been programmed:

```
Send string: '2223333'
Write : 07 32 32 32 33 33 33 33 
Read:   07 00 02 00 01 00 FF 01   (Reverse) 
```
```
Send string: '2222222'
Write : 07 32 32 32 32 32 3 32 
Read:   07 00 00 00 00 00 FF 01   (Normal) 
```
This way I can switch to reverse mode, although it is the right way.

As for how things have been implemented Xiaomi I agree with you. He has done what he wanted.

Adminiuga, 2020-04-14:
Sounds like you are making progress :thumbs_up: just don't do `self​.attributes.update` this would make changes for all, instead make a local copy and update the local copy, something like 
```
attributes = Basic.attributes.copy()
attributes.update ...
```

Gamester17, 2020-09-14:
Has this issue been resolved with latest Home Assistant, zigpy, its radio libraries, and/or latest zha-device-handlers/zha-quirks?

albalaing, 2020-09-16:
With version 0.114.4 it is still not possible to set a bool value

closed 2020-09-25 by dmulcahey

albalaing, 2020-10-13:
86/5000
I don't know if it will be a mistake. 
But the change that Adminiuga proposed in api.py was in line:
```vol.Required(ATTR_VALUE): vol.Any(int, cv.bool, cv.string),```
And it has been done in the previous line:
``` vol.Required(ATTR_ATTRIBUTE): vol.Any(int, cv.boolean, cv.string),```
Therefore, it is still not possible to enter Bool values

Adminiuga, 2020-10-13:
it is `cv.boolean` https://github.com/home-assistant/core/blob/52c93edb53b7b47a1d37107076853d8d3236be82/homeassistant/helpers/config_validation.py#L149 I see no difference with the original proposal.

You need to be more specific of exactly what you are doing and how

Adminiuga, 2020-10-13:
Doh, it is `ATTR_VALUE` 🤦

Adminiuga, 2020-10-13:
but it does work, cause I use it currently without issues, doesn't it?

Adminiuga, 2020-10-13:
it does work and the change was correct:
`vol.Required(ATTR_ATTRIBUTE): cv.positive_int,` --> `vol.Required(ATTR_ATTRIBUTE): vol.Any(int, cv.boolean, cv.string),`

When calling the service, make sure the value parameter is not in double quotes

albalaing, 2020-10-14:
If I make the call to service zha.set_zigbee_cluster_attribute
```
ieee: '00:15:8d:00:03:66:96:cd'
endpoint_id: 1
cluster_id: 16
cluster_type: in
attribute: 65286 
value: true
manufacturer: 4447
```

Adminiuga, 2020-10-14:
doh, i do see the problem now

smmoroz, 2020-11-05:
Hi, @Adminiuga ,

I'm runnning 0.117.4 now. And there is an issue with the fix https://github.com/home-assistant/core/pull/41850/commits/08b856afe4fa2b4d6609de66c2cd5a68a7daff7f.

If an attribute type is `zigpy.types.basic.data16` then for any numeric `value` as an argument for `zha.set_zigbee_cluster_attribute` I have this error:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 135, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1448, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1483, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 552, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 932, in set_zigbee_cluster_attributes
    response = await zha_device.write_zigbee_attribute(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 559, in write_zigbee_attribute
    response = await cluster.write_attributes(
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 351, in write_attributes
    args = self._write_attr_records(attributes)
  File "/usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py", line 342, in _write_attr_records
    a.value.value = python_type(value)
TypeError: 'int' object is not iterable
```
In order to solve the issue I made this change:
```
vol.Required(ATTR_VALUE): cv.string
```
And passed `value : 11`. I guess that it was considered as `[1, 1]` during the conversion phase. See debug output below:

```
2020-11-05 03:00:59 DEBUG (MainThread) [zigpy.zcl] [0x0dc4:1:0xfc01] XXX attr_id = 0 (type = <class 'zigpy.types.basic.data16'>), a = Attribute(attrid=0, value=<TypeValue type=data16, value=['1', '1']>) (a.type = <class 'zigpy.zcl.foundation.Attribute'>), value = 11 (type = <class 'str'>)
2020-11-05 03:00:59 DEBUG (MainThread) [zigpy.zcl] [0x0dc4:1:0xfc01] YYY python_type = <class 'zigpy.types.basic.data16'> python_type(value) = ['1', '1']
```
I also guess that another way to solve the issue is to pass `value : "11"`. 

Is it your original idea to deal with `data16` types?

Adminiuga, 2020-11-05:
how do you call the service, exactly. Cause I can confirm it works, cause I use it:
```
2020-11-04 00:05:05 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=set_zigbee_cluster_attribute, service_data=ieee=50:0b:00:00:00:00:00:00, endpoint_id=1, cluster_id=65281, attribute=16, value=100>
2020-11-04 00:05:05 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xD385, EmberApsFrame(profileId=260, clusterId=65281, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=74), 75, b'\x04\x9c\x11J\x02\x10\x00)d\x00')
2020-11-04 00:05:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'001b'
2020-11-04 00:05:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040101ff01014001000083c0cc85d3ffff061c9c114a040004'
2020-11-04 00:05:05 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=65281, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=131), 192, -52, 0xd385, 255, 255, b'\x1c\x9c\x11J\x04\x00']
2020-11-04 00:05:05 DEBUG (MainThread) [zigpy.zcl] [0xd385:1:0xff01] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4508 tsn=74 command_id=Command.Write_Attributes_rsp>
2020-11-04 00:05:05 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD385](TH1124ZB): set: 100 for attr: 16 to cluster: 65281 for ept: 1 - res: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
```

Adminiuga, 2020-11-05:
the schema is `vol.Required(ATTR_VALUE): vol.Any(int, cv.boolean, cv.string)` so if you give it an `int`, you'll get an `int` for the output.

smmoroz, 2020-11-05:
This works for attributes with simple types. I tested int and bool. But it doesn't work for data16. If I pass 0x0101 it does not work. But if I pass string "11" it is properly converted to 0x0101.

Adminiuga, 2020-11-05:
pass decimal.

Adminiuga, 2020-11-05:
ah, it's a list. List won't work with the service

Adminiuga, 2020-11-05:
what is that attribute? you sure it is a list of 2 * uint-8?

smmoroz, 2020-11-05:
Here it is:

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/legrand/dimmer.py

```
class LegrandCluster(CustomCluster, ManufacturerSpecificCluster):
    """LegrandCluster."""

    cluster_id = MANUFACTURER_SPECIFIC_CLUSTER_ID
    name = "LegrandCluster"
    ep_attribute = "legrand_cluster"
    manufacturer_attributes = {
--->        0x0000: ("dimmer", t.data16),
        0x0001: ("led_dark", t.Bool),
        0x0002: ("led_on", t.Bool),
    }
```

The attributes are explained [here](https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Legrand-Netatmo-corner.md#cluster-0xfc01)

`data16` is [here](https://github.com/zigpy/zigpy/blob/dev/zigpy/types/basic.py):

```
class data16(FixedList, item_type=uint8_t, length=2):
    """General data, Discrete, 16 bit."""

    pass
```

BTW, as you can see, my trick with `"11"` string will not work for anything with 2 bytes, e.g. 0xffff.

BTW, as a side topic, can you please point me to the code snippet, how to make these `manufacturer_attributes` as HA entity attributes or even entities themselves.

Adminiuga, 2020-11-05:
wanna try `vol.Required(ATTR_VALUE): vol.Any(int, cv.boolean, [int], cv.string)` ? would need to call the services with
```yaml
args: [1, 1]
```

or
```yaml
args:
 - 1
 - 1
```

Adminiuga, 2020-11-05:
or have you tried a string like `"\x01\x01"` ?

claudegel, 2023-02-12:
I'm trying to add an attribute in manufacturer specific cluster for a ZHA quirck but I can't find the zigpy type for bool.
This is what I get from zha-toolkit scan ieee:
```
"attributes": {
            "0x0001": {
              "attribute_id": "0x0001",
              "attribute_name": "1",
              "value_type": [
                "0x10",
                "Bool",
                "Discrete"
              ],
              "access": "WRITE|READ",
              "access_acl": 3,
              "manf_id": 4508,
              "attribute_value": 1
```
what type t.??? I should use

Adminiuga, 2023-02-12:
t.Bool https://github.com/zigpy/zigpy/blob/dev/zigpy/types/named.py#L71

claudegel, 2023-02-13:
On last thing @Adminiuga, when we have an attribute of type t.enum8 in manufacturer cluster. the value we get is something like
enum8.undefined_0x04. How can I build enumeration list so we know what is the value 0x04 and all other possible values in the quirks.

puddly, 2023-02-13:
You subclass it (using `t.enum8` alone is never something you should do):

https://github.com/zigpy/zigpy/blob/819f8e7dc97e27627335a398dac22ad9a9b7688a/zigpy/zcl/clusters/lightlink.py#L60-L62

claudegel, 2023-02-13:
Thanks @puddly, it's really simple to implement. Do you have an idea for value_type Array for one of my manufacturer attribute? can't find in zigpy/types

puddly, 2023-02-13:
I don't think we have any devices using `Array` but it can be found as `zigpy.zcl.foundation.Array`.