zigpy/zigpy #326: Alot of warnings "FOREIGN KEY constraint failed" in log

MaximumSU, 2020-03-21:
I setup new zha network with some XBee modules. I adding, and configuring 8 modules, disable unneeded switches and sensors in device configs. All looking good except of one node. That node send the IO sampling of enabled pins every 1 minute. And every minute i see in log 6 messages
"FOREIGN KEY constraint failed":
```
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUJ\x1a\x03F\xe8\xe8\x00\x92\xc1\x05\x01\x01\x0c\xb8\x00\x00('
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:4a:1a, 0x0346, 232, 146, 1, b'010cb8000028')
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy.zcl] [0x0346:232:0x0092] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=0>
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy.zcl] [0x0346:232:0x0092] ZCL request 0x0000: [{'digital_pins': [0, 0, 0, 1, 1, 1, 0, 1, 0, 0, 1, 1, 0], 'analog_pins': [0, 0, 0, 0, 0, 0, 0, 0], 'digital_samples': [0, 0, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0], 'analog_samples': [0, 0, 0, 0, 0, 0, 0, 0]}]
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
```

Adminiuga, 2020-03-21:
The XBee modules don't behave like normal Zigbee devices, as they don't report active endpoints and clusters.
Stop HA and modify `zigbee.db`. Insert endpoint description for Xbee, like
```
INSERT INTO endpoints VALUES('00:xb:ee:ad:re:ss:11:83',232,260,24321,1);
```

That should fix the other problem too.

MaximumSU, 2020-03-21:
> 232,260,24321,1

What does this meen? Need to do it for all nodes with the same values?

Adminiuga, 2020-03-21:
Schema for the `endpoints` table:
```
CREATE TABLE endpoints (ieee ieee, endpoint_id, profile_id, device_type device_type, status, FOREIGN KEY(ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
```

Yes, do it for all XBees nodes which do IO sampling.

MaximumSU, 2020-03-21:
I look into endpoints table - the problematic node is here (but contain another digits):

00:13:a2:00:41:55:4a:1a | 230 | 49413 | 1 | 1
-- | -- | -- | -- | --
00:13:a2:00:41:55:4a:1a | 232 | 49413 | 1 | 1

Do i need to change the only digits for the "232" profile_id?
And the record for coordinator node looks strange:

00:13:a2:00:41:55:4a:30 | 230 | NULL|NULL | 0
-- | -- | -- | -- | --

Adminiuga, 2020-03-21:
Hrm... Add the 2nd one, but that begs the question whether the quirk is correct.

MaximumSU, 2020-03-21:
I try to add the new records (for problematic node and for coordinator node) with you parameters, but all what i have after starting "ValueError: '1' is not a valid Status" and ZHA not start:
```
2020-03-21 21:00:58 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/serial0 for zha
ValueError: '1' is not a valid Status

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/config_entries.py", line 216, in async_setup
    hass, self
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zha/__init__.py", line 101, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zha/core/gateway.py", line 137, in async_initialize
    self.application_controller = radio_details[CONTROLLER](radio, database)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_xbee/zigbee/application.py", line 32, in __init__
    super().__init__(database_file=database_file)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/application.py", line 52, in __init__
    self._dblistener.load()
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 321, in load
    self._load_endpoints()
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 382, in _load_endpoints
    ep.status = zigpy.endpoint.Status(status)
  File "/usr/lib/python3.7/enum.py", line 310, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.7/enum.py", line 564, in __new__
    raise exc
  File "/usr/lib/python3.7/enum.py", line 548, in __new__
    result = cls._missing_(value)
  File "/usr/lib/python3.7/enum.py", line 577, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: '1' is not a valid Status
```

Adminiuga, 2020-03-21:
Why is `'1'` in quotes? did you add it as a number or as a string?
can you do `sqlite3 zigbee.db .dump` ?

MaximumSU, 2020-03-21:
OK, about '1' it was the DB Explorer error - it interpreted the empty data type as TEXT. I set the correct data type, but no luck. When i added the new entry with "232" endpoint_id for coordinator node - it simply removed by HA from DB just after starting. When i changed the record with "232" endpoint_id for problematic node with your digits - no warnings, but data for that node stops updating also.

Adminiuga, 2020-04-02:
Fixed by #343

closed 2020-04-02 by Adminiuga